Koenkk/zigbee2mqtt #19925: [New device support]: Human Presence Sensor Pro - 24G Mmwave Human Presence Sensor Tuya Zigbee

klaneos, 2023-11-30:
### Link

https://www.aliexpress.com/item/1005005946786561.html?spm=a2g0o.order_list.order_list_main.47.702018024eGrXT

### Database entry

0xa4c138c36367379b (0x968B)

### Comments

Identified by Zigbee2MQTT as Zigbee model 'TS0601' and Manufacturer Name  '_TZE204_kyhbrfyl'

Thank you in advance.

klaneos, 2023-11-30:
Also found on TuYa EXPo site https://expo.tuya.com/product/1104014
Name: NEO ZigBee Human Presence Sensor Pro
Model: NAS-PS09B2

PM04290, 2023-12-19:
I have the same device; impossible to pair with coordinator.... is there a specific action to do it ? or just power On ?

jdmc, 2024-02-21:
Did anybody tried the following using a Tuya Zigbee Gateway?
I have the same device but missing a Tuya Gateway to configure it and find the info that is needed for accepting this device a supported in [Zigbee2MQTT] 
 
Support new TuYa devices
https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html 
Find Tuya Data Points
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#find-tuya-data-points 

Also checking that there are other topics with the same device that could give a possible solution for this device:

https://github.com/Koenkk/zigbee2mqtt/issues/19292 >> this has a possible integration

https://github.com/Koenkk/zigbee2mqtt/issues/20062

closed 2024-09-20 by github-actions[bot]