Koenkk/zigbee2mqtt #19931: Device Goes to 1% instead of Off

boojew, 2023-11-30:
### What happened?

When turning the device off in Z2M or HA, the device will revert to the state of 1%

### What did you expect to happen?

Go off :) 

### How to reproduce it (minimal and precise)

Using this device, 9/10 times, I just need to turn it on - wait any period of time - and turn it off. 

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20221226

### Adapter

zzh

### Debug log

```
Info 2023-11-30 12:55:32MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:35MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":36,"state":"ON"}'
Info 2023-11-30 12:55:35MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:35MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:36MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:36MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:36MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":36,"state":"ON"}'
Info 2023-11-30 12:55:36MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:37MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Info 2023-11-30 12:55:37MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":137,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":112,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":112,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"OFF"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":112,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":112,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"OFF"}'
Info 2023-11-30 12:55:46MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":112,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:49MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Info 2023-11-30 12:55:50MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Info 2023-11-30 12:55:52MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:52MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:52MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:53MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Info 2023-11-30 12:55:53MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:53MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:53MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:53MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:54MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:55:54MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Info 2023-11-30 12:56:03MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"OFF"}'
Info 2023-11-30 12:56:03MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Info 2023-11-30 12:56:05MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Debug 2023-11-30 12:58:19Received MQTT message on 'zigbee2mqtt/Kitchen - Fireplace Lights/set' with data '{"state":"ON","brightness":147}'
Debug 2023-11-30 12:58:19Publishing 'set' 'brightness' to 'Kitchen - Fireplace Lights'
Info 2023-11-30 12:58:19MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":147,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":43,"state":"ON"}'
Debug 2023-11-30 12:58:19Received Zigbee message from 'Kitchen - Fireplace Lights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-11-30 12:58:19MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":147,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
Debug 2023-11-30 12:58:22Received Zigbee message from 'Kitchen - Fireplace Lights', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":147}' from endpoint 1 with groupID 0
Info 2023-11-30 12:58:22MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":147,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
[... cut ...]
Debug 2023-11-30 12:58:36Received Zigbee message from 'Kitchen - Fireplace Lights', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":2}' from endpoint 1 with groupID 0
Info 2023-11-30 12:58:36MQTT publish: topic 'zigbee2mqtt/Kitchen - Fireplace Lights', payload '{"brightness":2,"do_not_disturb":null,"level_config":{"on_off_transition_time":50},"linkquality":40,"state":"ON"}'
```

Device Dev Console
```
{
    "brightness": 3,
    "level_config": {
        "on_off_transition_time": 50
    },
    "linkquality": 40,
    "state": "ON",
    "do_not_disturb": null
}
```

boojew, 2023-12-01:
I didnt mention, but this is the device [TS0052](https://www.zigbee2mqtt.io/devices/TS0052.html#tuya-ts0052)

boojew, 2023-12-18:
After looking, this problem was also documented in https://github.com/Koenkk/zigbee2mqtt/issues/17239, but the issue was closed.

wooooooooooook, 2024-01-11:
I too am experiencing the same problem. Whether I turn it off in ha or z2m, it stays on at 1%. When I look at the log, it is recorded as on right after it turns off.

wooooooooooook, 2024-01-12:
#19847 
this converter works!

upandabout, 2024-02-27:
> I too am experiencing the same problem. Whether I turn it off in ha or z2m, it stays on at 1%. When I look at the log, it is recorded as on right after it turns off.

Did you set the transition time under the Settings (specific) tab? If yes then try to delete this value. For me this made the FS-05R unit work again switching the light completely off.

peca89, 2024-03-11:
I can confirm that trying to set the transition value under Settings (Specific) tab breaks turning off completely. The workaround is to unset the transition value (delete the value from text box making it empty and click outside of it). 

I really really don't like transitions, though, and inability to disable it makes me unhappy about the device...

upandabout, 2024-03-18:
Yepp, I confirm it was the transition time that was the cause of the problem to me as well. There was a reasent update of the device, but I don't know if I need to kick the module from my device list and re-add it or if it just update itself? Perhaps the reconfigure button will do the trick, if so do I need to click "permit join" again before clicking reconfigure, logic tells me it's not needed to click  "permit join" but perhaps the reconfigure button is made for something completely different, I hav not ben able to find any meaning full info on the intended use of that button. Perhaps someone can tell me what is purpose is?

closed 2024-10-16 by github-actions[bot]