Koenkk/zigbee2mqtt #19935: [New device support]: Awow ms-c 4 TRV modelID: 'TS0601', manufacturerName:_TZE200_thbr5z34

kajapa, 2023-11-30:
### Link

https://www.amazon.de/dp/B08M3R3FVW/ref=cm_sw_r_cp_api_glt_i_HRTNR1KWZHZDBZTCRX3B

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0x842e14fffef23f56","nwkAddr":52401,"manufId":4098,"manufName":"_TZE200_thbr5z34","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"-��,\u0012.��,\u0012","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_thbr5z34","powerSource":3,"appVersion":64,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701373331286,"defaultSendRequestWhen":"immediate"}

### Comments

Hey! 
With converter below current temperature, target temperature and battery level are displayed correctly.
Changing to any mode often cause valve to switch to auto instead of opening or closing. 
This valve seems to be similar to: [_TZE200_zion52ef](https://github.com/Koenkk/zigbee2mqtt/issues/10850)
THANK YOU in advance for any help!

### External converter

```shell
const legacy = require('/config/zigbee2mqtt/lib/legacy');
const fz = {...require('/config/zigbee2mqtt/converters/fromZigbee'), legacy: require('/config/zigbee2mqtt/lib/legacy').fromZigbee};
const tz = {...require('/config/zigbee2mqtt/converters/toZigbee'), legacy: require('/config/zigbee2mqtt/lib/legacy').toZigbee};
const exposes = require('/config/zigbee2mqtt/lib/exposes');
const reporting = require('/config/zigbee2mqtt/lib/reporting');
const extend = require('/config/zigbee2mqtt/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('/config/zigbee2mqtt/lib/tuya');

const tuyaLocal = {
  dataPoints: {
  sh4Mode: 2,
  sh4HeatingSetpoint: 16,
  sh4LocalTemp: 24,
  sh4ChildLock: 30,
  sh4Battery: 34,
  sh4FaultCode: 45,
  sh4ComfortTemp: 101,
  sh4EcoTemp: 102,
  sh4VacationPeriod: 103,
  sh4TempCalibration: 104,
  sh4ScheduleTempOverride: 105,
  sh4RapidHeating: 106,
  sh4WindowStatus: 107,
  sh4Hibernate: 108,
  sh4ScheduleMon: 109,
  sh4ScheduleTue: 110,
  sh4ScheduleWed: 111,
  sh4ScheduleThu: 112,
  sh4ScheduleFri: 113,
  sh4ScheduleSat: 114,
  sh4ScheduleSun: 115,
  sh4OpenWindowTemp: 116,
  sh4OpenWindowTime: 117,
  sh4RapidHeatCntdownTimer: 118,
  sh4TempControl: 119,
  sh4RequestUpdate: 120,
  },
};
const fzLocal = {
  sh4_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
      const dpValue = tuya.firstDpValue(msg, meta, 'sh4_thermostat');
      const dp = dpValue.dp;
      const value = tuya.getDataValue(dpValue);
      function weeklySchedule(day, value) {
        // byte 0 - Day of Week (0~7 = Mon ~ Sun) <- redundant?
        // byte 1 - 1st period Temperature (1~59 = 0.5~29.5°C (0.5 step))
[... 289 lines cut ...]
  ],
  toZigbee: [
    tzLocal.sh4_thermostat_current_heating_setpoint,
    tzLocal.sh4_thermostat_comfort_temp_preset,
    tzLocal.sh4_thermostat_eco_temp_preset,
    tzLocal.sh4_thermostat_away,
    tzLocal.sh4_thermostat_mode,
    tzLocal.sh4_thermostat_child_lock,
    tzLocal.sh4_thermostat_calibration,
    tzLocal.sh4_thermostat_schedule_override_setpoint,
    tzLocal.sh4_thermostat_schedule,
    tzLocal.sh4_thermostat_get_data,
    
  ],
  onEvent: tuya.onEventSetLocalTime,
  exposes: [
    e.battery(), e.battery_low(), e.child_lock(),
    exposes.climate().withSetpoint('current_heating_setpoint', 0.5, 29.5, 0.5)
                     .withLocalTemperature()
                     .withSystemMode(['auto','heat','off'])
  ],
};

module.exports = device;
```

mak27arr, 2024-05-29:
Get error: z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT
Current version: 1.37.1-1

tureka, 2024-12-10:
I have 5 of them and unfinished quirk for ZHA

closed 2025-02-17 by github-actions[bot]