Koenkk/zigbee2mqtt #19947: Extension Support for Aqara T1 LED Stripe

ennilein, 2023-12-01:
### What happened?

After adding the Aqara T1 LED Stripe with 3 x 1 Meter extensions it works only in white mode over the complete length. When i switch to a color only the main string is illuminated.

### What did you expect to happen?

All extension should illuminated in all colors.

### How to reproduce it (minimal and precise)

I add the light stripe several times without any changes. I add it to the original aqara hub and it works fine.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus TI CC2652P

Xirtam2K, 2023-12-20:
I have the same problem. I bought 2 sets T1 with 2 extensions both. In withe there Ara no problems,  in color there is no light in the extensions 
 I tried different combinations and the problem stays the same. 
I use the aqara hub 3

NT1970, 2024-01-01:
Same issue here, 3 meter and 1 meter extension connected via the M2 hub. White works fine but colours don't extend to the extension. Tried reseting the strip with no success.

fabssor, 2024-01-01:
For me the length adjustment is working as expected with version 1.34.0. But with the new version (1.35.0) the color adjustment is removed completly.

closed 2024-07-30 by github-actions[bot]