Koenkk/zigbee2mqtt #19951: MQTT Publish Command Broken in 1.34.0-1 (Float passed as String)

julianfs, 2023-12-01:
### What happened?

When publishing a set command to a Hive receiver, the following command works in 1.33.2-1

```
- service: mqtt.publish
data:
  qos: 0
  retain: false
  payload: |-
    { 
      "system_mode":"heat",
      "temperature_setpoint_hold":"1",   
      "occupied_heating_setpoint":"18"
    }
  topic: zigbee2mqtt/climate.hive_receiver_ufh/set
```

When running the same command in 1.34.0-1 it fails with the log entry:
`Zigbee2MQTT:error 2023-12-01 12:04:38: Publish 'set' 'occupied_heating_setpoint' to 'climate.hive_receiver_ufh' failed: 'Error: 'occupied_heating_setpoint' is not a number, got string (18)'
`

The log entry suggests that the numeric value is now being passed as a string instead of float.

### What did you expect to happen?

Float passed as float.

### How to reproduce it (minimal and precise)

Call the service as above, reproduced every call.

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

latest

### Adapter

Conbee II

julianfs, 2023-12-01:
I should add, rolling back to 1.33.2-1 resolves the issue.

NathanVanCleave, 2023-12-02:
Same here. Trying to recall a scene errors out because it sees the scene number payload as a string in 1.34.0. Downgraded to 1.33.2 for now.

passibe15, 2023-12-02:
What happens if you change `"occupied_heating_setpoint":"18"` to `"occupied_heating_setpoint":18`? Is it still broken then?

roberthawdon, 2023-12-02:
> What happens if you change `"occupied_heating_setpoint":"18"` to `"occupied_heating_setpoint":18`? Is it still broken then?

That works fine for me. I am using the Proxy Thermostat workaround ( https://github.com/greeeny101/Hive-Thermostat-Zigbee2MQTT ) which was required to stop the Hive Thermostat from dropping to 1°C when attempting to change the temperature.

I removed the quotes from the trigger.payload in the automation on this line: https://github.com/greeeny101/Hive-Thermostat-Zigbee2MQTT/blob/f19901611bcd77ff17c99c5062e4ee3fa8e09ff4/SLR1/config_automation/hive.yaml#L19

Now the temperature is passed as a number as Zigbee2MQTT is expecting.

The real question is, is the proxy even required now? Has this update fixed the issue that the workaround was required for?

julianfs, 2023-12-02:
I'm sure I tried passing the parameter as 18 (and not "18") and still got the string error. Reading other logged similar issues I'm quite sure there's something amiss with the 1.34 release.

@roberthawdon the issue you describe with the thermostat returning to 1c is a 'feature' of Hive, as it expects a specific mqtt payload to work, so not an issue as such.

The normal 'climate' service calls won't work for Hive, you have to use the payload I mention in the 1st post above, sending both the "temperature_setpoint_hold" & "occupied_heating_setpoint" parameters to avoid that 1c niggle.

The payloads are nicely documented here [https://www.zigbee2mqtt.io/devices/SLR2b.html](url), but note that the boost 'emergency_heating' option doesn't work, as that's a non-standard setting.

julianfs, 2023-12-02:
In that case I will upgrade & try again. I wonder if it's because the payload contains two floats as strings. I was only correcting the second one as that's what was mentioned in the error return.

I've removed the quotes on & tested with 1.33.2 and it continues to work as expected.

fenty17, 2023-12-05:
Hi folks. Just to say that removing the quotes from the temp in my hive scripts and automations fixed this problem for me.

closed 2023-12-05 by Koenkk

dgthomson, 2024-01-03:
I think I have just been caught out by this after upgrading Z2M and having spent a couple of days troubleshooting. My payloads to a Hive SLR1b used strings instead of floats, so the 'on' command was being ignored. Maybe a documentation change is needed to highlight this change in behaviour.