Koenkk/zigbee2mqtt #19952: Aqara Thermostat E1 (SRTS-A01). Add hvac_action report to the converter

DaN66O, 2023-12-01:
### Is your feature request related to a problem? Please describe

This TRV does not know how to report hvac_action, because of this, it is difficult to work with it in Home Assistant.
Knowing temperature and current_temperature, is it possible to create an hvac_action variable in the converter based on the readings of the target temperature and the current temperature
and add it to the climate supported_features?

### Describe the solution you'd like

Based on the readings of the target temperature and the current temperature, create the hvac_action variable and add it to the climate supported_features.

### Describe alternatives you've considered

Based on the values of temperature and current_temperature, create the variable hvac_action in the Home Assistant and then create a new climat

### Additional context

no information available

closed 2024-06-29 by github-actions[bot]

niorko, 2024-10-26:
Is it even possible? Radiator valve could ask for heating sooner then the temperature drops or vice versia. Calculating hvac_action is not possible I guess, or is it?

DaN66O, 2024-10-28:
This is where they tried to solve it programmatically in the home assistant
https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1382697913
I thought it could be solved in the converter.

niorko, 2024-10-28:
> This is where they tried to solve it programmatically in the home assistant [#13993 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1382697913) I thought it could be solved in the converter.

`
hvac_action_template: >
      {% if state_attr('climate.real_trv', 'preset_mode') == 'manual' and state_attr('climate.real_trv', 'system_mode') == 'heat' and state_attr('climate.real_trv', 'temperature') > state_attr('climate.real_trv', 'current_temperature') + 1 %}
        heating
      {% else %}
        idle
      {% endif %}
`

His calculation is not right :-(. It determines hvac_action based on the current temperature and target temperature. If target tempersture is higher, then it is consideted as "heating".

I did similar approach and if you use BetterThermostat it does the same thing - but you can set threshold. It is really far from what I want, but could work.

DaN66O, 2024-10-28:
The whole point of my question was not to make new thermostats but to implement it in a standard converter.
p.s.
I don't use BetterThermostat for the same reason.

marcuscps, 2025-01-05:
I also think the logic is incorrect.
I believe `hvac_action` should be determined using hysteresis.

I took a stab on the problem from the HomeAssistant perspective [here](https://community.home-assistant.io/t/best-way-to-signal-aqara-trvs-are-calling-for-heat-to-a-heat-pump-ashp/821529/4).

To summarise, I believe it should work like this:

1 - Initially the `hvac_action` should be solely determined by `currect_temperature` and `temperature`:
* `heat` if `currect_temperature` < `temperature`
* `idle` otherwise

2 - From there, hysteresis kicks in.
* `heat` if `currect_temperature` < `temperature` - `hysteresis`
* 'idle` if `currect_temperature` >= `temperature` + `hysteresis`
* **unchanged** otherwise.

I'm reasonably sure the device implements a similar hysteresis logic internally. However it'd have to be confirmed experimentally.

The correct value for `hysteresis` also needs to be confirmed. [trullock](https://github.com/trullock) suggested it should be 1 on the "add support post" (as mentioned by @niorko ), but I'm not so sure about it. Some strange scenarios might rise in case it is wrong. E.g. If internally it is 0.5 and we use 1, the device will never reach the upper bound (1 degree above set point), and the sensor will be kinda stuck on `heat` pretty much indefinitely while internally it is actually `idle`. This can cause some very wasteful behaviour ($$). 

Ultimately, this parallel implementation of hysteresis can always conflict and cause competing behaviour and ends up resulting on "heat" forever. Aqara should expose the status directly from the device, but unfortunately they don't.

I'm travelling abroad for some time, but I'll try to test it once I'm back home. Ideally one should determine how many degrees above and below set temperature the valve opens/closes completely.

marcuscps, 2025-01-05:
Check results:

[image]

On the top, the original implementation without hysteresis.
On the center, the hysteresis implementation.
On the bottom the switch with 10 minutes delay.

Results are only relevant from around 23:30 forward, as before it wasn't the final solution. 
Notice how much energy is saved. And the warmth feeling is pretty much the same or slightly better with the hysteresis solution (temperature overshoots a bit).