Koenkk/zigbee2mqtt #19953: Moes BHT-002-GCLZB - TS0601_TZE200_aoclfnxz - Current heating setpoint returning to 0 after every change in v1.34.0-1

BassBoi, 2023-12-01:
### What happened?

After changing the setpoint (in program mode o or in manual) the current heating setpoint is returning to 0 degrees after the heating ends.

### What did you expect to happen?

Stay at the set temperature

### How to reproduce it (minimal and precise)

Update to 1.34.0-1, set heating so it will start heating. Wait till heating is done. heating returns to 0degress instead of setpoint

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

20230507

### Adapter

Sonoff dongle-P

julianfs, 2023-12-01:
Possibly the same as #19951 (float heating value being passed as string)

BassBoi, 2023-12-14:
So [https://github.com/Koenkk/zigbee2mqtt/issues/19951](url) was closed by my problem still exist.
I am not a power user and have no idea how and where change some values and strings... 
Will this be worked on or should i provide some information you need ?
Thank you.

julianfs, 2023-12-14:
I suspect the problem will be within your home assistant implementation, it's not a zigbee2mqtt issue.

Do you know how HA is using your Hive? Any automations or scripts you can find?

You're looking for yaml like this:
`{
   "system_mode_heat":"heat",
   "temperature_setpoint_hold_heat":"1",
   "occupied_heating_setpoint_heat":"20" // Replace with desired temperature. Between 5 and 32 C
}`

In the example above, you would need to remove the quotes around the numbers 1 & 20, which I think fixes the issue.

BassBoi, 2023-12-14:
Ok, i have figured something out, as my thermostat does not support changing the setpoint in 0,5 degrees due to manufacture limitations, it only happens (dropping to 0 degrees) when i set the temperature by 0.5 degrees.
In 1.33.2 my Moes BHT-002-GCZLB was not able to set the temperature by 0,5 degrees, it went by 1 degree - this is however possible in 1.34.0-1. 
There is no option in z2m to change (select) the step for _TZE200_aoclfnxz.
Maybe a device specific thing ? 
Can i somehow change it locally ?
Thank you for your patience.

julianfs, 2023-12-14:
I don't have that device, so don't know for sure, but looking in the herdsman converters file [(here)](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/moes.ts) shows a 'heatingStepSize' of 0.5 which would need changing to 1. 

@Koenkk can you modify? Looking at this discussion [here](https://github.com/Koenkk/zigbee2mqtt/discussions/6453#discussioncomment-2752075) would also show another confirmation that a step of 1c is required.

```
 {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ztvwu4nk'},
            {modelID: 'TS0601', manufacturerName: '_TZE204_5toc8efa'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_5toc8efa'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ye5jkfsb'},
            {modelID: 'TS0601', manufacturerName: '_TZE204_aoclfnxz'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_u9bfwha0'},
            {modelID: 'TS0601', manufacturerName: '_TZE204_u9bfwha0'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        fromZigbee: [legacy.fz.moes_thermostat],
        toZigbee: [legacy.tz.moes_thermostat_child_lock, legacy.tz.moes_thermostat_current_heating_setpoint, legacy.tz.moes_thermostat_mode,
            legacy.tz.moes_thermostat_standby, legacy.tz.moes_thermostat_sensor, legacy.tz.moes_thermostat_calibration,
            legacy.tz.moes_thermostat_deadzone_temperature, legacy.tz.moes_thermostat_max_temperature_limit,
            legacy.tz.moes_thermostat_min_temperature_limit, legacy.tz.moes_thermostat_program_schedule],
        whiteLabel: [
            tuya.whitelabel('Moes', 'BHT-002/BHT-006', 'Smart heating thermostat', ['_TZE204_aoclfnxz']),
        ],
        exposes: (device, options) => {
            const heatingStepSize = device?.manufacturerName === '_TZE204_5toc8efa' ? 0.5 : 1;
            return [e.linkquality(), e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(), e.min_temperature_limit(),
                e.climate().withSetpoint('current_heating_setpoint', 5, 35, heatingStepSize, ea.STATE_SET)
                    .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
                    .withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                    .withPreset(['hold', 'program']),
                e.temperature_sensor_select(['IN', 'AL', 'OU']),
                e.composite('program', 'program', ea.STATE_SET).withDescription('Time of day and setpoint to use when in program mode')
                    .withFeature(exposesLocal.hour('weekdays_p1_hour'))
                    .withFeature(exposesLocal.minute('weekdays_p1_minute'))
                    .withFeature(exposesLocal.program_temperature('weekdays_p1_temperature'))
                    .withFeature(exposesLocal.hour('weekdays_p2_hour'))
                    .withFeature(exposesLocal.minute('weekdays_p2_minute'))
                    .withFeature(exposesLocal.program_temperature('weekdays_p2_temperature'))
                    .withFeature(exposesLocal.hour('weekdays_p3_hour'))
                    .withFeature(exposesLocal.minute('weekdays_p3_minute'))
                    .withFeature(exposesLocal.program_temperature('weekdays_p3_temperature'))
                    .withFeature(exposesLocal.hour('weekdays_p4_hour'))
                    .withFeature(exposesLocal.minute('weekdays_p4_minute'))
                    .withFeature(exposesLocal.program_temperature('weekdays_p4_temperature'))
                    .withFeature(exposesLocal.hour('saturday_p1_hour'))
                    .withFeature(exposesLocal.minute('saturday_p1_minute'))
                    .withFeature(exposesLocal.program_temperature('saturday_p1_temperature'))
                    .withFeature(exposesLocal.hour('saturday_p2_hour'))
                    .withFeature(exposesLocal.minute('saturday_p2_minute'))
                    .withFeature(exposesLocal.program_temperature('saturday_p2_temperature'))
                    .withFeature(exposesLocal.hour('saturday_p3_hour'))
                    .withFeature(exposesLocal.minute('saturday_p3_minute'))
                    .withFeature(exposesLocal.program_temperature('saturday_p3_temperature'))
                    .withFeature(exposesLocal.hour('saturday_p4_hour'))
                    .withFeature(exposesLocal.minute('saturday_p4_minute'))
                    .withFeature(exposesLocal.program_temperature('saturday_p4_temperature'))
                    .withFeature(exposesLocal.hour('sunday_p1_hour'))
                    .withFeature(exposesLocal.minute('sunday_p1_minute'))
                    .withFeature(exposesLocal.program_temperature('sunday_p1_temperature'))
                    .withFeature(exposesLocal.hour('sunday_p2_hour'))
                    .withFeature(exposesLocal.minute('sunday_p2_minute'))
                    .withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
                    .withFeature(exposesLocal.hour('sunday_p3_hour'))
                    .withFeature(exposesLocal.minute('sunday_p3_minute'))
                    .withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
                    .withFeature(exposesLocal.hour('sunday_p4_hour'))
                    .withFeature(exposesLocal.minute('sunday_p4_minute'))
                    .withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
            ];
        },
        onEvent: tuya.onEventSetLocalTime,
    },

```

Koenkk, 2023-12-15:
@julianfs the step size is 1 for all except `_TZE204_5toc8efa` which uses 0.5

jvcsw, 2023-12-22:
Hi, same problem here. 

I explain my scenario here for help us:

It isn't an HA issue because happens with HA stopped.

When I set a new temperature with decimals (14.5º) from the thermostat card or from zigbee2mqtt UI, the card it goes to 0º at once.

[image]

When I change the temperature again without decimals, the thermostat card works fine.

[image]

The same occurs changing the values from z2m webUI (with HA started and stopped).

jvcsw, 2023-12-22:
Using dev-branch works fine. Thank you @Koenkk .

jvcsw, 2023-12-27:
Hello again @Koenkk, 

current dev branch returns always that the thermostat preset mode is "hold".
When sets the value to "program" the change is ignored.

Koenkk, 2023-12-27:
@jvcsw 

> When sets the value to "program" the change is ignored.

Is this a new bug or already present in e.g. z2m 1.33.1?

jvcsw, 2024-01-04:
I see now that in yesterday's dev version it works fine.

BassBoi, 2024-01-04:
I can confirm this has been fixed in stable release 1.35.0-1, thank you!

closed 2024-01-04 by Koenkk