Koenkk/zigbee2mqtt #19955: Support for Conbee III in Home Assistant zigbee2mqtt

thphilipp, 2023-12-01:
### Is your feature request related to a problem? Please describe

When will the support für zigbee2mqtt and home assistant with an Cobnbee III Stick come?

### Describe the solution you'd like

I dont know

### Describe alternatives you've considered

I dont know

### Additional context

I dont know

julianfs, 2023-12-01:
It's in the release notes of the latest version as being supported.

julianfs, 2023-12-01:
Sorry, I don't have a Conbee III so can't help. 

All I saw was that in the release notes 'Support for Conbee III' is listed under Features.
[](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.34.0)

julianfs, 2023-12-01:
[https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.34.0](url)

thphilipp, 2023-12-02:
Unfortunately, this is a mistake, this version is definitely not compatible with the HA Zigbee2MQtt version.You should pass this on to those responsible.Thanks.

sfo, 2023-12-08:
My ConBee III is working flawlessly under 1.34.0 with the following settings (according to [this guide](https://www.zigbee2mqtt.io/guide/adapters/#other)):
```yaml
serial:
  port: /dev/ttyUSB0
  adapter: deconz
  baudrate: 115200
```

cigas4, 2023-12-28:
Having problem with new ConBee 3 adapter. Zigbee2MQTT Addon fails to start.

Adapter: ConBee 3
ConBee 3 Firmware: deCONZ_ConBeeIII_0x264e0900 08-Dec-2023
Zigbee2mqtt addon version: 1.18.1-1
HA version: 2023.12.4

Addon logs:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[18:45:59] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[18:45:59] INFO: MQTT available, fetching server detail ...
[18:45:59] INFO: MQTT server settings not configured, trying to auto-discovering ...
[18:45:59] INFO: Configuring 'mqtt://core-mosquitto:1883' mqtt server
[18:46:00] INFO: Previous config file found, checking backup
[18:46:00] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[18:46:00] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[18:46:00] INFO: Handing over control to Zigbee2mqtt Core ...

> zigbee2mqtt@1.18.1 start /app
> node index.js

CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2023-12-28 18:46:11: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-12-28 18:46:11: Failed to start zigbee
Zigbee2MQTT:error 2023-12-28 18:46:11: Exiting...
Zigbee2MQTT:error 2023-12-28 18:46:11: undefined
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2023-12-28T16_46_11_791Z-debug.log
[18:46:11] INFO: Handing over control to Zigbee2mqtt Core ...
```

Addon Config:
```
data_path: /config/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  user: Z2M
  password: 12345
serial:
  port: /dev/serial/by-id/usb-dresden_elektronik_ConBee_III_DE03187432-if00-port0
  adapter: deconz
  baudrate: 115200
advanced:
  log_level: warn
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
```

xhousemd, 2024-02-05:
> My ConBee III is working flawlessly under 1.34.0 with the following settings (according to [this guide](https://www.zigbee2mqtt.io/guide/adapters/#other)):
> 
> ```yaml
> serial:
>   port: /dev/ttyUSB0
>   adapter: deconz
>   baudrate: 115200
> ```

Thats help me so much thank you !

asiar1993, 2024-07-02:
> > My ConBee III is working flawlessly under 1.34.0 with the following settings (according to [this guide](https://www.zigbee2mqtt.io/guide/adapters/#other)):
> > ```yaml
> > serial:
> >   port: /dev/ttyUSB0
> >   adapter: deconz
> >   baudrate: 115200
> > ```
> 
> Thats help me so much thank you !

Did you manage to resolve the connection issue?

xhousemd, 2024-07-02:
Tak wszystko działa

closed 2025-01-10 by github-actions[bot]