Koenkk/zigbee2mqtt #19957: Nodon SIN-4-1-20 error at zigbee2mqtt startup

chris-1243, 2023-12-01:
### What happened?

I have just updated Zigbee2mqtt to the latest version (1.34.0) and I have the following errors. It is something which was not present on the previous version.

The device seems to work some minutes after the boot.

Maybe the firmware of the sonoff dongle-P ?

### How to reproduce it (minimal and precise)

update from zigbee2mqtt 1.33.2 to 1.34.0 and Nodon module SIN-4-1-2perfectly working and already paired to the network

### Zigbee2MQTT version

1.34.0 (commit: aae7312)

### Adapter firmware version

20231112

### Adapter

Sonoff Dongle-P

### Debug log

The log et startup with zigbee-herdsman log as well

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13527072/log.txt)

Koenkk, 2023-12-02:
@AlexisPolegato could you check this? Looks like some versions don't support `nodonTransitionTime`

AlexisPolegato, 2023-12-02:
Ok, I've removed it here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6606

I will investigate this later

chris-1243, 2023-12-02:
Thanks for your support. Should I try the latest dev to see if the problem is solved already?

Koenkk, 2023-12-03:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/6606 , thanks @AlexisPolegato 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-12-03 by Koenkk