Koenkk/zigbee2mqtt #19960: Issue with chaing Operation Mode on Aqara switches

ant0nkr, 2023-12-01:
### What happened?

After updating zigbee2mqtt 1.34.0 all aqara switches started to return an error: Publish 'set' 'operation_mode' to 'switch.gym' failed: 'Error: undefined is not a object, got string ("decoupled")'
Same information appears in Log: Error 2023-12-01 16:05:30Publish 'set' 'operation_mode' to 'switch.gym' failed: 'Error: undefined is not a object, got string ("decoupled")'

### What did you expect to happen?

It supposed to change Switch mode to Decopled or Control Relay, depends on selected value.

### How to reproduce it (minimal and precise)

Go to Zigbee2MQTT UI, choose any Aqara switch change Opeation Mode to Decoupled or Control Relay. See the error.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

CC2652R

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13527644/log.txt)

helderfmf, 2023-12-01:
I got exactly the same environment and error.

Rumlik, 2023-12-01:
Same here. But adapter EZSP v8

ant0nkr, 2023-12-01:
> @ant0nkr How did you revert back to 1.33.2?

The only way to revert back is to restore from a backup, afaik

TB-SE, 2023-12-02:
Same here. Happened to notice it during a clean move to a new coordinator (worked in version 1.33.2).

neekul, 2023-12-02:
Same. Hope this can be fixed in a new release. I use the operation mode select in a number of automations 😞

Koenkk, 2023-12-02:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-12-02 by Koenkk

nickbock, 2023-12-07:
Hey,

I just switched to latest-dev docker image and the error stays the same. 

`Publish 'set' 'operation_mode' to 'Switch XXXX' failed: 'Error: undefined is not a object, got string ("decoupled")'`

Zigbee2MQTT version
1.34.0-dev commit: [f11eb24](https://github.com/Koenkk/zigbee2mqtt/commit/f11eb24)
Coordinator type
zStack3x0
Coordinator revision
20230507
Frontend version
0.6.148
Zigbee-herdsman-converters version
16.1.0
Zigbee-herdsman version
0.25.3

itrankolov, 2023-12-07:
I can also confirm that it is not working with WS-EUK01. Still gives the same error:
Publish 'set' 'operation_mode' to 'Corridor Bedrooms' failed: 'Error: undefined is not a object, got string ("decoupled")'

Koenkk, 2023-12-07:
I forgot to integrate the changes, please try again with the latest dev branch in 2 hours from now.

itrankolov, 2023-12-07:
It works now. Thanks!

Koenkk, 2023-12-08:
Next release = 1 January

Gaara81, 2023-12-08:
> Next release = 1 January

So we have to wait 20 days without our "smart" home not working at all? There is a way to fix it meanwhile?

jerrm, 2023-12-08:
How are you running z2m?

KMTsvetanov, 2023-12-09:
not working with smart switch: AQARA H1 EU

itrankolov, 2023-12-09:
> > Next release = 1 January
> 
> So we have to wait 20 days without our "smart" home not working at all? There is a way to fix it meanwhile?

Install the development release, change the mode and switch to regular branch.

Gaara81, 2023-12-09:
> > > Next release = 1 January
> > 
> > 
> > So we have to wait 20 days without our "smart" home not working at all? There is a way to fix it meanwhile?
> 
> Install the development release, change the mode and switch to regular branch.

Thanks for the answer. Could you guide me how to do that? Or any reliable place that explains it? 
I'm worried to lose my current configs.

Thanks again

Gaara81, 2023-12-09:
I figured out myself. It's pretty easy... Thanks my aqara h1 works again.

KMTsvetanov, 2023-12-09:
For anyone that is still not working I needed to update my firmware.

I use Zigbee 3.0 USB Dongle Plus. Can you point me to the link with the firmwares for it ?

Follow for more and a video on how to do it. This worked out for me. 
You you have to disassemble your Dongle. (Make changes only when you are sure that it's the same from the inside as well)

https://github.com/Koenkk/zigbee2mqtt/discussions/16573#discussioncomment-7807494

jsalvino1223, 2023-12-11:
It changed it to decoupled, but can't change anything now on the Aqara Switch.

2023-12-12 00:50:17Publish 'get' 'operation_mode' to '2F Master Bedroom Aqara Main Switch' failed: 'Error: Read 0x54ef4410008bceaf/1 aqaraOpple([512], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4062 - 1 - 11 - 64704 - 1 after 10000ms)'
2023-12-12 00:50:40Publish 'get' 'operation_mode' to '2F Master Bedroom Aqara Main Switch' failed: 'Error: Read 0x54ef4410008bceaf/1 aqaraOpple([512], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4062 - 1 - 12 - 64704 - 1 after 10000ms)'
2023-12-12 00:51:03Publish 'get' 'operation_mode' to '2F Master Bedroom Aqara Main Switch' failed: 'Error: Read 0x54ef4410008bceaf/1 aqaraOpple([512], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4062 - 1 - 13 - 64704 - 1 after 10000ms)'
2023-12-12 00:51:26Publish 'get' 'operation_mode' to '2F Master Bedroom Aqara Main Switch' failed: 'Error: Read 0x54ef4410008bceaf/1 aqaraOpple([512], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4062 - 1 - 14 - 64704 - 1 after 10000ms)'
2023-12-12 00:51:49Publish 'get' 'operation_mode' to '2F Master Bedroom Aqara Main Switch' failed: 'Error: Read 0x54ef4410008bceaf/1 aqaraOpple([512], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4062 - 1 - 15 - 64704 - 1 after 10000ms)'

aaalex474, 2023-12-13:
Could we just send particular payload in dev console to change the operation mode? Mode change isn't day to day operation. Once configured it rests. It would be simpler than to use the dev branch.
Could someone figure out the right payload?

HGRT846, 2023-12-14:
> Could we just send particular payload in dev console to change the operation mode? Mode change isn't day to day operation. Once configured it rests. It would be simpler than to use the dev branch. Could someone figure out the right payload?

Hello I am new to HA but I found the way to do it, firstly yo need to download [MQTT client](https://mosquitto.org/download/), then after installing you have to launch mosquitto_pub via cmd, with command:

`mosquitto_pub -h MOSQUITTO_BROKER_IP -p 1883 -u USERNAME -P PASSWORD -t zigbee2mqtt/IEEE ADDRESS OF YOUR DEVICE/set/operation_mode -m "control_relay OR decoupled"`

In this way you just send message straight to MQTT broker, so I assume it should work even without Zigbee2mqtt edge

sous-park, 2024-03-25:
If you switch the Fast mode to multi or reverse in zigbee2mqtt and immediately press any switch key 5 times, the mode will change and there will be no error.