Koenkk/zigbee2mqtt #19968: [Wrong device]: EARU VOLTAGE METER  WRONG RECOGNIZED

basalisco, 2023-12-01:
### Link

https://it.aliexpress.com/item/1005005999108883.html?spm=a2g0o.order_list.order_list_main.20.6cde3696FUAHuy&gatewayAdapt=glo2ita

### Model

EAYCBM-Z-2P

### Description

voltage meter

### Vendor

EARU

### Picture (link)

https://ae01.alicdn.com/kf/S1ec89b600c7045fb953929e00a0fef99K/1P-2P-TUYA-APP-Zigbee-WIFI-consumo-energetico-intelligente-kWh-Meter-interruttore-del-circuito-interruttore-del.jpg

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0xe0798dfffedde7cd","nwkAddr":7423,"manufId":4098,"manufName":"_TZ3000_zrm3oxsh","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,1026,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u001c�,\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_zrm3oxsh","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":67,"powerSource":1}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":233,"rmsCurrent":2876,"activePower":563,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,8315],"divisor":100,"multiplier":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2702}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ed6ec","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":60,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1941818122},"lastSeen":1701468303243,"defaultSendRequestWhen":"immediate"}

### Notes

this device is recognized as [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)

andreas-glaser, 2024-03-25:
I'm having the same issue

andreas-glaser, 2024-03-25:
i've got the same issue. my breaker is recognized as

TS011F  /  _TZ3000_zrm3oxsh
https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html

piavia, 2024-03-30:
@Koenkk
I think I found out the data of this device _TZ3000_zrm3oxsh,
could you look at the link
https://github.com/sprut/Hub/issues/2977

Address: F4B3B1FFFE6FE3F0/7EFE

Structure
Endpoint 1 (ON_OFF_PLUG_IN_UNIT)
Input
0000_Basic
0000_ZCLVersion: 3 [UNSIGNED_8_BIT_INTEGER]
0001_ApplicationVersion: 67 [UNSIGNED_8_BIT_INTEGER]
0002_StackVersion: 0 [UNSIGNED_8_BIT_INTEGER]
0003_HWVersion: 1 [UNSIGNED_8_BIT_INTEGER]
0004_ManufacturerName: _TZ3000_zrm3oxsh [CHARACTER_STRING]
0005_ModelIdentifier: TS011F [CHARACTER_STRING]
0006_DateCode: [CHARACTER_STRING]
0007_PowerSource: 1 [ENUMERATION_8_BIT]
FFDE_Custom: 11 [UNSIGNED_8_BIT_INTEGER]
FFDF_Custom: �IA-� [CHARACTER_STRING]
FFE2_Custom: 31 [UNSIGNED_8_BIT_INTEGER]
FFE4_Custom: 0 [UNSIGNED_8_BIT_INTEGER]
FFFD_Custom: 2 [UNSIGNED_16_BIT_INTEGER]
FFFE_Custom: 0 [ENUMERATION_8_BIT]
E000_ManufacturerSpecific
D001_Custom: ByteArray [value=00 06] [ORDERED_SEQUENCE_ARRAY]
D002_Custom: ByteArray [value=00 0A] [ORDERED_SEQUENCE_ARRAY]
D003_Custom: AAAA [CHARACTER_STRING]
D004_Custom: 0 [UNSIGNED_16_BIT_INTEGER]
D005_Custom: 0 [UNSIGNED_16_BIT_INTEGER]
E001_ManufacturerSpecific
D000_TuyaUnknown: 0 [UNSIGNED_8_BIT_INTEGER]
D001_TuyaUnknown: 0 [BITMAP_32_BIT]
D002_TuyaUnknown: 0 [UNSIGNED_32_BIT_INTEGER]
D003_TuyaUnknown: 0 [UNSIGNED_32_BIT_INTEGER]
D004_TuyaUnknown: 0 [UNSIGNED_32_BIT_INTEGER]
D005_TuyaUnknown: 0 [UNSIGNED_32_BIT_INTEGER]
D010_RestartStatus: 2 [ENUMERATION_8_BIT]
D020_Custom: 0 [ENUMERATION_8_BIT]
D030_ButtonType: 0 [ENUMERATION_8_BIT]
0702_Metering
0000_CurrentSummationDelivered: 0 [UNSIGNED_48_BIT_INTEGER]
0200_Status: 0 [BITMAP_8_BIT]
0300_UnitOfMeasure: 0 [ENUMERATION_8_BIT]
0303_SummationFormatting: 0 [BITMAP_8_BIT]
0306_MeteringDeviceType: 0 [BITMAP_8_BIT]
FFFD_Custom: 1 [UNSIGNED_16_BIT_INTEGER]
0402_TemperatureMeasurement
0000_MeasuredValue: 2726 [SIGNED_16_BIT_INTEGER]
FFFD_Custom: 1 [UNSIGNED_16_BIT_INTEGER]
0003_Identify
0000_IdentifyTime: 175 [UNSIGNED_16_BIT_INTEGER]
FFFD_Custom: 1 [UNSIGNED_16_BIT_INTEGER]
0004_Groups
0000_NameSupport: 0 [BITMAP_8_BIT]
FFFD_Custom: 2 [UNSIGNED_16_BIT_INTEGER]
0B04_ElectricalMeasurement
0505_RMSVoltage: 225 [UNSIGNED_16_BIT_INTEGER]
0508_RMSCurrent: 0 [UNSIGNED_16_BIT_INTEGER]
050B_ActivePower: 0 [SIGNED_16_BIT_INTEGER]
FFFD_Custom: 1 [UNSIGNED_16_BIT_INTEGER]
0005_Scenes
0000_SceneCount: 0 [UNSIGNED_8_BIT_INTEGER]
0001_CurrentScene: 0 [UNSIGNED_8_BIT_INTEGER]
0002_CurrentGroup: 0 [UNSIGNED_16_BIT_INTEGER]
0003_SceneValid: false [BOOLEAN]
0004_NameSupport: 0 [BITMAP_8_BIT]
FFFD_Custom: 2 [UNSIGNED_16_BIT_INTEGER]
0006_OnOff
0000_OnOff: false [BOOLEAN]
4001_OnTime: 0 [UNSIGNED_16_BIT_INTEGER]
4002_OffWaitTime: 0 [UNSIGNED_16_BIT_INTEGER]
5000_Custom: 1 [ENUMERATION_8_BIT]
8000_MultipleClick: false [UNSIGNED_16_BIT_INTEGER]
8001_IndicatorMode: 1 [ENUMERATION_8_BIT]
8002_RestartStatus: 2 [ENUMERATION_8_BIT]
FFFD_Custom: 2 [UNSIGNED_16_BIT_INTEGER]
Output
0019_OtaUpgrade
0000_UpgradeServerID: FFFFFFFFFFFFFFFF [IEEE_ADDRESS]
0001_FileOffset: -1 [UNSIGNED_32_BIT_INTEGER]
0006_ImageUpgradeStatus: 0 [ENUMERATION_8_BIT]
FFFD_Custom: 3 [UNSIGNED_16_BIT_INTEGER]
000A_Time
FFFD_Custom: 1 [UNSIGNED_16_BIT_INTEGER]
Endpoint 242 (ZGP_PROXY_BASIC)
Output
0021_GreenPower

Koenkk, 2024-03-31:
Is the picture/description just wrong or are some functionalities like power measurements not working?

andreas-glaser, 2024-03-31:
> Is the picture/description just wrong or are some functionalities like power measurements not working?

@Koenkk The exposed functionality is limited to voltage/amperage/power monitoring, and the over/under voltage and other features are missing.

Koenkk, 2024-04-01:
Do you expect anything more besides that? For the over/under voltage you probably need to reverse engineer how this work by sniffing traffic of the TuYa hub https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

kap-Gun, 2024-04-15:
> Do you expect anything more besides that? For the over/under voltage you probably need to reverse engineer how this work by sniffing traffic of the TuYa hub https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Dumb question, can it be paired with a Tuya gateway to set the over/under voltage then pair it back to Z2M?
I have bought a sniffer and will see what information I can get from the breaker and the tuya gateway.

closed 2024-11-13 by github-actions[bot]