Koenkk/zigbee2mqtt #19971: Ikea E2123 not firing most dots button events anymore

fthiery, 2023-12-02:
### What happened?

Since upgrading from 1.33.2 to 1.34.0 (using hassio addon), at least the following events of my Ikea E2123 (running firmware 1.0.35) are not fired anymore:
- dots_1_short_release
- dots_1_long_release
- dots_2_short_release
- dots_2_long_release

Only dots_1_double_press and dots_2_double_press are still working.

I did delete/re-pair the device multiple times (note that it won't delete without forcing), without success.

Rolling back to 1.33.2 makes these actions work again

### What did you expect to happen?

All documented events at https://www.zigbee2mqtt.io/devices/E2123.html#ikea-e2123 work

### How to reproduce it (minimal and precise)

Press dots 1 and dots 2 buttons (short or long) <-- no action is being reported in Z2M

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Sonoff P (zStack3x0)

purpleturtle1988, 2023-12-02:
I have the same problem. The switch was also added, but without any improvement

ASNNetworks, 2023-12-02:
Only buttons are play/pause, skip track, previous track, volume up, volume down.

Koenkk, 2023-12-02:
Will be fixed in https://github.com/Koenkk/zigbee-herdsman/pull/813

fthiery, 2023-12-03:
Awesome, thanks koenkk

HFlor, 2023-12-30:
Hello,

when will the problem be resolved?

NPohl, 2023-12-30:
Hi, just bought 2 SYMFONISK today and noticed the same Problem. Both came with Firmware version 1.0.35.
Glad this was already reported and fixed,
Thanks everyone!

HFlor, 2023-12-30:
How is it fixed? I have zibee2mqtt version 1.34.0 and the dot keys are missing there.

NPohl, 2023-12-30:
If I understand https://github.com/Koenkk/zigbee2mqtt/discussions/20003#discussioncomment-7779886 correctly it's fixed in the dev branch and will be part of the next release.

Koenkk, 2023-12-30:
Will be fixed in the 1 January Z2M release.

closed 2023-12-30 by Koenkk