Koenkk/zigbee2mqtt #19973: [Wrong device]: Connecte Smart Power Socket 16A

GD-Dal, 2023-12-02:
### Link

https://www.connecte.no/stikkontakt-flush

### Model

45 009 90

### Description

Smart Power Socket 16A

### Vendor

Connecte

### Picture (link)

https://static.wixstatic.com/media/bb46bf_cad46057a0d5432cb1640da8bac8d031~mv2.png/v1/fill/w_672,h_662,al_c,q_90,usm_0.66_1.00_0.01,enc_auto/StikkontaktILavtbyggende_f_500px-72dpi%20test.png

### Database entry

{"id":40,"type":"Router","ieeeAddr":"0x84ba20fffe2aa9f4","nwkAddr":59769,"manufId":4098,"manufName":"_TZ3000_fqoynhku","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,5,4,6,1794,2820,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_fqoynhku","powerSource":1,"zclVersion":3,"appVersion":97,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,5]}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":243,"activePower":0,"rmsCurrent":0}},"genOnOff":{"attributes":{"onOff":0,"tuyaBacklightMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff060c7e","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff060c7e","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff060c7e","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":97,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701515028154,"defaultSendRequestWhen":"immediate"}

### Notes

It gets wrongly detected as [Tuya TS0121_plug](https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug)

The Zigbee2MQTT logs also gives this error:
`Publish 'set' 'power_outage_memory' to '0x84ba20fffe2aa9f4' failed: 'Error: Write 0x84ba20fffe2aa9f4/1 genOnOff({"moesStartUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

And the Home Assistant log gives this error:
`Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:129
Integration: MQTT (documentation, issues)
First occurred: 12:03:44 PM (20 occurrences)
Last logged: 12:12:49 PM`

Thanks

Pythm, 2023-12-10:
I can confirm that I also get this error, but only on about half of my Connecte devices. The others are able to set the "power_outtage_memory" attribute.

EirikFinvold, 2023-12-15:
I did that as well on some of my devices. After upgrading the firmware (all mine now have appVersion=115) it works fine.

Pythm, 2023-12-15:
> I did that as well on some of my devices. After upgrading the firmware (all mine now have appVersion=115) it works fine.

How did you update firmware? Futurehome app or did you manage to find a firmware and update via Zigbee2MQTT?

EirikFinvold, 2023-12-15:
I used a Tuya hub. Since it's possible through the Tuya hub I guess it should be possible via z2m as well, but I haven't tried.
One thing I had to to in z2m though was to fix the reporting on four of my devices. They spammed the zigbee network with power updates several times every second!

closed 2024-07-13 by github-actions[bot]

cortado, 2025-07-12:
still same problem. Any suggestion on how I can change the zigbee model, so it works?