Koenkk/zigbee2mqtt #19975: [New device support]: Namron Stove Guard

GD-Dal, 2023-12-02:
### Link

https://www.namron.com/products/namron-zigbee-komfyrvakt-med-veggsensor-hvit/

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c94537","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}

### Comments

The zigbee functionality in this the device is described at page 9 in this [link](https://www.elektroimportoren.no/docs/lib/1402793-Brukerveiledning-5.pdf)

I have started to create a converter, but hit the wall

In Z2M, it looks like this:
[image]

The temperature varies. Often down to -300 C.
The link quality does seem way to low.
And the logs say the battery reports status every 5 seconds, which I would think is totally unnecessary.

Help is appreciated 

### External converter

```shell
const definition = {
    zigbeeModel: ['1402790', '1402791', '1402792', '1402793', '1402794', '1402795', '1402796', '1402797', '1402798'],
    model: '140279X',
    vendor: 'Namron',
    description: 'Namron Stove Guard',
    fromZigbee: [fz.on_off, fz.battery, fz.temperature, fz.metering, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    exposes: [e.temperature(), e.battery(), e.switch(), e.power(), e.energy()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(10);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genOnOff', 'haElectricalMeasurement', 'genPowerCfg', 'msTemperatureMeasurement']);
            await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acCurrentMultiplier', 'acCurrentDivisor']);
            await reporting.onOff(endpoint);
            await reporting.temperature(endpoint);
            await reporting.activePower(endpoint);
	    await reporting.batteryPercentageRemaining(endpoint);
        },
};

module.exports = definition;
```

adamchengtkc, 2024-03-16:
I have the newer version that shows as Brun Holding Power Control Unit. It works for me by changing
```
    zigbeeModel: ['Power Control Unit'],
    model: 'Power Control Unit',
    vendor: 'Brun Holding',
```

GD-Dal, 2024-03-27:
A year has passed since I first reported this in. 
Pretty much given up hope now.
Too bad I'm not able to add it myself

Koenkk, 2024-03-28:
Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

adamchengtkc, 2024-04-20:
With generated external converter
`Exception while calling fromZigbee converter: Definition 'Power Control Unit' has not endpoint defined}`

Generated  converter
`const {onOff, electricityMeter, batteryPercentage, temperature, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Power Control Unit'],
    model: 'Power Control Unit',
    vendor: 'Brun Holding',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), electricityMeter(), batteryPercentage(), temperature({"endpointNames":["1"]}), identify()],
    meta: {},
};

module.exports = definition;`

GD-Dal, 2024-05-16:
> The config provided by @GD-Dal works with no issue

It does? Does not work for me :)

GD-Dal, 2024-05-16:
> Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

Have tried.
It gives me

`const {deviceEndpoints, identify, onOff, electricityMeter, battery, temperature} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['1402790'],
    model: '1402790',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"10":10}}), identify(), onOff({"powerOnBehavior":false}), electricityMeter(), battery(), temperature()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;`

But the errors are still there and the device is listed as unsupported

GD-Dal, 2024-08-15:
Any news on this? Anyone got it working?
the generated external converter does not work.
I see errors like this
```
z2m: Publish 'get' 'voltage' to '0xf4ce3667f33e156d' failed: 'Error: ZCL command 0xf4ce3667f33e156d/1 haElectricalMeasurement.read(["rmsVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
z2m: Publish 'get' 'current' to '0xf4ce3667f33e156d' failed: 'Error: ZCL command 0xf4ce3667f33e156d/1 haElectricalMeasurement.read(["rmsCurrent"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

```
And the temperature says -327.68°C

Thanks

adamchengtkc, 2024-08-22:
I got a new replacement and it doesn't work with the config I posted earlier. Looks like a different firmware.
As for -327.68C, it is a normal behavior when the cook top is off. It will report correct temperature when you switch on the cook top

closed 2025-01-10 by github-actions[bot]