Koenkk/zigbee2mqtt #19976: Zigbee not loading after update tolatest release

Trebor87, 2023-12-02:
### What happened?

I updated  to 1.34.0 and after update it refused to start

I run my Zigbee using Channel 25, so I had to modify the channel back to 11 using /homeassistant/zigbee2mqtt/configuration.yaml for it to start correctly

Is it me having a wrong setting somewhere or a bug ?

I normally change the channel settings in the Zigbee / advanced page or via the Yaml page

### How to reproduce it (minimal and precise)

install update when not in channel 11

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20231030.2

### Adapter

sonoff D

Trebor87, 2023-12-02:
managed to get the log if it helps

info  2023-12-01 23:04:02: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-01.23-04-02' filename: log.txt
info  2023-12-01 23:04:02: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
info  2023-12-01 23:04:02: Starting zigbee-herdsman (0.25.0)
error 2023-12-01 23:04:02: Configuration is not consistent with adapter state/backup!
error 2023-12-01 23:04:02: - PAN ID: configured=63585, adapter=63585
error 2023-12-01 23:04:02: - Extended PAN ID: configured=5228b8c2e1d20545, adapter=5228b8c2e1d20545
error 2023-12-01 23:04:02: - Network Key: configured=076b4cb260950782fff573a1ee92bfc5, adapter=076b4cb260950782fff573a1ee92bfc5
error 2023-12-01 23:04:02: - Channel List: configured=25, adapter=11
error 2023-12-01 23:04:02: Please update configuration to prevent further issues.
error 2023-12-01 23:04:02: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
error 2023-12-01 23:04:02: Re-commissioning your network will require re-pairing of all devices!
error 2023-12-01 23:04:02: Error while starting zigbee-herdsman
error 2023-12-01 23:04:02: Failed to start zigbee
error 2023-12-01 23:04:02: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-12-01 23:04:02: Exiting...
error 2023-12-01 23:04:02: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

andrejsoucek, 2023-12-02:
Might be connected to a wrong version displayed in update information. Latest version is shown as:
1.99.0-1
Updated Zigbee2MQTT to version ``

Trebor87, 2023-12-03:
So do you mean that after changing the channel to x you need to restart the dongle using the red restart as shown in the image? 
[image]

KennoPa, 2023-12-03:
> > `error 2023-12-01 23:04:02: - Channel List: configured=25, adapter=11`
> 
> When you changed the channel you didn't restart Z2M, so it was still using channel 11. Now you've restarted it's throwing an error as a result.

Actually I did. I run Z2M with default, changed channel in ui, restarted and got this error.
Then went to yaml, changed value back to 11, started Z2M and all started well. Then did stop Z2M and changed channel in yaml to 15, started and got the error again. Then changed back to 11, started and again all okay.

Where did I made mistake?

DubhAd, 2023-12-04:
> My instance fails to start too, since updating to the latest version. Please fix ASAP, this is a serious issue.

What's in _your_ log about the problem? This thread was opened by somebody who's adapter was set to a different channel than in their config.

closed 2024-07-02 by github-actions[bot]