Koenkk/zigbee2mqtt #19977: 1.34 : MQTT payload contains power value even if no consumption Nous A1Z

Skuair, 2023-12-02:
### What happened?

Since version 1.34, most of the time when the consumption is null on the smart plug Nous A1Z, the MQTT payload contains the last active value for the power, whereas it is supposed to be 0.

In logs I can see the ZigBee message received that has 0 as activePower, but just after MQTT message is sent with the last non-null value in power.

Debug 2023-12-02 13:57:28 Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID 0
Info 2023-12-02 13:57:28 MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":5.63,"indicator_mode":"on/off","last_seen":"2023-12-02T12:57:28.783Z","linkquality":112,"power":74,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":222}'

Here MQTT has 74 as power but ZigBee message has 0. 74 is the last value sent when it was not 0.

### What did you expect to happen?

I expect that power is 0 int he MQTT payload.

### How to reproduce it (minimal and precise)

Use Nous A1Z smart plug and check when there is no consumption.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle-Plus E

buurman, 2023-12-07:
Same issue with my NOUS AZ1 since 1.34, after turning off devices and/or turning off the plug the last power value before turning it off remains. 

Zigbee model: TS011F
Zigbee manufacturer: _TZ3000_2putqrmw)
Adapter: Sonoff Dongle-Plus P

Koenkk, 2023-12-08:
Could you provide the debug log when power should change to 0?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Skuair, 2023-12-08:
I switched to debug mode for Z2M and for Zigbee-herdsman. Here are the log just right before power drops to 0 and after.

```
2023-12-08T16:03:54.211Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,5,0): 1556a1a9512a884b7e
2023-12-08T16:03:54.211Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-12-08T16:03:54.212Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-12-08T16:03:54.213Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1480010500
2023-12-08T16:03:54.213Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2023-12-08T16:03:54.214Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success
2023-12-08T16:04:04.211Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2023-12-08T16:04:04.211Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2023-12-08T16:04:04.212Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2023-12-08T16:04:04.212Z zigbee-herdsman:adapter:ezsp:uart --> DATA (5,2,0): 1500010500
2023-12-08T16:04:04.213Z zigbee-herdsman:adapter:ezsp:uart --> [525721a9512a5d737e]
2023-12-08T16:04:04.214Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
2023-12-08T16:04:04.219Z zigbee-herdsman:adapter:ezsp:uart <-- [2657a1a9512ac3767e]
2023-12-08T16:04:04.219Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,6,0): 2657a1a9512ac3767e
2023-12-08T16:04:04.220Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2023-12-08T16:04:04.220Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2023-12-08T16:04:04.221Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1580010500
2023-12-08T16:04:04.221Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2023-12-08T16:04:04.222Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success
Zigbee2MQTT:debug 2023-12-08 17:04:10: Saving state to file /config/zigbee2mqtt/state.json
2023-12-08T16:04:14.215Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2023-12-08T16:04:14.216Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2023-12-08T16:04:14.216Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2023-12-08T16:04:14.218Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,3,0): 1600010500
2023-12-08T16:04:14.219Z zigbee-herdsman:adapter:ezsp:uart --> [635421a9512ad98d7e]
2023-12-08T16:04:14.220Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2023-12-08T16:04:14.225Z zigbee-herdsman:adapter:ezsp:uart <-- [3754a1a9512a72807e]
2023-12-08T16:04:14.226Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,7,0): 3754a1a9512a72807e
2023-12-08T16:04:14.226Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2023-12-08T16:04:14.227Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2023-12-08T16:04:14.228Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1680010500
2023-12-08T16:04:14.229Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
2023-12-08T16:04:14.229Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2023-12-08T16:04:14.798Z zigbee-herdsman:adapter:ezsp:uart <-- [4754b1a97d312a15b658904124ab1593499c16b76b15939874f5de8183f47b1edfeacfc2df7e]
2023-12-08T16:04:14.799Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,7,0): 4754b1a9112a15b658904124ab1593499c16b76b15939874f5de8183f47b1edfeacfc2df7e
2023-12-08T16:04:14.799Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2023-12-08T16:04:14.799Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2023-12-08T16:04:14.801Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1690014500000401040b0101400100005890c0f85dffff0818e20a080521780102
2023-12-08T16:04:14.802Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":88},"lastHopLqi":144,"lastHopRssi":-64,"sender":24056,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,226,10,8,5,33,120,1]}}
2023-12-08T16:04:14.803Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":88},"lqi":144,"rssi":-64,"sender":24056,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,226,10,8,5,33,120,1]}}
2023-12-08T16:04:14.814Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":226,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":376}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":24056,"endpoint":1,"linkquality":144,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2023-12-08 17:04:14: Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":376}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-12-08 17:04:14: MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0.38,"energy":9.7,"indicator_mode":"on/off","last_seen":"2023-12-08T16:04:14.815Z","linkquality":144,"power":76,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage" [... cut ...]
Zigbee2MQTT:info  2023-12-08 17:04:36: MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":9.7,"indicator_mode":"on/off","last_seen":"2023-12-08T16:04:36.817Z","linkquality":152,"power":76,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":223}'
```

You can see when around this line when it drops to 0:
```Zigbee2MQTT:debug 2023-12-08 17:04:15: Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0```

Let me know if you want other logs.

Koenkk, 2023-12-08:
can you provide a screenshot of the z2m frontend -> device -> reporting tab?

Skuair, 2023-12-08:
Here are the settings:
[image]

(By the way, I increased intervals and removed some of reports (already done in 1.33) that are not relevant for me but still getting some related reports, but it is another topic, you can check later my question https://github.com/Koenkk/zigbee2mqtt/discussions/19909)

Koenkk, 2023-12-09:
Please reduce the min rep interval to 5

buurman, 2023-12-09:
I updated to 1.34 again and tried a min rep interval of 5, that does seem to work better. But I really don't need an update every 5 seconds of most devices so what is going wrong here? In 1.33.2 there was no problem with a min rep interval of 300.

jaroslawp, 2023-12-10:
I had same problem with 1.33.4 with default min rep intervals set to 5, downgraded to 1.33.2 which worked OK previously, then upgraded to 1.33.4 ... and that fixed it for me .. (using Home Assistant add-on)

Skuair, 2023-12-10:
Like @buurman I don't need frequent reports, but yes it seemed to work with that 5s interval in a quick test.
The question is : is it a bug or is it a new behavior of this version fit this device?

Koenkk, 2023-12-10:
Can you check if the 5 seconds reports are fixed with this [external converter](https://gist.github.com/Koenkk/7252623a5e31bf66accdad4fcc4640dc)

With this converter it should only report within 5 seconds when the change is bigger than: 5W or 0.05 A or 5V

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

buurman, 2023-12-10:
I'll rather stay on 1.33.2 than mitigate with a workaround. Like @Skuair said, what causes this behavior in 1.34 and can't this be reverted?

Skuair, 2023-12-11:
I tried your converter but I got na error at z2m startup:
```
/app/data/extension/externally-loaded.js:56
    ota: ota.zigbeeOTA,
         ^

ReferenceError: ota is not defined
    at /app/data/extension/externally-loaded.js:56:10
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:154:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:161:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:171:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

And not sure to understand, with this converter I have to keep 5 seconds in the report settings or I can put back 300 I had before ?

Koenkk, 2023-12-11:
Whoops, updated the converter. You should keep 5 seconds but it should not report every 5 seconds unless the the power change is bigger than 5W.

Skuair, 2023-12-11:
Updated the converter, but some errors.

First, at startup, but it seems not break the starting:
```
[20:19:18] INFO: Preparing to start...
[20:19:19] INFO: Socat not enabled
[20:19:22] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-12-11 20:19:36: Logging to console only'
Zigbee2MQTT:info  2023-12-11 20:19:36: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-11 20:19:37: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:info  2023-12-11 20:19:40: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-12-11 20:19:40: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:info  2023-12-11 20:19:40: Currently 1 devices are joined:
Zigbee2MQTT:info  2023-12-11 20:19:40: Congélateur (0xa4c1380d39ec562a): A1Z - Nous Smart plug (with power monitoring) (Router)
Zigbee2MQTT:info  2023-12-11 20:19:40: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2023-12-11 20:19:41: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:info  2023-12-11 20:19:41: Connected to MQTT server
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'zigbee2mqtt/Congélateur/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-12-11 20:19:41: Started frontend on port 8099
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'homeassistant/switch/0xa4c1380d39ec562a/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Congélateur/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Congélateur/set","device":{"identifiers":["zigbee2mqtt_0xa4c1380d39ec562a"],"manufacturer":"Nous","model":"Smart plug (with power monitoring) (A1Z)","name":"Congélateur"},"name":null,"object_id":"congélateur","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Congélateur","unique_id":"0xa4c1380d39ec562a_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'homeassistant/sensor/0xa4c1380d39ec562a/power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Congélateur/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380d39ec562a"],"manufacturer":"Nous","model":"Smart plug (with power monitoring) (A1Z)","name":"Congélateur"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:power-settings","name":"Power outage memory","object_id":"congélateur_power_outage_memory","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Congélateur","unique_id":"0xa4c1380d39ec562a_power_outage_memory_zigbee2mqtt","value_template":"{{ value_json.power_outage_memory }}"}'
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'homeassistant/select/0xa4c1380d39ec562a/power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Congélateur/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Congélateur/set/power_outage_memory","device":{"identifiers":["zigbee2mqtt_0xa4c1380d39ec562a"],"manufacturer":"Nous","model":"Smart plug (with power monitoring) (A1Z)","name":"Congélateur"},"entity_category":"config","icon":"mdi:power-settings","name":"Power outage memory","object_id":"congélateur_power_outage_memory","options":["on","off","restore"],"origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Congélateur","unique_id":"0xa4c1380d39ec562a_power_outage_memory_zigbee2mqtt","value_template":"{{ value_json.power_outage_memory }}"}'
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'homeassistant/sensor/0xa4c1380d39ec562a/indicator_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Congélateur/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380d39ec562a"],"manufacturer":"Nous","model":"Smart plug (with power monitoring) (A1Z)","name":"Congélateur"},"enabled_by_default":false,"name":"Indicator mode","object_id":"congélateur_indicator_mode","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Congélateur","unique_id":"0xa4c1380d39ec562a_indicator_mode_zigbee2mqtt","value_template":"{{ value_json.indicator_mode }}"}'
Zigbee2MQTT:info  2023-12-11 20:19:41: MQTT publish: topic 'homeassistant/select/0xa4c1380d39ec562a/indicator_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Congélateur/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Congélateur/set/indicator_mode","device":{"identifiers":["zigbee2mqtt_0xa4c1380d39ec562a"],"manufacturer":"Nous","mod [... cut ...]
Zigbee2MQTT:error 2023-12-11 20:24:52: Exception while calling fromZigbee converter: globalStore is not defined}
Zigbee2MQTT:debug 2023-12-11 20:24:52: ReferenceError: globalStore is not defined
    at Object.convert (/app/data/extension/externally-loaded.js:24:38)
Zigbee2MQTT:info  2023-12-11 20:24:52: MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":11.59,"indicator_mode":"on/off","last_seen":"2023-12-11T19:24:52.381Z","linkquality":144,"power":74,"power_outage_memory":"restore","state":"ON","voltage":225}'
```

Koenkk, 2023-12-12:
Updated: https://gist.github.com/Koenkk/7252623a5e31bf66accdad4fcc4640dc

Skuair, 2023-12-12:
Same error about the globalStore var after received a power message:

```
Zigbee2MQTT:debug 2023-12-12 22:25:20: Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":225}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-12-12 22:25:20: Configuring 'Congélateur'
Zigbee2MQTT:error 2023-12-12 22:25:20: Exception while calling fromZigbee converter: globalStore is not defined}
Zigbee2MQTT:debug 2023-12-12 22:25:20: ReferenceError: globalStore is not defined
    at Object.convert (/app/data/extension/externally-loaded.js:24:38)
```

Koenkk, 2023-12-13:
Updated: https://gist.github.com/Koenkk/7252623a5e31bf66accdad4fcc4640dc again

Skuair, 2023-12-14:
So, no more errors when using the converter.
I checked with a report set to 5 seconds, it seems ok now, here is an example of log:
```
Debug 2023-12-14 08:47:36Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
Info 2023-12-14 08:47:36MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":13.24,"indicator_mode":"on/off","last_seen":"2023-12-14T07:47:42.888Z","linkquality":140,"power":5,"power_outage_memory":"restore","state":"ON","voltage":221}'
Debug 2023-12-14 08:48:34Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":221}' from endpoint 1 with groupID 0
Info 2023-12-14 08:48:34MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":13.24,"indicator_mode":"on/off","last_seen":"2023-12-14T07:48:40.504Z","linkquality":136,"power":5,"power_outage_memory":"restore","state":"ON","voltage":221}'
Debug 2023-12-14 08:48:34Received Zigbee message from 'Congélateur', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1324]}' from endpoint 1 with groupID 0
Info 2023-12-14 08:48:34MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":13.24,"indicator_mode":"on/off","last_seen":"2023-12-14T07:48:40.645Z","linkquality":136,"power":5,"power_outage_memory":"restore","state":"ON","voltage":221}'
Info 2023-12-14 08:48:45MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":13.24,"indicator_mode":"on/off","last_seen":"2023-12-14T07:48:40.645Z","linkquality":136,"power":0,"power_outage_memory":"restore","state":"ON","voltage":221}'
Info 2023-12-14 08:48:45MQTT publish: topic 'zigbee2mqtt/Congélateur', payload '{"child_lock":"UNLOCK","current":0,"energy":13.24,"indicator_mode":"on/off","last_seen":"2023-12-14T07:48:40.645Z","linkquality":136,"power":0,"power_outage_memory":"restore","state":"ON","voltage":221}'
```

I can see that after a zigbee message containing power, the mqtt payload is correct after the third publish, i suppose this is your converter that has done its job (11 seconds after the zigbee message).

But it seems to work only when i configure a report to 5 seconds, is it possible to keep my 300 seconds i set before ?
If i understand well your converter, you wait minimumInterval * 2 + 1, so if i put back 300, it will be 601 seconds (11min) before publishing the right 0 value, right ? (i don't need immediate feedback of 0 power, i can wait 10 minutes, it is not a problem)

Koenkk, 2023-12-15:
> But it seems to work only when i configure a report to 5 seconds, is it possible to keep my 300 seconds i set before ? 

Why do you want this? See also my message from https://github.com/Koenkk/zigbee2mqtt/issues/19977#issuecomment-1850663611

Skuair, 2023-12-15:
I don't want to spam my zigbee network from this device, i don't need too many reports.

But in fact, even with a report set to 5 seconds, i don't see a zigbee message log every 5 seconds. So, sorry, but i think i misunderstand how the report settings works : eg, activePower -> minInterval 5s, maxInterval 3600s, min change 50W -> what is the first event ? minInterval or minChange ? This is what i'm not sure to understand (i'm a zigbee newbie).

Koenkk, 2023-12-16:
It will publish only after minInterval if the values change more than the minChange. As long as the minChange is not reached, it will not publish anything (until it reaches the maxInterval)

Skuair, 2023-12-16:
Thanks for your explanations.

So, for this fix, I have to keep the converter forever for this device or you will integrate the fix in future version ?

Koenkk, 2023-12-17:
I integrated the final changes, they are a bit different than in the ext converter (and I cannot provide one because it required some more changes). Could you check the latest dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Koenkk, 2023-12-19:
Pushed a fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Skuair, 2023-12-20:
Now i have this one (with 3 attemps):
```
Error 2023-12-20 22:00:33 Failed to configure 'Congélateur', attempt 1 (Error: Read 0xa4c1380d39ec562a/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24056 - 1 - 4 - 0 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
```

Koenkk, 2023-12-21:
That's strange, it previously became past this part, can you try to re-pair the device?

Skuair, 2023-12-21:
Ok, in fact the addon version was not up to date, i forgot to uninstall and resinstall.
Now it is up to date, i don't see any errors. Just, the reports setting of the plug have been restored to default settings it seems.

Koenkk, 2023-12-21:
Great, so everything is ok now?

Skuair, 2023-12-21:
Yes seems ok now. Thanks.

closed 2023-12-21 by Skuair