Koenkk/zigbee2mqtt #20004: Part of Zigbee2mqtt entities are no longer being provided by the mqtt integration

cancan92, 2023-12-03:
### What happened?

Hello
I'm using Zigbee2mqtt on HA for 4 years and everything was great. This week suddenly I had a mistake, loosing identities coming from Zigbee2mqtt . Ok I have more than 100 zigbee products but it always was ok.
For exemple : I have a switch for my "Bibliothèque". The status is coming right from zigbee but I can't see the identity "switch.bibliotheque" as usual on lovelace (see picture 1 & 2). It is strange because I can still see "sensor.bibliotheque_power".

From Zigbee2mqtt web frontend I am able to use this switch.bibliotheque (see picture 3) !?!

But thus automations and scripts using those entities are no longer working ...
It looks like lot of zigbee switches, lights are no more reachable when other products like temperature, wall switches are ok

I am using Mosquitto broker version: 6.4.0 and last version 1.34.0-1 of Zigbee2mqtt (it is the same trouble with version Edge) and Conbee II key with a cable. I even tried to change with a new Conbee II but no change.

I also tried to delete from HA such an identity no more reachable. I also deleted the MQTT integration. I launched the integration again. It became no more reachable (see picture 4), even when I delete the equipment from the zigbee network and subscribe it again on the zigbee network !

Do you know what is the mistake ? I'm lost ...

Thanks in advance

[image]

[image]

[image]

[image]

### What did you expect to happen?

I expected to still see my zigbee switch entities from HA to use with scripts & automations 

### How to reproduce it (minimal and precise)

Alway appends

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

NA

### Adapter

Conbee II

### Debug log

no bug seen on logs

cancan92, 2023-12-04:
Here is the version of my Conbee II

[image]

DubhAd, 2023-12-04:
What version of Home Assistant? You need to be on _at least_ 2023.8.0, but ideally newer.

cancan92, 2023-12-04:
I have version HA Core 2023.7.3. 
I tried with the lastest version 2023.11..3 without solving the bug. I came back to the previous version

DubhAd, 2023-12-05:
You need to be on at least 2023.8.0.

Changes were made in Z2M 1.33.0 that require this.

cancan92, 2023-12-05:
@DubhAd 🥇 THANKS A LOT ! I don't know what I have done before but I just installed again version 2023.11.3 and now it works fine. Thanks again for helping me. It's a so real pleasure to find back my switches and their scripts and automations linked 👍

closed 2023-12-05 by cancan92