Koenkk/zigbee2mqtt #20007: [New device support]:

mrdroopy, 2023-12-03:
### Link

https://www.ikea.com/se/sv/p/parasoll-doerr--foenstersensor-smart-vit-80504308/

### Database entry

{"id":56,"type":"EndDevice","ieeeAddr":"0x142d41fffe5285b0","nwkAddr":42788,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"PARASOLL Door/Window Sensor","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,1,3,32,2821,4096,64636,64641],"outClusterList":[3,4,6,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"PARASOLL Door/Window Sensor","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":8,"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230516","swBuildId":"1.0.19"}},"genPollCtrl":{"attributes":{"checkinInterval":172800}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeabcdb7","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1026,"inClusterList":[0,3,1280],"outClusterList":[3],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffeabcdb7","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230516","swBuildId":"1.0.19","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1701639657291,"defaultSendRequestWhen":"active","checkinInterval":43200}

### Comments

Not on the NL ikea site yet, but it is on the swedish one. It is available as of now in NL, at least in Delft.
Didn't receive a battery status yet, but that may take a while. the "Door state" property works as a charm with supplied external converter.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PARASOLL Door/Window Sensor'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'PARASOLL Door/Window Sensor', // Vendor model number, look on the device for a model number
    vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
    description: 'IKEA PARASOLL Door/Window Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_contact_alarm_1], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.contact()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

rjblake, 2023-12-04:
Installed same from IKEA Delft yesterday. According to print on the device, it is Model E2013. Not entirely sure what the device exposes, but ended up adding using the following:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
// const ea = exposes.access;

const fzLocal = {    
    ikea_contact: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            const lookup = {0: 'closed', 1: 'open'};
            return {
                contact: !((zoneStatus & 1) > 0),
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
};

const definition = {
    zigbeeModel: ['PARASOLL Door/Window Sensor'],
    model: 'E2013',
    vendor: 'IKEA',
    description: 'PARASOLL Door/Window Sensor',
    fromZigbee: [fzLocal.ikea_contact],
    toZigbee: [],
    exposes: [e.contact(), e.battery_low()],
};

module.exports = definition;
`

Koenkk, 2023-12-04:
@rjblake looks good, battery % might work with [this ext converter](https://github.com/Koenkk/zigbee2mqtt/issues/20007)

rjblake, 2023-12-13:
@Koenkk - I tried it, but seems there is no battery % being reported. Seems it only exposes the open/close and battery low status.

Koenkk, 2023-12-13:
Thanks for checking, since this device is supported now I will close this.

closed 2023-12-13 by Koenkk

europeke, 2023-12-21:
Hi! 
I have the Ikea parasoll successfully paired (Pressing 4x pairing under 5sec did it), but it is not supported by zigbee2mqtt. I am running 1.34.

Is this defintion mentioned above already merged in 1.34? I had same issue with Ikea vallhorn but updateing from previous version to z2m 1.34 added the support and now is working like a charme.

I am not familiar with adding the defintion by my own. So this is just a qustion. I can wait some weeks :)

rufferson, 2023-12-21:
For me it works with this ext-converter (including battery)

```
$ cat IKEA-Parasoll.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PARASOLL Door/Window Sensor'],
    model: 'E2013',
    vendor: 'IKEA',
    description: 'PARASOLL Door/Window Sensor',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
    toZigbee: [],
    exposes: [e.battery(), e.contact()],
};

module.exports = definition;
```

europeke, 2023-12-22:
Where to add this definition?

rufferson, 2023-12-22:
As written [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device) just create a file in `data/` folder:
```
data $ ls
IKEA-Parasoll.js            configuration.example.yaml  configuration.yaml          database.db                 log                         state.json
data $ grep -B1 IKEA-Parasoll.js configuration.yaml
external_converters:
  - IKEA-Parasoll.js
```
Restart z2m after adding the config section

europeke, 2023-12-22:
Got it. Many thanks

usma0118, 2023-12-23:
Is this supported now or should i do workaround as described?

rufferson, 2023-12-26:
It is [merged](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6632) into converters' master, so will be available in dev branch of z2m, but not in z2m master yet

larastafarian, 2023-12-27:
Hi, I done all the above steps but device still shows up as unsupported? Any ideas what im doing wrong?

Added file IKEA-Parasoll.js with the below in config/zigbee2mqtt/ 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PARASOLL Door/Window Sensor'],
    model: 'E2013',
    vendor: 'IKEA',
    description: 'PARASOLL Door/Window Sensor',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
    toZigbee: [],
    exposes: [e.battery(), e.contact()],
};

module.exports = definition;

Also added Log level:debug & external converters: - IKEA-Parasoll.js in config yaml in the config/zigbee2mqtt/ 

advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  external_converters:
    - IKEA-Parasoll.js

rufferson, 2023-12-28:
Did you restart the z2m after changing the config? What debug is saying when you start the service and add the device? i.e. debug might complain about syntax either at start or at device pairing. I didn't re-pair the device though, it paired ok just was showing unsupported. But i did have several typos in converter before it accepted it.

JBS5, 2023-12-31:
When you change the sensor from closed to open to closed very fast, the state stays at open. After changing the sensor to open and then to close again, the state is changed to closed correcty. So it seems it has some trouble with very fast changes.

rufferson, 2023-12-31:
yes noticed that as well, luckily with door inertia it's not easy but forced me to move magnet closer to the sensor even though there's some room for sensitivity. But if it stays at the rim of sensitivity it generates this kind of noise and hence state of the sensor is unpredictable. When it's close the door need to pick the speed to go far enough which gives that reliable state change.
Perhaps new fw upgrade will add some sort of debounce and state reconfirmation.

x2005254, 2024-01-02:
Hello,

I have same issue device is showed, but the problem I have is as soon I added the key bellow on configuration.yaml 
- IKEA-Parasoll.js

as soon z2m is restarted the added line is deleted, should I stop first z2m modify the file then start ?

JBS5, 2024-01-02:
> Hello,
> 
> I have same issue device is showed, but the problem I have is as soon I added the key bellow on configuration.yaml
> 
> * IKEA-Parasoll.js
> 
> as soon z2m is restarted the added line is deleted, should I stop first z2m modify the file then start ?

I use the Z2M UI for adding External Convertor file references: Settings > External Convertors:

[image]

x2005254, 2024-01-02:
> I use the Z2M UI for adding External Convertor file references: Settings > External Convertors: [...]

thanks for the quick fix :) and happy new year !!!
[image]

LeoWinterDE, 2024-01-03:
Please add the device to https://zigbee.blakadder.com/

cubic3d, 2024-01-03:
I'm not sure if the device is added properly. It seems to be missing OTA capabilities and I'm not sure if the binary `Tamper` attribute applies to the device. I don't see any button or sensor for the back lids nor does the value change if it's open.

JBS5, 2024-01-04:
> I'm not sure if the device is added properly. It seems to be missing OTA capabilities and I'm not sure if the binary `Tamper` attribute applies to the device. I don't see any button or sensor for the back lids nor does the value change if it's open.

`e.tamper` will be removed with a new PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6632#issuecomment-1872521682

But, I do not have a very stable IKEA Parasoll as well. State is not always updating, and every now en then it gets unavailable.

rufferson, 2024-01-04:
true dat, was working quite ok for several weeks but since today afternoon is not available, led constantly on, in z2m log is spamming with device_announced.
Although I suspect there could be contribution of my conbee2 dongle to this state, a week ago I had similar situation with PIR sensor, was solved only by force-removing and re-adding it.

sebw, 2024-01-09:
@rufferson most likely Conbee 2 shenanigans.. I sold mine and got a Sonoff gateway instead, which has been flawless for the past couple months. Conbee 2 (my first gateway entering the Zigbee world) made me think the whole Zigbee protocol was flaky. I'm not sure why it gets praised so much. 

Just got some Parasoll now and just paired through a TRADFRI repeater and Sonoff gw. I'll report back if things are unstable.

JBS5, 2024-01-09:
@sebw I am using a Sonoff as well and have the same unstable Parasoll for now.

sebw, 2024-01-09:
@JBS5 I can try to reproduce. Does it become unstable when not opening the window/door for a while? Or unstable even if the sensor is fairly "active"?

JBS5, 2024-01-09:
@sebw For me, both.

usma0118, 2024-01-10:
@sebw facing same after upgrading to 1.35.1... trying to figure out of Z2M docker restarts due to something

rufferson, 2024-01-10:
Ok I bought cc2652p stick and moved this contact to it. Will see if it is any better but another thing I noticed is that this contacts devours battery quite rapidly, which contributes to my stability issues. like leaving open state and never returning back only for me to figure out the battery is dead.

Aragur, 2024-01-16:
Does anybody have difficulties paring?
I successfully paired 2 yesterday. Today I cannot pair anymore. Nothing in the logs.