Koenkk/zigbee2mqtt #20009: translation error for NL batterylevel

mrdroopy, 2023-12-03:
### What happened?

In Dutch, on the "Toont" tab, the battery level text shows "Batterij **resertend** in %, kan tot 24u duren voor dit gerapporteerd wordt."
****

### What did you expect to happen?

should be "resterend"

### How to reproduce it (minimal and precise)

select battery powered zigbee device, go to "Toont" tab.

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

Koenkk, 2023-12-04:
Please click the pencil (edit) icon [here](https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/src/i18n/locales/nl.json) to create a fix for it

aumaticslex, 2023-12-06:
fixed and created pull request.

closed 2024-07-04 by github-actions[bot]