Koenkk/zigbee2mqtt #20011: Operation Mode on Xiaomi Aqara WS-USC03 switches throws an undefined error.

paperdigits, 2023-12-04:
### What happened?

When trying to change the `Operation_mode` in all of my Xiaomi Aqara WS-USC03, I gt the error 

```
error 2023-12-03 17:13:46Publish 'set' 'operation_mode' to 'kitchen_light_switch' failed: 'Error: undefined is not a object, got string ("decoupled")'
```

### What did you expect to happen?

I expected the `Operation_mode` to swtich.

This worked in the previous version.

### How to reproduce it (minimal and precise)

1. Go to the "Exposes" tab for the device
2. Change the `Operation Mode`
3. Get the error above

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Slaesh

### Debug log

error 2023-12-03 17:23:43Publish 'set' 'operation_mode' to 'kitchen_light_switch' failed: 'Error: undefined is not a object, got string ("decoupled")'
debug 2023-12-03 17:23:43Error: undefined is not a object, got string ("decoupled") at Object.assertObject (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:511:15) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:2488:19) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:529:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:141:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27) at WebSocket.emit (node:events:517:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20) at Receiver.emit (node:events:517:28)

Ajhomepn, 2023-12-04:
Also Xiaomi WS-EUK03 whit the same issue, it was working before the update of the 1/12/2023

pusheax0, 2023-12-04:
WS-USC01 as well with latest dev commit and Sonoff ZBDongle-P with latest firmware.

Pel1can111, 2023-12-04:
Got a bunch of WS-EUK04's. Same error, rolling back to previous version solved it.

Koenkk, 2023-12-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/19960

closed 2023-12-04 by Koenkk

pfurrie, 2023-12-04:
Our WS-USC04 switches are having the same issue, at least those that we installed only a fews days ago.  Those installed a week before that are still in decoupled mode, but now I don't dare try toggling them away from that mode.

The "State" value for a currently working switch (is still decoupled) is as follows:
```
{
    "consumption": 0,
    "current": 0,
    "device_temperature": 27,
    "energy": 0,
    "flip_indicator_light": "ON",
    "linkquality": 255,
    "operation_mode_bottom": "decoupled",
    "operation_mode_top": "decoupled",
    "power": 0,
    "power_outage_count": 3,
    "power_outage_memory": true,
    "state": "OFF",
    "state_bottom": "OFF",
    "state_top": "OFF",
    "voltage": 122,
    "action": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```
The operational mode for the switches than won't decouple are listed as "null"

pfurrie, 2023-12-04:
Okay, restored Z2M to version 1.33.2-1, and now the decouple action is working again.

paperdigits, 2023-12-04:
The problem for me is that .34.0 fixes a lot of issues I was having with interactions with Home Assistant, and I'd rather stay on 34 and have this issue than roll back and have issues with HA.

pfurrie, 2023-12-04:
What sort of other problems are you having pre-.34.0 ?  I could be having those as well, and don't know what I'm getting.  Maybe I can just configure the decoupling on the switches, and **then** switch to the same version you have.  When we were on the newer version, the switches I'd already configured were still working decoupled, and only had a problem when that setting was adjusted.

paperdigits, 2023-12-04:
From Home Assistant, using the last version of 1.33.x, toggling lights on/off or changing colors of lights was quite slow or sometimes didn't happen. This means that my SO's use via homekit was also quite bad :( going to 1.34 solves those issues but now I can't change the switch mode. Not a big deal as only one of my switches isn't configured as I'd like. HA/homekit/siri are snappy again.