Koenkk/zigbee2mqtt #20020: SYMFONISK sound remote gen2 partially broken in 1.34.0

SaschaRH, 2023-12-04:
### What happened?

After updating  Zigbee2MQTT to 1.34 the '.' and '..' buttons are no longer working. Got it to work again after rollback to 1.33.2.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

ChrisHaPunkt, 2023-12-04:
Same here, 
1 and 2 dots:
Long press: not working
Single Press: not working
Double Press: working

Downgrade to 1.33.2-1 and the long and single press working again

markus-barta, 2023-12-04:
**Same problem here: 1 and 2 dot buttons:**
Single Press, Long press: not working
Double Press: working

TeHashX, 2023-12-04:
See https://github.com/Koenkk/zigbee2mqtt/issues/19971#issuecomment-1837231063

SaschaRH, 2023-12-04:
Closed since it's a duplicate of [#19971 ]

Koenkk, 2023-12-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/19971

closed 2023-12-04 by Koenkk