Koenkk/zigbee2mqtt #20022: SNZB-03 Unsupported

MegoVaTx, 2023-12-04:
### What happened?

Friendly name
0xa4c1380c6f1d632f
Description
Last seen
N/A
Availability

[Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
EndDevice
Zigbee Model
SNZB-03
Zigbee Manufacturer
eWeLink
Support status
Not supported

IEEE Address
0xa4c1380c6f1d632f
Network address
0xD5EF
Firmware version
0122052017
Power
Battery 
Interview completed
True

Zigbee2MQTT version
1.29.2-dev commit: [82fd062](https://github.com/Koenkk/zigbee2mqtt/commit/82fd062)
Coordinator type
EZSP v8
Coordinator revision
6.7.8.0 build 373
Coordinator IEEE Address
0x000d6f001740c949
Frontend version
0.6.124

Im using Zigbee2MQTT Edge and its staing that's its unsupported even though it suppose to be.

Received message from unsupported device with Zigbee model 'SNZB-03' and manufacturer name 'eWeLink'
Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### What did you expect to happen?

Fix support status

### How to reproduce it (minimal and precise)

Pair 
SNZB-03

### Zigbee2MQTT version

1.29.2-dev 

### Adapter firmware version

6.7.8.0 build 373

### Adapter

CECOMINOD016164

Koenkk, 2023-12-04:
You are using an outdated z2m version, update to the latest (1.34.0)

closed 2023-12-04 by Koenkk

MegoVaTx, 2023-12-11:
Thank you, 
To update you Remove the addon and then install it again, it installs the latest version.