Koenkk/zigbee2mqtt #20028: TS011F _TZ3000_okaz9tjs plug No Energy Consumption and no OTA update

arisermpo87, 2023-12-04:
### What happened?

I have 3 of these plugs. They are paired and they show as TS011F _TZ3000_okaz9tjs / [TS011F_plug_1]. 
They report Power, Current and Voltage bur no Energy (appears as "null").
Also the FW version is 0122052017 and when I check for updates I get "Failed to check if update available (No image available for imageType '524').

### What did you expect to happen?

Have energy metering.
Update FW OTA.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Slaesh CC2652RB

### Debug log

Info 2023-12-04 22:47:40MQTT publish: topic 'zigbee2mqtt/TV plug', payload '{"child_lock":"UNLOCK","current":231,"energy":null,"indicator_mode":"off/on","linkquality":109,"power":16,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":226}'

DataGhost, 2023-12-05:
I have these too, _TZ3000_okaz9tjs TS011F_plug_1, same firmware version, same firmware update failure message. They also report current in mA instead of A, but Z2M doesn't know that and displays thousands of amps. I've verified with a current meter that it's accurate in mA.

But it seems a bit strange to me that energy consumption isn't reported for these plugs. I also have a few _TZ3000_gvn91tmx (also TS011F_plug_1) that report it correctly, seemingly through messages like:
```
zigbee2mqtt  | Zigbee2MQTT:debug 2023-12-05 23:05:49: Received Zigbee message from 'S Droger', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,738]}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-05 23:05:49: MQTT publish: topic 'zigbee2mqtt/S Droger', payload '{"child_lock":"UNLOCK","current":0,"energy":7.38,"indicator_mode":"off","linkquality":192,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":235}'
```

The _TZ3000_okaz9tjs report the same message but that's not picked up somehow:
```
zigbee2mqtt  | Zigbee2MQTT:debug 2023-12-05 23:06:14: Received Zigbee message from 'S nieuw KO1ok', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,47]}' from endpoint 1 with groupID 0
[...]
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-05 23:06:33: MQTT publish: topic 'zigbee2mqtt/S nieuw KO1ok', payload '{"child_lock":"UNLOCK","current":21,"energy":null,"indicator_mode":"off/on","linkquality":108,"power":2,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":237}'
```

I'm not sure if I understand all of this correctly but to me it seems that the messages are practically identical but just not picked up at all for the latter plug. Haven't done many tests with them either so I'm not sure if it reports in the same units (divisor) but I believe it should have been exposed in the state regardless.

My setup is Z2M (docker) 1.34.0 aae7312, Sonoff ZBDongle-E running coordinator 6.10.3.0 build 297.

pjanuario, 2023-12-07:
I also bought 3 plugs (the only one i already tested) present same symptoms, is there any way to push the firmware upgrade manually to check if it's firmware issue?

Schaumschlaeger, 2023-12-08:
i have the same issue.
Modell: TS011F
Manufacturer: _TZ3000_0zfrhq4i
Firmware: 0122052017
Zigbee2MQTT Version: 1.34.0

```"Failed to check if update available for 'xxx' (No image available for imageType '524')"```

```
{
"child_lock":"UNLOCK",
"current":0.45,
"energy":8.14,
"indicator_mode":"off/on",
"linkquality":192,
"power":72,
"power_outage_memory":"off",
"state":"ON",
"update":{"installed_version":-1,
"latest_version":-1,
"state":null},
"update_available":null,"voltage":227
}

ve1koz111, 2023-12-08:
Have the same problem. Following

allthesebugsv2, 2023-12-09:
I also have a few problems with the _TZ3000_okaz9tjs plug. None of these issues are present with the _TZ3000_gvn91tmx plug, so I hope it is all fixable. I especially hope so after reading @DataGhost's comment as the information is published. I am also on 1.34.0 using a Sonoff ZBDongle-E.

- no energy consumption and OTA update as mentioned here
- wrong unit for the current: the plug reports mA while z2m displays A
- changing any reporting interval (like Min rep change) for activePower (haElectricalMeasurement) is not offered and not working for this plug while it works for _gvn91tmx
- there is an error during interview when the plug joins z2m, maybe related to all of this (though it is added successfully in the end):
```
Failed to configure '0xa4c138fbcbe518b6', attempt 2 (Error: ConfigureReporting 0xa4c138fbcbe518b6/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":10}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22) at Object.activePower (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:204:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3290:17) at Configure.configure (/app/lib/extension/configure.ts:117:13))
```
This error might be related with me trying to set this attribute once. But generally the reporting seems to be awry; for one of my  okaz9tjs's the reporting tab is completely empty.

kennerblick, 2023-12-10:
still the same probleme here with my:
Zigbee-Modell
TS011F
Zigbee-Hersteller
_TZ3000_typdpbpg

CoccoAndrea, 2023-12-11:
Have the same problem. Following.

{
    "child_lock": "UNLOCK",
    "current": 0,
    "indicator_mode": "off",
    "linkquality": 112,
    "power": 0,
    "power_outage_memory": "restore",
    "state": "ON",
    "voltage": 237,
    "energy": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}

Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_typdpbpg

Koenkk, 2023-12-11:
Could you provide the data/database.db entry of this device?

arisermpo87, 2023-12-11:
I hope this is what you are looking for:

{"id":42,"type":"Router","ieeeAddr":"0xa4c138c399331e03","nwkAddr":37314,"manufId":4107,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"manufacturerName":"_TZ3000_okaz9tjs","zclVersion":3,"appVersion":112,"modelId":"TS011F","powerSource":1}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":226,"rmsCurrent":178,"activePower":15}},"seMetering":{"attributes":{"currentSummDelivered":[0,491]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc41b2","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc41b2","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc41b2","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1702319383019,"defaultSendRequestWhen":"immediate"}

arisermpo87, 2023-12-11:
Ok. Now energy reporting is working!

The device is missing on the OTA page.

arisermpo87, 2023-12-11:
No problem. Thanks a lot for your help and all your work.

DataGhost, 2023-12-11:
That fixes the issues for me as well, except that it's now reported as BSD29_1 with an incorrect picture. If I understand it correctly, it's now only polling the energy? But I'm curious though, why the energy messages I saw in my logs were not picked up. It seems to indicate that those plugs shouldn't require any polling.

allthesebugsv2, 2023-12-11:
> That fixes the issues for me as well, except that it's now reported as BSD29_1 with an incorrect picture. If I understand it correctly, it's now only polling the energy? But I'm curious though, why [...]

I can confirm these observations.

But I still have empty reporting tabs for the _TZ3000_okaz9tjs plug:
[image]

While it looks like this for the _TZ3000_gvn91tmx plug:
[image]

Manually trying to set activePower for haElectricalMeasurement leads to the following error:
```
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0xa4c138d42680f5e8/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":10}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'
```

DataGhost, 2023-12-12:
I think it's much simpler. Looks to be entirely related to https://github.com/Koenkk/zigbee2mqtt/discussions/19680#discussioncomment-7667035. We're seeing the same INVALID_DATA_TYPE error there. It seems that after using the external converter linked in this thread the plugs kept reporting or getting polled for energy, even with the external converter unloaded. So I removed one of the devices from the network, restarted Z2M, re-added it so it was back to the old (wrong) behaviour and verified it was again reporting current incorrectly, and no energy. I then added my own external converter, basically a copy of TS011F_plug_1 from tuya.ts, but extended the check https://github.com/Koenkk/zigbee-herdsman-converters/blob/0c21893dd438aeb2b5fb26b78ee1d5b4f6bc2684/src/devices/tuya.ts#L3294 to include this device manufacturer. This made it configure properly, after which energy became available (I guess without any polling). That also fixed the current measurement. Probably because those two were going to be set up only after that error.

I could make a patch/PR but the change is basically:
```
- if (device.manufacturerName !== '_TZ3000_0zfrhq4i') {
+ if (device.manufacturerName !== '_TZ3000_0zfrhq4i' && device.manufacturerName !== '_TZ3000_okaz9tjs') {
```
and I'm not sure if that's all the way up to coding style etc. It also doesn't fix OTA which will remain unavailable for this specific device until someone pulls the image (if available) and I'm unsure if showing the imagetype error is considered problematic.

allthesebugsv2, 2023-12-12:
@DataGhost Would you mind posting your converter somewhere? I am not really familiar with JavaScript and somehow always mess up the syntax :(

DataGhost, 2023-12-12:
@allthesebugsv2  I made a small change to it (removed "160, 100, 69, 68, 65, 64, " from the fingerprint map) to hopefully minimise the possible side-effects for other "unrelated" plugs (which I don't have but other people might) but otherwise this is what I'm using now: https://gist.github.com/DataGhost/572e79e08c0e7eb22773f54c1648b263

allthesebugsv2, 2023-12-13:
@DataGhost thank you very much, that works very well!

The only thing not working (and I guess we explicitly turn that off with the converter) is setting different reporting intervals (or minimal reported changes) for activePower. Which is a shame since that is the most interesting measurement, at least for me

DataGhost, 2023-12-13:
I have no clue, very little experience with Z2M (in terms of code), I don't have any thing / experience to debug the protocol either. I assume if it was trivial for the plugs in the other issue, that would have been chosen instead. I personally don't use the reporting tab so I'm not sure what you'd want to change there, for me it's reporting the way I want it to report.

Koenkk, 2023-12-13:
@DataGhost looks good, could you make a PR? (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts)

DataGhost, 2023-12-13:
Sure. Just checking because I see another plug in this thread: @pjanuario @kennerblick @CoccoAndrea could either of your verify that my updated gist https://gist.github.com/DataGhost/572e79e08c0e7eb22773f54c1648b263 used as an external converter works for your _TZ3000_typdpbpg plugs?

@Koenkk I assume they're affected by the same thing so I'll include those plugs in the PR as well, but you might want to wait to get confirmation obviously.

Vince92500, 2023-12-15:
Hi all,
I have the _TZ3000_okaz9tjs and the issues previously reported.
I'm not familiar with the building process. Should I try to fix with the external converter or wait the fix is included in a new HA version ? How can I know when it would be integrated ?

DataGhost, 2023-12-15:
You don't need to build anything. When the PR I opened is approved, you'll be able to see that in this ticket and the fix should then appear in the next release which is generally the 1st of the month I believe. If you can live with only power and voltage reporting for now you don't need to do anything. But if you really want correct energy and current reporting right now, you'll have to use the external converter linked two comments up.

closed 2023-12-15 by Koenkk

pjanuario, 2023-12-15:
> Sure. Just checking because I see another plug in this thread: @pjanuario @kennerblick @CoccoAndrea could either of your verify that my updated gist [...]

Sorry I was traveling and with a lot of work.
@DataGhost how can i test the gist you posted to confirm it's fixing for `_TZ3000_typdpbpg`.
 I will test it tomorrow.

pjanuario, 2023-12-15:
> Sorry I was traveling and with a lot of work. @DataGhost how can i test the gist you posted to confirm it's fixing for `_TZ3000_typdpbpg`. I will test it tomorrow.

I guess this is the answer to my question right?
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device

pjanuario, 2023-12-15:
I just did a quick try to test it, but got some errors related with dependencies.

DataGhost, 2023-12-17:
@pjanuario That should be it yes, save the gist as some .js file and reference it in the config under external_converters, just as described in https://github.com/Koenkk/zigbee2mqtt/issues/20028#issuecomment-1850710040. It shouldn't throw any dependency errors, so something must have gone wrong there. Unless you post those errors there's no way to know what, though. In any case, it's merged now, so if you're running Z2M in docker, an alternative is to use the latest-dev docker image and see if that fixes your plugs. And afterwards obviously revert to latest or whatever you were using.

jaspweowaale, 2024-05-28:
Im not sure why, I also have this _TZ3000_okaz9tjs, look different and original it was  _TZ3000_cehuw1lw and yes it also does not measure power it looks like, to me as I ordered the same Item looks like bake and switch thing

Super happy with the original _TZ3000_cehuw1lw and I see the same for other Items, ones it sell they swap out with different models