Koenkk/zigbee2mqtt #20030: [New device support]: IKEA SOMRIG shortcut button E2211

cklam2, 2023-12-04:
### Link

https://www.ikea.com/se/sv/p/somrig--genvaegsknapp-vit-smart-50560334/

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0x5cc7c1fffe61fe84","nwkAddr":46447,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"SOMRIG shortcut button","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,1,3,4,32,4096,64636,64640],"outClusterList":[3,4,6,8,25,4096,64640],"clusters":{"genBasic":{"attributes":{"modelId":"SOMRIG shortcut button","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":8,"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230308","swBuildId":"1.0.20"}},"genPollCtrl":{"attributes":{"checkinInterval":172800}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8918e","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":6,"inClusterList":[0,3,4,64640],"outClusterList":[3,4,6,8,64640],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230308","swBuildId":"1.0.20","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1701725878866,"defaultSendRequestWhen":"active","checkinInterval":43200}

### Comments

New IKEA shortcut button, can buy in Netherlands but not listed in NL page. Tried following converter, unfortunately nothing happens/shows up in log when pressing buttons. Also not sure what I should put in exposes, Some help would be appreciated.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SOMRIG shortcut button'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'E2211', // Vendor model number, look on the device for a model number
    vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
    description: 'SOMRIG shortcut button', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Koenkk, 2023-12-05:
Will be added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6574

Borrel88, 2024-01-01:
What is the status of this issue? On the compatible devices page I can see the device should be compatible.
But when I pair this device it stills says it is not supported.

Koenkk, 2024-01-01:
Supported in z2m 1.35.0!

closed 2024-01-01 by Koenkk

uncle-fed, 2024-01-06:
@Koenkk There seems to be a documentation mismatch. The docs state that the `action` values all start with the word `dots_` while in reality they start with `1_`  or `2_`. The docs also do not mention the `...double...`  presses for both buttons even though they are recognized.

I also cannot see how to efficiently deal with the fact that all of the types of button presses (single, double, hold) always generate 2 actions and the first one is always the same ( `_initial_press`). If you want your automations to distinguish between single / double / hold then you have to ignore the `initial_press` because it is always the same for any type of button click and check the second action that follows it (`short_release`, `double_press` and `long_press`). That would be reasonable, if there wasn't nearly a second delay between the `initial_press` and `short_release` (even if the button is pressed momentarily). For the automations that depend on the most simple click there would be close to one second latency in response. Very unfortunate for this otherwise perfect value for money.

Is this a hardware implementation / limitation? Does it also take almost a second to recognize a simple button click with the official IKEA hub? I wonder... Had high hopes that this would be THE button but looks like my quest for a perfect Zigbee button continues :-(   See [this ](https://github.com/Koenkk/zigbee2mqtt/issues/20407) as well.

Koenkk, 2024-01-07:
The actions are fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6834 already.

About the delays, you can check the actual commands send by the device in the debug log, if there is a delay, the switch simply takes time to sent them (I guess to detect if you are single, double or long pressing)

cklam2, 2024-01-07:
The delay comes from the device. With a simple single click the second Zigbee messages comes 1 second after the first one. 
Regarding the events, if you don't care how the user pressed the button, just check the `initial_press` event. It's instantly, actions execute immediately. For all other button actions:

Short press
1. `initial_press`
2. `short_release` (1 sec after initial press)

Double press
1. `initial_press`
2. `double_press` (immediately after double pressed)

Long press
1. `initial_press`
2. `long_press` (immediately after initial press)
3. `long_release` (after releasing button. Diff between long_press and long_release is how long it has been pressed)

Short, followed by long press:
1. `initial_press` (IKEA bug? device only sends 1 Zigbee message) 

Long, followed by short press:
1. `initial_press`
2. `long_press`
3. `long_release`
4. `double_press`

uncle-fed, 2024-01-07:
@cklam2 thanks a lot for laying it out. I can pretty much confirm all of it from my findings, including the bug where only the `initial_press` is being sent. My point was that of you do want to distinguish types of key presses (which was my original intention) you're facing a delay of 1 seconds for most simple action that will be used most often. Unacceptable for me (or rather for my parents who wouldn't be convinced at all if their main light started being "slow" while me claiming it is "smarter" now :))

I guess there is no way to avoid some extra latency if the button is supposed to detect double click. I wish there was a simple button that recognized single press instantly or hold (without generating that 'initial press' at all (and costed under 10 Eur :))

cklam2, 2024-01-07:
If you don't need the double click functionality, you can get the IKEA Rodret for €7.99 which looks similar to the Somrig.

uncle-fed, 2024-01-08:
I was looking at it on their web site but unfortunately my local IKEA store is out of stock and for a while now. I cannot be 100% sure but it looks like the Somrig has replaced Rodret. I hope I'm wrong..

cklam2, 2024-01-08:
Rodret was introduced in June '23 (as was the support for E2201 in zigbee2mqtt), don't think it becomes obsolete after such short period. Secondly, I was in IKEA today and there were many on the shelves. Nearly all Styrbars, which were used to control the demo bulbs have been replaced by Rodret. Thirdly, Somrig is a shortcut button to control scenes while Rodret is a power switch/dimmer.

devildant, 2024-01-22:
I'm the only one to have the link on/off and levelctrl which doesn't work?
I linked bulbs but nothing happens, ikea bug?

schnellerFuchs, 2024-03-01:
Having problems with long_press:

I still have the issue as mentioned above: If I press long, it first sends inital_press before it sends long_press. How can I avoid the initial_press?

Home Assistant 2024.2.5 and Zigbee2MQTT 1.36.0-1

thesebastianf, 2024-06-01:
> The delay comes from the device. With a simple single click the second Zigbee messages comes 1 second after the first one.
> Regarding the events, if you don't care how the user pressed the button, [...]

@Koenkk 
Is there a way that Not all presses result also additonally in  "Initial Press"? 
This basically means either only Use Initial Press or only Use any other Action but Initial Press

Thank you

cklam2, 2024-06-01:
Those events come from the IKEA device, not generated by zigbee2mqtt.

cianpdx, 2024-11-10:
just tried to pair a somrig: 

`2024-11-10 13:38:26	Device '0x6c5cb1fffe080fab' joined
2024-11-10 13:38:26	MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x6c5cb1fffe080fab","ieee_address":"0x6c5cb1fffe080fab"},"type":"device_joined"}'
2024-11-10 13:38:26	Starting interview of '0x6c5cb1fffe080fab'
2024-11-10 13:38:26	MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x6c5cb1fffe080fab","ieee_address":"0x6c5cb1fffe080fab","status":"started"},"type":"device_interview"}'
2024-11-10 13:38:35	Received Zigbee message from '0x6c5cb1fffe080fab', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":172800}' from endpoint 1 with groupID null
2024-11-10 13:38:35	Successfully interviewed '0x6c5cb1fffe080fab', device has successfully been paired
2024-11-10 13:38:35	Device '0x6c5cb1fffe080fab' with Zigbee model 'SOMRIG shortcut button' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
2024-11-10 13:38:35	MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x6c5cb1fffe080fab","ieee_address":"0x6c5cb1fffe080fab","status":"successful","supported":false},"type":"device_interview"}'
2024-11-10 13:40:43	Received Zigbee message from '0x6c5cb1fffe080fab', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":16777248,"imageType":15112,"manufacturerCode":4476}' from endpoint 1 with groupID null`