Koenkk/zigbee2mqtt #20032: [New device support]: Melery Zigbee Tuya Smart Wall Socket AU Electrical Plug Outlet

MnM001, 2023-12-05:
### Link

https://vi.aliexpress.com/item/1005005857622120.html?spm=a2g0o.order_list.order_list_main.302.50651802W8lyGq&gatewayAdapt=glo2vnm

### Database entry

{"id":68,"type":"Router","ieeeAddr":"0xa4c13810d9bbf5e4","nwkAddr":7738,"manufId":4417,"manufName":"_TZ3000_8nyaanzb","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701752583137,"defaultSendRequestWhen":"immediate"}

### Comments

I have these power wall sockets.
I made an external convertor for them which used to work good, however I have now noticed that if i remove and add the device again when pressing the power on button both buttons on the device are turned on. The same for the off button.
The devices that have not been removed and added again behave normally.

The extremal connectors I use are mainly based on the Makegood AU power points. The devices that I am using have the following Zigbee Manufacturer ID:

_TZ3000_rgpqqmbj (this the black version)
_TZ3000_8nyaanzb (white version)
_TZ3000_iy2c3n6p (white version)

I will show the _TZ3000_8nyaanzb external connector I am using but all 3 external connectors are identical apart from the Zigbee Manufacturer ID.

I have no idea why if devices are removed and added again dont perform as they used to.
So I am taking this opportunity to fix them and to add them to Zigbee and dont use external convertors anymore.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // MG-AUZG01 requires multiEndpoint only for on_off
    // https://github.com/Koenkk/zigbee2mqtt/issues/13190
    MGAUZG01_on_off: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('onOff')) {
                const endpointName = utils.getKey(model.endpoint(meta.device), msg.endpoint.ID);
                return {[`state_${endpointName}`]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_8nyaanzb'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    ota: ota.zigbeeOTA,
    extend: tuya.extend.switch({
        powerOutageMemory: true, indicatorMode: true, endpoints: ['l1', 'l2']}),
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
};

module.exports = definition;
```

Koenkk, 2023-12-05:
```
    description: 'Double Zigbee power point',
    configure: tuya.configureMagicPacket, // <- add this line
    ota: ota.zigbeeOTA,
```

Koenkk, 2023-12-06:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-06 by Koenkk

MnM001, 2023-12-06:
Thanks for that. I only see one device added but as mentioned above there are 3 devices that behave exactly the same (the one external convertor provided was an example as all 3 are the same but different IDs)"

_TZ3000_rgpqqmbj (this the black version)
_TZ3000_8nyaanzb (white version)
_TZ3000_iy2c3n6p (white version)

Were all 3 added?

Cheers.

Koenkk, 2023-12-07:
Done now!

Koenkk, 2024-02-01:
@MnM001 you mentioned that for some the `state` does not update when toggling them physically, without using any external converter, for which device is this? (I need the manufacturer name, e.g. `_TZ3000_rgpqqmbj`)

MnM001, 2024-02-01:
Hi @Koenkk 

I will explain what happens:

1. I am still using external converters
2. When I press a physical button on the device it does not show in the z2m GUI
3. All the devices are based on the Makegood template, which I believe also have power monitoring built in and it is showing in the GUI to. These devices do not have power monitoring at all.
4. The above issues show for all devices - _TZ3000_rgpqqmbj (this the black version), _TZ3000_8nyaanzb (white version) and _TZ3000_iy2c3n6p (white version)

The external converters look like this now:

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // MG-AUZG01 requires multiEndpoint only for on_off
    // https://github.com/Koenkk/zigbee2mqtt/issues/13190
    MGAUZG01_on_off: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('onOff')) {
                const endpointName = utils.getKey(model.endpoint(meta.device), msg.endpoint.ID);
                return {[`state_${endpointName}`]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_rgpqqmbj'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    configure: tuya.configureMagicPacket,
    ota: ota.zigbeeOTA,
    extend: tuya.extend.switch({
        powerOutageMemory: true, indicatorMode: true, endpoints: ['l1', 'l2']}),
    endpoint: (device) => {
        return {l1: 1, l2: 2};
    },
};

module.exports = definition;
```

Thanks for looking at this.

Koenkk, 2024-02-03:
2. Can you check if this ext converter fixes that: https://gist.github.com/Koenkk/9ec792a6e6a89091f9e1fb61a98f541a
3. I guess not for the device which you use an external converter for?

MnM001, 2024-02-04:
Hi,

2. I have checked that converter - it does not fix the issue.
3. Yes correct - devices using external converters do not show the power metering entities. I was just mentioning this as it they make into z2m (without using convertors) as the Makegood template they will be showing these entities (I think).

Koenkk, 2024-02-04:
- Does the configure succeed? (press the yellow refresh button on the device page)
- Can you provide the debug log when the plugs states are changed via the button on them?

MnM001, 2024-02-05:
- After I updated one of the external connector all devices for that manufacturer started updating their configuration upon z2m start.
 - I have tried to reconfigure a device by pressing the yellow button, but that didn't fix the issue.
 - When I have noticed that it still doesn't work, I manually deleted a device from z2m and them add it again, but it still didn't work.
 - I don't have logs right now as I am testing that other thing, but as soon as I can I will upload them here.

MnM001, 2024-02-09:
I have updated all the external connector files with what you suggested.
I have captured 2 logs files.

1. First log file is from when z2m started and autoconfigured the devices.

[PowerPoints-1.zip](https://github.com/Koenkk/zigbee2mqtt/files/14215897/PowerPoints-1.zip)

2. Second log is when I have pressed the reconfigure button (the yellow one) in GUI for one device.

[PowerPoints-2.zip](https://github.com/Koenkk/zigbee2mqtt/files/14215900/PowerPoints-2.zip)

Hope this helps.

Koenkk, 2024-02-11:
Looks good, can you now provide:

> Can you provide the debug log when the plugs states are changed via the button on them?

MnM001, 2024-02-12:
I have used ServerRoompowerPoint2, ServerRoompowerPoint3, ServerRoompowerPoint4, and ServerRoompowerPoint5 for this.

Turned both power buttons on on all of them (from ServerRoompowerPoint2 to ServerRoompowerPoint5) and then turned the power button off on all of them.

[20240121.zip](https://github.com/Koenkk/zigbee2mqtt/files/14234488/20240121.zip)

Koenkk, 2024-02-12:
Fixed it for all detected as `MG-AUZG01`

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

MnM001, 2024-02-13:
@Koenkk - I have tested it and the buttons now work!! Thank you for fixing it.

However one issue still remain. 
As mentioned before while the power points I have are based on the MG-AUZG01 template they are really not the same device.
MG-AUZG01 provides power monitoring, while the ones I have does not. MG-AUZG01 are Makegood devices while mine are Mellery.
So the last issue that needs to be fixed is (perhaps) for the devices I have (_TZ3000_rgpqqmbj, _TZ3000_8nyaanzb and _TZ3000_iy2c3n6p to have their own Mellery entry in z2m which will not display any power monitoring.

Here is a pic of the issue:

[image]

Many thanks again.

MnM001, 2024-02-13:
Yes is kind of looks like that - but then again Makegood looks similar too.
Unfortunately there is no model number on the box or device itself.

Koenkk, 2024-02-14:
Pushed a fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

MnM001, 2024-02-15:
I don't think that worked as intended.

All power points are now detected as power plugs ([TS011F_plug_1) with the wrong options:

[image]

Koenkk, 2024-02-15:
forgot something indeed, fixed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

MnM001, 2024-02-15:
OK that seems to work as expected now :)

One more thing I noticed - not sure if it affects anything or not. I see that the Power Monitoring options are still on for one of the buttons. Not sure if it needs to be disabled or not like for the second button.

[image]

Koenkk, 2024-02-18:
> One more thing I noticed - not sure if it affects anything or not. 

You can safely remove these

> Also one more thing I have noticed is that there is no OTA option now for these devices.

Added it back.

MnM001, 2024-02-23:
Hi @Koenkk 

I have removed and then added back one of these power points.
All works as it should, except for the Power-on behavior - is always OFF.
Trying to change it but it doesn't take the new settings.
Could you please have a look at this?

MnM001, 2024-02-23:
Also one more thing - there are no bindings:

[image]

Koenkk, 2024-02-24:
- Did the power on behaviour work when they were still detected as MG-AUZG01?
  - If yes, can you provide the debug log when setting it now?

> Also one more thing - there are no bindings:

If device reports when changing state through the physical switch, you can ignore this.

MnM001, 2024-02-26:
Yes this used to work before.
Here is the log (I have started fresh with Zigbee E dongle so is less for you to go through).

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14400685/log.txt)

Koenkk, 2024-02-26:
Can you now provide the same log when it doesn't work?

Note to self; this is send when it works:

```
2024-02-26T06:33:45.995Z zigbee-herdsman:controller:endpoint Write 0xa4c1387072a84382/1 genOnOff({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
```

MnM001, 2024-02-26:
Yes the logs were captured when the Power-on Behavior doesnt work.

I press the options in the GUI and is always blank. Below is zipped file with a video showing what is happening.

[20240227 074220-1.zip](https://github.com/Koenkk/zigbee2mqtt/files/14412521/20240227.074220-1.zip)

Koenkk, 2024-02-27:
Think I've found the issue, pushed a fix

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

MnM001, 2024-03-03:
Hi,

I can confirm that this fixed the issue with the wall power points. However, it seems that it has also "moved" this issues to the light switches now. 

All the light switches I have now show the same issue with the Power-on Behavior that the wall power points had. They used to work fine before.
The Manufacturer IDs for the lights switches are:

_TZ3000_qewo8dlz
_TZ3000_mrduubod
_TZ3000_hhiodade
_TZ3000_18ejxno0

MnM001, 2024-03-04:
Checked my light switches this morning and they worked OK - I could select Power-on Behavior.

Maybe it was the update I did last night?

zigbee_herdsman_converters_version 18.42.0

Lets consider this fixed. I will keep an eye on them and if there are issues I will post back here.

MnM001, 2024-03-26:
Hi @Koen i the latest dev build 0.40.0 all the work above is undone.
The power points are now back to being detected as TS011F plugs with energy metering.

[image]

[image]

Apart form that all my light switches do not show in the GUI anymore. The just disappeared overnight.

Will appreciate if you can look and fix please.

MnM001, 2024-03-27:
Re-installed the latest image and all come back as it was before. No idea as it seems is still version 0.40.0