Koenkk/zigbee2mqtt #20035: 1.34.0: Lots of `Cannot read properties of undefined (reading 'isDevice')`

samip5, 2023-12-05:
### What happened?

Upgraded to 1.34.0 and noticed a lot of errors when adding devices as an example.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20220219

### Adapter

Tube's cc2652p2 PoE coordinator

samip5, 2023-12-05:
[image]

As an example

Koenkk, 2023-12-05:
Please provide the debug log.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

samip5, 2023-12-06:
Currently running the dev version in which I don't see those appearing anymore so maybe it's not actually a problem?

Koenkk, 2023-12-06:
Could already have been fixed there, closing this for now.

closed 2023-12-06 by Koenkk