Koenkk/zigbee2mqtt #20038: [Wrong device]: Zimismart SPM01-D2TZ recognised as Yagusmart SPM01-D2TZ

mrmaximas, 2023-12-05:
### Link

https://www.zemismart.com/products/spm01-d2tz-zm

### Model

 SPM01-D2TZ

### Description

Zemismart Tuya Zigbee Real-time Smart Energy Monitor / Electric Energy Meter 

### Vendor

Zimismart

### Picture (link)

https://ueeshop.ly200-cdn.com/u_file/UPAC/UPAC387/2304/products/10/507226290e.jpg.500x500.jpg

### Database entry

{"id":53,"type":"Router","ieeeAddr":"0x187a3efffe358554","nwkAddr":415,"manufId":4098,"manufName":"_TZE200_bcusnqt8","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_bcusnqt8","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1701777078797,"defaultSendRequestWhen":"immediate"}

mrmaximas, 2023-12-05:
Additional - I found issue with current converter.
When trying to set Current calibration from -1 to -19 it does not affect to the reports, but with -20 it does.

mrmaximas, 2023-12-05:
@Koenkk

emilic14, 2024-06-23:
There is another issue with this configuration, It can not see negative values when connected to a home with photo voltaic injection.
It actually adds the values of consumption and injection and it is showing it as a positive value... I would expect it to show the actual injected power as negative (or possitive if when no injection the usage is negative... depending how the device was placed).

Also the correct link and name of the company doing this device is [Zemismart](https://www.zemismart.com/products/spm01-d2tz-zm) (not zimismart)

closed 2025-01-09 by github-actions[bot]