Koenkk/zigbee2mqtt #20039: Innr smart plug sp240 not showing power monitoring

asterix778, 2023-12-05:
### What happened?

Innr SP240 isnt a supported device, I followed the manual on the website to add unsupported devices.
Made a innr.JS file with the following code: 

 

> const` fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
> 
> const definition =  {
    zigbeeModel: ['SP 240'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SP 240', // Vendor model number, look on the device for a model number
    vendor: 'inrr', // Vendor of the device (only used for documentation and startup logging)
    description: 'smart plug innr', // Description of the device, copy from vendor site. (only used for documentation and startup            
    fromZigbee: [fz.electrical_measurement, fz.on_off], // We will add this later 
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch(), e.power(), e.energy(), e.current(), e.voltage().withAccess(ea.STATE),], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};
> 
> module.exports = definition;

Zigbee starts up I'm able to switch the device on and off but the energy readings are "Null"

### What did you expect to happen?

What I expected to happen was that I had used the right exposes to add al the things this plug provides.
I hope that someone could help me further and that we can add this device to the supported list.

### How to reproduce it (minimal and precise)

Using the code above you can reproduce it. Tried removing the sp240 plug and re-adding it using the same code same effect

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

Ocramius, 2023-12-11:
Wanted to send a patch today, but it was already done:

 * https://github.com/Koenkk/zigbee-herdsman-converters/pull/6637
 * https://github.com/Koenkk/zigbee-herdsman-converters/commit/8eb43be096eedcc322813987de3d6baf30c470ac

closed 2023-12-11 by Koenkk

Ocramius, 2023-12-11:
> Can confirm that your code works I will follow up if the reporting is correct if so i will change this problem to an add new device request.

The reporting with the custom converter from @jorni2000 is a bit off: getting wrong `Current`, `Power` and `Energy` readings.

I tried adjusting the divisors too, with no success.

That said, at least the plug turns on/off and reports voltage correctly :-)

bikeymouse, 2023-12-22:
I'm using the Dev-branch (from Docker) and the SP240 is recognised properly, including all attributes.
However the reporting is not working. 

The "Exposes"page in the Z2M front-end  shows the attributes with a "refresh" button. 
When pressed it indeed sends the new info, but automatic reporting is not done.

The fact that it shows refresh buttons for the attributes suggest that polling is configured (but not working):
[image]

These are my reporting settings:

[image]

bikeymouse, 2023-12-24:
Update: I did a factory reset an re-included the SP240 and now it is reporting properly.

Kugelfang666, 2023-12-28:
I just picked up a pair of the SP 240, any idea when the support will be merged into main?

and7891, 2023-12-28:
I have the same Problem. Unfortunately, I'm just a simple user. When will the sp240 be available with a homeassistant starndard installation? Thanks so much

bikeymouse, 2023-12-28:
If you use the [dev-Docker image](https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-d6c035dfadbe19fd75d09b84c6cf702b7e5c9b322b56b75bc5428af1db572db3?context=explore) the plug works fine with Home Assistant.

gewoonivar, 2023-12-29:
Also curious, when to expect this in the regular release schedule?

and7891, 2023-12-29:
hello everyone,
Unfortunately, I'm new to Linux. Can someone give me a short step by step guide on how to switch to the dev branch?
The instructions in https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html  didn't work. I don't have the opt directory. I didn't get any further with it. it would be nice if someone could help me

Koenkk, 2023-12-30:
Will be supported in the 1 January release.

Ocramius, 2024-01-01:
Just to confirm here: removed my custom external converter, upgrading Zigbee2MQTT, restarting, and everything reports correct wattage.

Thanks @Koenkk, and happy new year!

gewoonivar, 2024-01-02:
Updated to latest version, works like a charm when pairing. I had to factory reset the unit, and now all values are imported into HA.

However getting updates of the attributes is another thing, as @bikeymouse also noticed.

I changed the timing for periodically updating to have instant updates for all attributes. The only one that is not updating automatically is the 'currentSumDelivered', when manually 'refreshed' the value is updated accordingly. This should be done automatically.

Any ideas on how to fix this @Koenkk ?

[image]

Koenkk, 2024-01-02:
Does it work after doing a reconfigure and not touching the rep change for currentSummDelivered?

gewoonivar, 2024-01-03:
After one day I see 2 sockets have updated the currentSummDelivered automatically.

One of which I changed to the original reporting setting (for this attribute). The other one has a modified reporting setting.

A 3rd socket I factory reset this morning and used for the TV, has no update as the TV has not been used.

How to interpret the reporting settings for this attribute? It seems random at this point when the sockets are updating this attribute (currentSummDelivered).
The other attributes strictly follow the configured reporting timings according to their (changed) config.

stefan1495, 2024-01-04:
Is the reset still required? Paired my new Plugs but they seem not to report automatically. I do have to trigger the report manually via the refresh button on Zigbee2Mqtt

gewoonivar, 2024-01-04:
It seems that a factory reset is required @stefan1495 , as far as I experienced. But it might be different with your units.

As can be seen in the screenshot, the Quooker has multiple times reheated, especially when making diner last night. However the attribute currentSummDelivered has only updated twice.

[image]

This is with the following reporting configuration:

[image]

Any ideas @Koenkk ?

Koenkk, 2024-01-04:
@gewoonivar if I understand correctly, you want to see updates more frequently? Could you provide the data/database.db entry of your device?

gewoonivar, 2024-01-05:
@Koenkk yes.

Strange thing, I remote started another plug (#4), which has the same reporting config as plug #3. Plug #3 is reporting on short intervals, except for the currentSummDelivered as mentioned.

Plug #4 doesn't update at all once switched on. Only on manual 'refreshes' the values of all attributes are updated.

**Plug #3:**
{"id":19,"type":"Router","ieeeAddr":"0xa4c138cabd88fcd3","nwkAddr":44142,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 240","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,4096,1794,2820,57345],"outClusterList":[10,25],"clusters":{"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":255}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":1,"acPowerMultiplier":1,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"acVoltageDivisor":1,"acVoltageMultiplier":1,"activePower":0,"rmsCurrent":21,"rmsVoltage":235}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,425]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e8baf","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e8baf","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e8baf","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":0,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":0,"maxRepIntval":65000,"repChange":5,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,10],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":54,"hwVersion":1,"dateCode":"20231019","swBuildId":"1.4.3","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1704439978453,"defaultSendRequestWhen":"immediate"}

**Plug #4:**
{"id":22,"type":"Router","ieeeAddr":"0xa4c1385d0c0bedaf","nwkAddr":63571,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 240","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,4096,1794,2820,57345],"outClusterList":[10,25],"clusters":{"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":255}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":1,"acPowerMultiplier":1,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"acVoltageDivisor":1,"acVoltageMultiplier":1,"activePower":11,"rmsCurrent":174,"rmsVoltage":238}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,35]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e8baf","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e8baf","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2e8baf","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":0,"maxRepIntval":65000,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":0,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":54,"hwVersion":1,"dateCode":"20231019","swBuildId":"1.4.3","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1704440086779,"defaultSendRequestWhen":"immediate"}

Koenkk, 2024-01-05:
Can you try to reduce the max rep interval to something like 20? Then it should report every 20 secs.

gewoonivar, 2024-01-05:
That works. Have set it to 60secs for each and reporting is good. Graph line gradually moves upwards with more consumption.

Is there a way (in de UI) to copy/paste these set of reporting parameters to multiple devices, instead of configuring them one-by-one?

Koenkk, 2024-01-05:
That's currently not possible (but it's a one time thing so shouldn't be a lot of work to do it manually)

bikeymouse, 2024-01-06:
> Can you try to reduce the max rep interval to something like 20? Then it should report every 20 secs.

I can confirm that I too - after the factory reset - have everything working with the default settings, except for the automatic reporting of the currentSummDelivered.

Changing the rep interval of that to a lower value indeed does work. However, does that not actually mean reconfiguring it be polled every 20 seconds, instead of relying on (less resource intensive) use push-updates?

For others also good to know: when I unplugged the plug from the mains and plugged it in again, reporting of all attributes was broken again. Had to do a factory reset again to fix that. Might be a firmware bug of the SP240?

gewoonivar, 2024-01-06:
> I can confirm that I too - after the factory reset - have everything working with the default settings, except for the automatic reporting of the currentSummDelivered.
>
> Changing the rep interval of [...]

Agreed, that (auto) push updates would be more favourable.

On the last remark, that's what I've experienced today to. Unplugging & plugging-in and the auto reporting is not functioning anymore. Very strange.

bikeymouse, 2024-01-06:
@gewoonivar 
Yeah, seems buggy. 

Btw: I did notice in your and my configuration that the currentSummDelivered is the only attribute in the default configuration that has no "minimum reporting change" set. 

I figured that perhaps that could be the issue so I have set this to "1" (would that be 1 Watt or 1 Kwh?) but that doesn't seem to have an impact, so still relying on the polling.

[image]

gewoonivar, 2024-01-06:
> @gewoonivar Yeah, seems buggy.
> 
> Btw: I did notice in your and my configuration that the currentSummDelivered is the only attribute in the default configuration that has no "minimum reporting change" set.
> 
> I figured that perhaps that could be the issue so I have set this to "1" (would that be 1 Wh or 1 Kwh?) but that doesn't seem to have an impact, so still relying on the polling.
> 

Correct, on some plugs I changed that to 1 or 0 for testing, with no desired result. I believe empty was the default value after factory reset.

Koenkk, 2024-01-07:
Can you try sending to `zigbee2mqtt/bridge/request/device/configure_reporting` payload `{"id":"PLUG_NAME","cluster":"seMetering","attribute":"currentSummDelivered","minimum_report_interval":5,"maximum_report_interval":600,"reportable_change":[0, 5]}`? If that doesn't work try a lower value for `reportable_change`, e.g. `[0, 1]`

bikeymouse, 2024-01-07:
I tried this, but I'm not sure if the command is accepted. See no difference in behaviour. also.
Why the brackets and the space before the "5" in  "[0, 5]"? And shouldn't we use a decimal point, so use "0.01"?

Koenkk, 2024-01-07:
Because internally this is a bigint type, can you provide the debug log when sending this and your data/database.db entry after **stopping** zigbee2mqtt?

bikeymouse, 2024-01-07:
Ah, Ok thanks for explaining ;-)

Here are the relevant parts of the debug log (it actually now did report the energy, together with the other attributes, probably because of changes in those):

```
debug 2024-01-07 16:04:47: Received MQTT message on 'zigbee2slzb/bridge/request/device/configure_reporting payload' with data '{"id":"Wasmachine",
"cluster":"seMetering",
"attribute":"currentSummDelivered",
"minimum_report_interval":5,
"maximum_report_interval":1800,
"reportable_change": [0, 1]}'

debug 2024-01-07 16:06:32: Received Zigbee message from 'Wasmachine', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":37}' from endpoint 1 with groupID 0
info  2024-01-07 16:06:32: MQTT publish: topic 'zigbee2slzb/Wasmachine', payload '{"current":0.04,"energy":0.33,"last_seen":"2024-01-07T16:06:32+01:00","linkquality":105,"power":0,"power_on_behavior":"previous","state":"ON","voltage":219}'
debug 2024-01-07 16:07:08: Received Zigbee message from 'Wasmachine', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":225}' from endpoint 1 with groupID 0
info  2024-01-07 16:07:08: MQTT publish: topic 'zigbee2slzb/Wasmachine', payload '{"current":0.04,"energy":0.33,"last_seen":"2024-01-07T16:07:08+01:00","linkquality":105,"power":0,"power_on_behavior":"previous","state":"ON","voltage":225}
```

So trying to set max report interval to 1800 seconds and reportable change to 0,1 kWh.
In the front-end it however still shows 3600 and 1 kWh.

And this is the part of the database.db

```
{
  "id": 53,
  "type": "Router",
  "ieeeAddr": "0xa4c13820a37ccd9c",
  "nwkAddr": 39931,
  "manufId": 4454,
  "manufName": "innr",
  "powerSource": "Mains (single phase)",
  "modelId": "SP 240",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 266,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        4096,
        1794,
        2820,
        57345
      ],
      "outClusterList": [
        10,
        25
      ],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 1,
            "startUpOnOff": 255
          }
        },
        "haElectricalMeasurement": {
          "attributes": {
            "acPowerDivisor": 1,
            "acPowerMultiplier": 1,
            "acCurrentDivisor": 1000,
            "acCurrentMultiplier": 1,
            "acVoltageDivisor": 1,
            "acVoltageMultiplier": 1,
            "activePower": 0,
            "rmsCurrent": 37,
            "rmsVoltage": 225
          }
        },
        "seMetering": {
          "attributes": {
            "divisor": 100,
            "multiplier": 1,
            "currentSummDelivered": [
              0,
              33
            ]
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0026d101b6",
          "endpointID": 1
        },
        {
          "cluster": 2820,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0026d101b6",
          "endpointID": 1
        },
        {
          "cluster": 1794,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0026d101b6",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1,
          "manufacturerCode": null
        },
        {
          "cluster": 2820,
          "attrId": 1291,
          "minRepIntval": 30,
          "maxRepIntval": 65000,
          "repChange": 1,
          "manufacturerCode": null
        },
        {
          "cluster": 2820,
          "attrId": 1285,
          "minRepIntval": 300,
          "maxRepIntval": 65000,
          "repChange": 4,
          "manufacturerCode": null
        },
        {
          "cluster": 2820,
          "attrId": 1288,
          "minRepIntval": 300,
          "maxRepIntval": 65000,
          "repChange": 20,
          "manufacturerCode": null
        },
        {
          "cluster": 1794,
          "attrId": 0,
          "minRepIntval": 10,
          "maxRepIntval": 3600,
          "repChange": 1,
          "manufacturerCode": null
        }
      ],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 20,
  "stackVersion": 54,
  "hwVersion": 1,
  "dateCode": "20231019",
  "swBuildId": "1.4.3",
  "zclVersion": 8,
  "interviewCompleted": true,
  "meta": {
    "configured": 1324213189
  },
  "lastSeen": 1704640028125,
  "defaultSendRequestWhen": "immediate"
}
```

gewoonivar, 2024-01-07:
With another plug just (again) faced the issue of no auto updates after unplugging & plugging-in the unit. I'm able to switch the plug (on/off), but attribute updates are not pushed, i.e. manual refresh required.

Is this something for Innr, or Z2M?
As the plug is 'switchable'

Koenkk, 2024-01-08:
@bikeymouse could you also provide the herdsman debug log of this? 

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

@gewoonivar the Innr firmware is bugged, I already notified someone from Innr and they are looking into it.

gewoonivar, 2024-01-08:
Ok thanks @Koenkk . Any experience with their proces/speed of updates? Can it be distributed via Zigbee, as with other devices?

Otherwise I'll return the plugs and find another solution.

Koenkk, 2024-01-09:
> Any experience with their proces/speed of updates?

No clue, depends on Innr

> Can it be distributed via Zigbee, as with other devices?

Yes

bikeymouse, 2024-01-11:
> @bikeymouse could you also provide the herdsman debug log of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Sure! 
I'm not sure what to look for, so I sent the MQTT configure command to the SP240 and copied all logging from a couple of seconds before and after that (I do see data to the network address 39931). See attached log file.
Please let me know if you need more info.

[herdsman-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13909122/herdsman-log.txt)

Koenkk, 2024-01-13:
I don't see any logging after `Received MQTT message on 'zigbee2slzb/bridge/request/device/configure_reporting payload'`, what is logged after this?

bikeymouse, 2024-01-13:
Nothing actually, which surprised mee. So I guessed that the commnand is first processed before echoing back the command. Should I retest?

Koenkk, 2024-01-15:
The configuration is not applied because the wrong topic is used, use `zigbee2slzb/bridge/request/device/configure_reporting` (not `zigbee2slzb/bridge/request/device/configure_reporting payload`)

Peter376, 2024-03-17:
I’m not tech savvy and just installed four SP240 10 days ago. Power (watts) history is fine but sum delivered energy is very buggy. I did in fact move the plugs from one wall socket to the other to test many devices. What should I do?
Thank you.

Jeruntu, 2024-04-14:
My HA run's on a Raspberry Pi 4b 4G

Core: 2024.4.3
Supervisor: 2024.04.0
Operating System: 12.2
Frontend: 20240404.2

I've got 4 of these plugs and they all work fine. Pairing went well and all the values are reported automatically, except for the currentSumDelivered. When I set the max interval to 20 seconds, it starts polling so that works. However I would like to have push notifications. Is it still the case that doesn't work, and if so, is that because of Innr sp 240 firmware, or on the zigbee2mqtt side? I bought them for energy monitoring, so if that's not working well, I rather buy other ones.

Ocramius, 2024-04-14:
@Jeruntu I've been running ~10 plugs for a while: my experience has been that they work fine, until there's a hiccup in the coordinator (in my case, due to a reboot), and then I had to reconfigure them all.

Some never came back, but I didn't stress it either.

I would suggest checking with Innr if there's a way to upgrade their firmware: AFAIK not currently supported by Z2M?

Jeruntu, 2024-04-20:
I contacted Innr support and they mention push notifications for Energy `CurrentSumDelivered` is not supported, and not Zigbee compliant. Thier answer:

- _The SP 24x reports voltage, current, and active power to any other interested device (for example a hub/bridge) that creates a binding to the Electrical Measurement cluster on the SP 24x.
The SP 24x supports the Configure Attribute Reporting command to let the receiving device determine the intervals with which data is reported.
The SP 24x does NOT report attributes autonomously, without being asked to do so by a binding to the cluster. Tuya based smart plugs do this, always to address 0x0000, and quite possibly also the old Innr SP 120 may do this, but this is not Zigbee compliant! Address 0x0000 is usually the hub in a system but not necessarily, and in addition there may be other devices in a system than the hub on other addresses that may be interested in the attribute reports._
 
- _The “CurrentSummationDelivered” attribute of the Metering cluster is not defined as reportable by Zigbee and therefore not reported by the SP 24x.
Interested devices must regularly retrieve the value, or retrieve it on demand when the user wants to see it._

asterix778, 2024-04-20:
> I contacted Innr support and they mention push notifications for Energy `CurrentSumDelivered` is not supported, and not Zigbee compliant. Their answer:
> 
> * _The SP 24x reports voltage, current, and active power to any other interested device (for example a hub/bridge) that creates a binding to the Electrical Measurement cluster on the SP 24x.
>   The SP 24x supports the Configure Attribute Reporting command to let the receiving device determine the intervals with which data is reported.
>   The SP 24x does NOT report attributes autonomously, without being asked to do so by a binding to the cluster. Tuya based smart plugs do this, always to address 0x0000, and quite possibly also the old Innr SP 120 may do this, but this is not Zigbee compliant! Address 0x0000 is usually the hub in a system but not necessarily, and in addition there may be other devices in a system than the hub on other addresses that may be interested in the attribute reports._
> * _The “CurrentSummationDelivered” attribute of the Metering cluster is not defined as reportable by Zigbee and therefore not reported by the SP 24x.
>   Interested devices must regularly retrieve the value, or retrieve it on demand when the user wants to see it._

What I did to fix this is to make a helper in homeassist that counts the used energy so that i could put in the energy dashboard and have accurate data.

bikeymouse, 2024-04-21:
I don't understand why you would need a helper for this? Or that you think that energy reporting is not working will if you just polling which works fine.

I you set the minimum reporting interval to e.g. 300 and the maximum interval at e.g. 3600 the plug will be polled every hour for an update. You can then just the Energy entity sensor from the device in HomeAssistant to see the value and display in you dashboard. 

Of course you can set the maximum interval to a lower value if you want faster updates (but why would you?).

Peter376, 2024-05-03:
Is this issue z2mqtt specific? I use the sp240 with ZHA and currentsumdelivered works out of the box
[image]

Kurdie2023, 2024-05-18:
Hi all,

I bought 4 innr SP 240 in a box. At one of the four the energy is shown correctly. The other three show sporadic steps of it ( several hours in between) . The firmware of the plugs seem to be all the same an all are connected to ZHA with a skyconnect. I don't know how to fix it.

if anybody has an idea…

tvwerkhoven, 2024-07-05:
I have the same issue with SP240 (and SP120) plugs on ZHA, [reported here](https://github.com/zigpy/zha-device-handlers/issues/3243). @Peter376 I think my plugs did work well out of the box, but after a while (reboot, network change, not sure what triggered it) the `sp_240_summation_delivered` degrades and only updates sporadically, e.g. once per day. My other [Robb Zigbee plugs](https://www.robbshop.nl/robb-smarrt-slimme-stekker-zigbee-3680w) don't suffer from this (no affiliation).

Given this issue occurs on both ZHA and zigbee2mqtt, _and_ it doesn't occur on other hardware it seems an innr-specific bug. This seems in line with the answer @Jeruntu received from their helpdesk.

The only solutions I see is to
1. integrate power manually in HA with Riemann sum (not very elegant) OR
2. use other hardware plugs.

What's strange is that the plug **does** irregularly report an updated energy value. If we can identify and recreate whatever triggers the update, that might help.

Here's one of my SP240 plugs
[image]

Peter376, 2024-12-14:
Any Updates yet from innr? I use ZHA (home assistant) and whenever there is a power outage, the innr SP240: a. Lose sumDelivered, b. Pressing button on/off on device does switch the plug on/off but the switch in HA doesn’t change. 
After removing the device from HA and re-adding (using same name) all functionality is restored. But it’s a drag to keep having to think about it. 
Any news or help greatly appreciated!

thomasreiser, 2025-01-13:
I have the same issues. When I remove the INNR SP 240 from the outlet (power them off) and put them into again, they stop promoting energy usage correctly.

RonaldRegister, 2025-02-25:
I'm also having the issue that SP240 is no longer reporting/updating its kWh delivered. It was working for 9 months straight without any issue (don't know how many updates per day, but it was enough). Last week I had a regional power outage, and things went sour from there. Voltage and Power do regularly update.

Energy dash is now wrong with negative untracked consumption, and kWh's from INNR devices don't or rarely update. Only updates delivered kWh after rebooting HA (thus restarting Zigbee integration) or disable/enable the Zigbee device.

I've read a lot, but I am unable to get a clear answer if this is fixable with an firmware update from INNR, also have not found the precise way to update INNR firmware from ZHA.

Ocramius, 2025-02-25:
From what I saw on my end, firmware updates (currently got all my SP240 to build `20250210`, `1.9.29`) led to much more stability in reporting: had multiple power blips/outages, and the plugs continued reliably reporting.

stefan1495, 2025-02-25:
> I'm also having the issue (ON ZHA) that SP240 is no longer reporting/updating its kWh delivered. It was working for 9 months straight without any issue (don't know how many updates per day, but it was enough). Last week I had a regional power outage, and things went sour from there. Voltage and Power do regularly update.
> 
> Energy dash is now wrong with negative untracked consumption, and kWh's from INNR devices don't or rarely update. Only updates delivered kWh after rebooting HA (thus restarting Zigbee integration) or disable/enable the Zigbee device.
> 
> I've read a lot, but I am unable to get a clear answer if this is fixable with an firmware update from INNR, also have not found the precise way to update INNR firmware from ZHA.

You do have to do an OTA Update for your Plugs. Innr fixed plenty of things there and zigbee2mqtt has changed its converter to reflect those changes.

RonaldRegister, 2025-02-26:
Well thank you, three in a row; @Peter376 @stefan1495 @Ocramius !
No arguing with this, updates shall be done. Found some info on how to do it with ZHA, I'm not so keen on installing z2mqtt just for updates. 

For now, the 'Reconfigure'-option in ZHA looks to be a temporary fix. And I got back some kWh reporting. 

Info I found for ZHA updating:
[https://community.home-assistant.io/t/zha-firmware-update-process/736374](url)
[https://community.home-assistant.io/t/innr-zigbee-plugs-not-updating-energy/728244](url)

Will try and report back here for others to reference