Koenkk/zigbee2mqtt #20040: Z2M bridge-config MQTT topic not updated while running

emandtf, 2023-12-05:
### What happened?

I'm monitoring "_zigbee2mqtt - bridge - config_" MQTT topic to have a quick monitor if last time I set Debug level = "_debug_" or I left "_Permit join_" flag enabled. 
Unfortunately this MQTT topic seems not be updated after initial start.

### What did you expect to happen?

Values ​​have to reflect Z2M settings.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

Latest Conbee 2

### Adapter

Conbee 2

mundschenk-at, 2023-12-08:
Isn't that topic deprecated? The information should be available in `zigbee2mqtt/bridge/info`.

mundschenk-at, 2023-12-08:
I realized I was thinking of `zigbee2mqtt/bridge/request/config/#` which [have been deprecated according to the manual](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request). However, which topic are you actually listening to?

emandtf, 2023-12-08:
As I wrote before, the most simple: zigbee2mqtt/bridge/config

mundschenk-at, 2023-12-08:
Well, it's not mentioned (anymore?) in the documentation and set in `legacyBridge.ts`, so I'd say it's deprecated. You should be able to get all that information from `zigbee2mqtt/bridge/info` in JSON format (`config` key).

closed 2024-01-02 by emandtf