Koenkk/zigbee2mqtt #20044: New Device (Bosch Outdoor Siren #6313) BSIR-EZ Not Pairing

johnsethwatkins, 2023-12-05:
### What happened?

I am using the Home Assistant add-on and have updated to 1.34.0-1 which equates to z2m 1.34.0.
As it is now a supported device I have bought the Bosch Outdoor Siren. It's new out of the box.
The device will not pair, it is marked as Unsupported in the front end and the log has "failed to interview" errors.

### What did you expect to happen?

Pairing requested, expected success

### How to reproduce it (minimal and precise)

Enable pairing, switch on device. Interview starts but fails.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbeee 3.0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13562514/log.txt)

lieblinger, 2023-12-07:
Same here! 

Trying to pair the siren in smart mode and it's popping up in Z2M (as supported) and after a second it leaves the network. This is happening again and again.

danieledwardgeorgehitchcock, 2023-12-07:
You'll need to add the install code from the back of the device into Zigbee2MQTT before it will pair

lieblinger, 2023-12-07:
Had to add the installer code a couple of times and it works now. 

I'm not sure if this is related to the siren, but after enabling the siren Z2M crashed a couple of times. Devices were not pingable and the docker container was not reachable until I restarted it. 

After the siren got added everything works normaly.

danieledwardgeorgehitchcock, 2023-12-07:
Probably best you report the error as a separate issue in order to not cross pollinate problems

johnsethwatkins, 2023-12-07:
The install code information from @danieledwardgeorgehitchcock has worked. Many thanks for this. I have already submitted a doc update to include this information on the device page.

closed 2023-12-07 by johnsethwatkins