Koenkk/zigbee2mqtt #20045: VIMAR devices don't connect to the network after a blackout

LeoeLeoeL, 2023-12-05:
### What happened?

After a simulated blackout of 15 minutes, Vimar devices didn't connect to the network anymore.
I have:
2 https://www.zigbee2mqtt.io/devices/14592.0.html#vimar-14592.0
5 https://www.zigbee2mqtt.io/devices/14594.html#vimar-14594
1 https://www.zigbee2mqtt.io/devices/14595.0.html#vimar-14595.0

Among 8 devices, only the one that is less than 2 meter from the coordinator connected again.

During the blackout only the "light line" has been affected. Plugs remained ON, Home Assistant, Zigbee2MQTT and the broker remained ON too.

### What did you expect to happen?

I expect everything works as before.

### How to reproduce it (minimal and precise)

Power-off the line at home for 15 minutes.
Then power on the line.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

 20230507

### Adapter

Sonoff Dongle P zStack3x0

LeoeLeoeL, 2024-01-31:
I think Z2M has a problem with Vimar devices (Deconz didn't have).
[image]
Network is not reliable with them.
Sometimes Z2M "sees" them offline but they don't and sometimes they really are off the network.
Devices are very close each other so it's not a coverage problem.

LeoeLeoeL, 2024-02-06:
Today, I added a device in the mesh.
Half a hour without current.
When I gave current all devices connected to zigbee2MQTT except VIMAR devices. Still they are offline after one hour.
Somebody could think about a firmware problem but under Deconz this problem was absent.

LeoeLeoeL, 2024-06-22:
I hope new releases solve the issue. We arrived at 1.38 but Vimar devices still are unavailable after a blackout.

saveriol, 2024-09-01:
I also have this issue with Z2M 1.38 and Vimar devices.

saveriol, 2024-09-01:
Debug logs when I try to turn on one of the devices offline:

`[2024-08-31 11:22:22] info: 	z2m: Successfully changed options
[2024-08-31 11:22:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"2q3jc-1"}'
[2024-08-31 11:22:28] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Luce terrazzo sala/set' with data '{"state":"ON"}'
[2024-08-31 11:22:28] debug: 	z2m: Publishing 'set' 'state' to 'Luce terrazzo sala'
[2024-08-31 11:22:28] debug: 	zh:controller:endpoint: ZCL command 0xf4ce36c553d84b7a/10 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-08-31 11:22:28] debug: 	zh:ember: ~~~> [ZCL to=2131 apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":10,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":0,"disableDefaultResponse":false,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":222,"commandIdentifier":1}]
[2024-08-31 11:22:28] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=251 Len=24]
[2024-08-31 11:22:28] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=1](ackRx=3)
[2024-08-31 11:22:28] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-31 11:22:28] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=3 frmTx=4)
[2024-08-31 11:22:28] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=1](frmRx=1) Added to rxQueue
[2024-08-31 11:22:28] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=4)
[2024-08-31 11:22:28] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=251 Len=7]
[2024-08-31 11:22:28] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=130 messageTag=15 status=OK]
[2024-08-31 11:22:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-31 11:22:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
[2024-08-31 11:22:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=2](frmRx=2) Added to rxQueue
[2024-08-31 11:22:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=4)
[2024-08-31 11:22:33] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=251 Len=22]
[2024-08-31 11:22:33] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=ZIGBEE_DELIVERY_FAILED], [type=DIRECT], [indexOrDestination=2131], [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":10,"options":4160,"groupId":0,"sequence":130}], [messageTag=15]
[2024-08-31 11:22:33] debug: 	zh:controller:endpoint: Error: ZCL command 0xf4ce36c553d84b7a/10 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":10,"options":4160,"groupId":0,"sequence":130})
[2024-08-31 11:22:33] error: 	z2m: Publish 'set' 'state' to 'Luce terrazzo sala' failed: 'Error: ZCL command 0xf4ce36c553d84b7a/10 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":10,"options":4160,"groupId":0,"sequence":130})'
[2024-08-31 11:22:33] debug: 	z2m: Error: ZCL command 0xf4ce36c553d84b7a/10 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":10,"options":4160,"groupId":0,"sequence":130})
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:98:31)
    at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:454:42)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5016:14)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:770:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45)
    at processImmediate (node:internal/timers:476:21)
[2024-08-31 11:22:36] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Luce terrazzo sala/set' with data '{"state":"OFF"}'
[2024-08-31 11:22:36] debug: 	z2m: Publishing 'set' 'state' to 'Luce terrazzo sala'
[2024-08-31 11:22:36] debug: 	zh:controller:endpoint: ZCL command 0xf4ce36c553d84b7a/10 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint" [... cut ...]
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:770:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45)
    at processImmediate (node:internal/timers:476:21)
[2024-08-31 11:22:44] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"warning"}},"transaction":"2q3jc-2"}'
`

saveriol, 2024-09-01:
When manually turning on/off one of the offline devices, I see this on debug log:

`[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=1](frmRx=1) Added to rxQueue
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=6)
[2024-09-01 14:46:37] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=98:"undefined" Seq=189 Len=13]
[2024-09-01 14:46:37] debug:    zh:ember:ezsp: <=x= Ignored unused/unknown [CBFRAME: ID=98:"undefined" Seq=189 Len=13]
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=2](frmRx=2) Added to rxQueue
[2024-09-01 14:46:37] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=6)
[2024-09-01 14:46:37] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=189 Len=31]
[2024-09-01 14:46:37] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":10,"destinationEndpoint":1,"options":3392,"groupId":0,"sequence":141}], [packetInfo:{"senderShortId":11746,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":124,"lastHopRssi":-69,"lastHopTimestamp":0}], [messageContents=08bf0a00001001]
[2024-09-01 14:46:37] debug:    zh:controller: Data is from unknown device with address '11746', skipping...
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=3](frmRx=3) Added to rxQueue
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=6)
[2024-09-01 14:46:40] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=98:"undefined" Seq=189 Len=13]
[2024-09-01 14:46:40] debug:    zh:ember:ezsp: <=x= Ignored unused/unknown [CBFRAME: ID=98:"undefined" Seq=189 Len=13]
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=4](frmRx=4) Added to rxQueue
[2024-09-01 14:46:40] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=6)
[2024-09-01 14:46:40] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=189 Len=31]
[2024-09-01 14:46:40] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":10,"destinationEndpoint":1,"options":3392,"groupId":0,"sequence":142}], [packetInfo:{"senderShortId":11746,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":128,"lastHopRssi":-68,"lastHopTimestamp":0}], [messageContents=08c00a00001000]
[2024-09-01 14:46:40] debug:    zh:controller: Data is from unknown device with address '11746', skipping...
`

please note:

`Data is from unknown device with address '11746', skipping...`

this device on Z2M dashboard has this address:

Network address
0xECF1 / 60657

LeoeLeoeL, 2024-09-01:
What coordinator do you have?

saveriol, 2024-09-01:
> What coordinator do you have?

Home Assistant Yellow and its built in skyconnect chip

LeoeLeoeL, 2024-09-02:
Z2M 1.40.0 doesn't solve the issue

saveriol, 2024-09-02:
@LeoeLeoeL what driver are you using?

LeoeLeoeL, 2024-09-03:
The coordinator?
Sonoff Dongle P with FW 20230922 because the latest official doesn't work with Vimar devices. They go offline.

LeoeLeoeL, 2024-09-04:
Do you know a faster way to set Vimar device in pairing mode than what is described in the manual?

saveriol, 2024-09-05:
@LeoeLeoeL I think I found what's the issue with Z2M and Vimar devices.
That's my theory:

- When you power off the devices, a lot of Zigbee routers go offline
- When you power on the devices again, for some reason I will investigate with Vimar, some devices decide to change their network address
- When devices change their network address, they emit a "Device announcement" frame, which is normally captured by the coordinator and Z2M updates the network address of the device with the new one.
- Since a lot of routers are offline or are starting up, that device announcement isn't correctly captured by Z2M.

A fix without need of repairing the device is updating the device network address manually in Z2M. This, in my case, restored the connectivity of the devices.

Procedure:

- Enable and keep enabled Z2M advanced device availability (Gear icon -> settings -> availability -> select Availability Advanced)
- After powering on the devices, wait to see which one are offline in the "devices" page of Z2M. It takes 10 minutes with the default availability settings
- If there is some devices offline, enable debug logs (Logs -> logging level debug)
- go on the logs' directory, i.e.: `/root/homeassistant/zigbee2mqtt/log/2024-09-03.12-27-45` (replace with most current directory)
- tail and grep for unknown devices: `tail -f log.log |grep "Data is from unknown device with address"`
- go on every device offline and manually turn on/off the switches. You will see in the logs something like: `Data is from unknown device with address '11746', skipping...`
- do this for every device and take note of the number seen (i.e. 11746)
- go on the device page of the devices offline in Z2M and take note of the old network addresses in decimal, i.e. `Network address 0x872F / 34607`
- after you took notes of every device offline, turn off Z2M
- go to Z2M config directory, i.e. `/root/homeassistant/zigbee2mqtt/`
- backup the database `cp database.db database.db.backup.20240905`
- for each device, replace the old network address with the new one. You can do it manually or via a command like `sed --in-place 's/\"nwkAddr\":34607/\"nwkAddr\":11746/' database.db` where 34607 is the old network address and 11746 is the new one
- When you replaced everything, restart Z2M

I know it seems complicated but in reality is less tedious than repairing Vimar devices.
I will investigate if there is a mode to avoid changing addresses so frequently contacting Vimar support and also if there is a solution to do a fix via Z2M (though I think this is more of a Vimar and zigbee issue rather than Z2M)

LeoeLeoeL, 2024-09-05:
If you need to set in pairing mode Vimar devices there is another annoying procedure.
You have to flash BT firmware and then zigbee again. You gain you don't risk to lose some other devices after a power off.
Regarding the problem written in this thread, I have many Vimar devices connected via Deconz (due to this issue) and they don't have any problem after a blackout. So I think the problem  could be how Z2M integrated Vimar devices.
At last... the vimar switch (the yellow one) is offline since 48 hours but it's till seen
[image]

Why?

saveriol, 2024-09-05:
>If you need to set in pairing mode Vimar devices there is another annoying procedure.

Yes, I also used that method but I find manually changing the network address easier.

>So I think the problem could be how Z2M integrated Vimar devices.

Maybe Deconz is able to query the end devices for the updated network address; this is not implemented in Z2M. So maybe yes, Deconz is better for this specific issue. But Vimar devices shouldn't change network address on power off. That's a nonsense.

Nerivec, 2024-09-11:
@saveriol Can you confirm the behavior for ember is still present after 1.39.1? If it is, can you check your logs, see if you find `Network address '0x` in there, after you know a device changed address?

saveriol, 2024-09-11:
@Nerivec I can confirm. I'm currently in:

`Zigbee2MQTT version
1.40.1 commit: unknown
Coordinator type
EmberZNet
Coordinator revision
7.4.3 [GA]
Coordinator IEEE Address
0xf082c0fffe63e9b2
Frontend version
0.7.4
zigbee-herdsman-converters version
20.12.1
zigbee-herdsman version
0.57.3`

The logs are the same as above.
i can't find `Network address '0x`

saveriol, 2024-09-22:
As mentioned in https://github.com/Koenkk/zigbee-herdsman/pull/1183 this should be fixed in version 2.0.0

Nerivec, 2024-09-22:
As mentioned in the PR, the limitation is currently that the device needs to talk to Z2M to trigger the update. But as soon as the device reports "something", it should update automatically, which means, as a last resort, triggering the device (physical button/switch), should allow to force it.

saveriol, 2024-09-22:
@Nerivec that's obviously an huge step forward! I'm looking forward to test it and report back. Maybe, could be useful to be able to trigger manually the discover, I'm thinking about being not at home and not able to manually flip the switches. Maybe (excuse me if I'm telling an heresy) as part of the interview process especially if the device is marked offline?

saveriol, 2024-09-23:
Switching to dev branch, I can confirm that this works and a device which was offline for two days came back online:

````
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=115 Len=31]
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":10,"destinationEndpoint":1,"options":3392,"groupId":0,"sequence":111}], [packetInfo:{
"senderShortId":63157,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":156,"lastHopRssi":-61,"lastHopTimestamp":0}], [messageContents=08660a00001000]
**[2024-09-23 14:07:36] debug:    zh:controller: Trying to identify unknown device with address '63157'**
[2024-09-23 14:07:36] debug:    zh:ember: ~~~> [ZDO IEEE_ADDRESS_REQUEST UNICAST to=0xFFFFFFFFFFFFFFFF:63157 messageTag=11 payload=0bb5f60000]
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=116 Len=26]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=4 frmRx=6](ackRx=4)
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=4 frmTx=5)
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=6](frmRx=6) Added to rxQueue
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=5)
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=116 Len=7]
[2024-09-23 14:07:36] debug:    zh:ember: ~~~> [SENT ZDO UNICAST messageTag=11 apsSequence=180 status=OK]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=5 frmTx=5)
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=7](frmRx=7) Added to rxQueue
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=5)
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=116 Len=22]
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=63157], [apsFrame={"profileId":0,"clusterId":1,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":180}], [messageTag=11]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=5 frmTx=5)
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=0](frmRx=0) Added to rxQueue
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=5)
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=98:"undefined" Seq=116 Len=13]
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: <=x= Ignored unused/unknown [CBFRAME: ID=98:"undefined" Seq=116 Len=13]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=5 frmTx=5)
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=1](frmRx=1) Added to rxQueue
[2024-09-23 14:07:36] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=5)
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=116 Len=36]
[2024-09-23 14:07:36] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":0,"clusterId":32769,"sourceEndpoint":0,"destinationEndpoint":0,"options":3392,"groupId":0,"sequence":112}], [packetInfo:{"senderShortId":63157,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":156,"lastHopRssi":-61,"lastHopTimestamp":0}], [messageContents=0b003cea46f5e636cef4b5f6]
[2024-09-23 14:07:36] debug:    zh:controller: Received ZDO response: clusterId=IEEE_ADDRESS_RESPONSE, status=SUCCESS, payload={"eui64":"0xf4ce36e6f546ea3c","nwkAddress":63157,"startIndex":0,"assocDevList":[]}
[2024-09-23 14:07:36] debug:    zh:controller: IEEE address from '0xf4ce36e6f546ea3c:63157'
**[2024-09-23 14:07:36] debug:    z2m: Device 'Luce studio' reconnected**

````

LeoeLeoeL, 2024-09-23:
Glad to read that.
When the new release will be available I'll test the window covering device and the dimmer switch too.

saveriol, 2024-09-23:
@LeoeLeoeL this should work for every device that changes its network address and the event is not received by zigbee2mqtt.

LeoeLeoeL, 2024-09-23:
Those devices went sometimes offline even without a blackout (you can see above the post of January 31th) so that they could have a different behaviour).

saveriol, 2024-09-23:
I experienced changes of network address even without powering off the devices, so it can be... you need to test. As for the adapter, this fix is independent of the adapter

saveriol, 2024-10-02:
@LeoeLeoeL the fix is included in Z2M release 1.40.2 [Latest](https://github.com/Koenkk/zigbee2mqtt/releases/latest) released a few hours ago.

closed 2024-10-02 by Nerivec

LeoeLeoeL, 2024-10-02:
> @LeoeLeoeL the fix is included in Z2M release [1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2) released a few hours ago.

Tks for confirmation.
I began to move the first covering device to Z2M again.

LeoeLeoeL, 2024-10-20:
Do you have a DimmerSwitch_v1.0 by Vimar?
Rebooting Z2M (after a freeze of the application), the device didn't go online.
I suspect the integration for that device is not so OK.

LeoeLeoeL, 2024-10-20:
Anyway, to fix the Vimar dimmer switch, I powered off the line for 10 seconds and then powered on.
Outcome: Dimmer switch is online now but I lost 2 Vimar covering devices.
Not a good result!

saveriol, 2024-10-20:
@LeoeLeoeL you need to turn on debug logs and upload them if you want us to have a look at what's happening in your network...

LeoeLeoeL, 2024-10-21:
Enabled debug.
Anyway in the mean time, I restarted Z2M and performed another powercycle.
Now it's OK (until the next peoblem :-) )

LeoeLeoeL, 2024-11-07:
@saveriol Are you able to contact Vimar for this strange behavior of the dimmer?
Each time I press the "interview" (blue square) or "Reconfigure" (orange square) the device changes its Network address.
It's the only device that makes that. I tried with other Vimar devices (5 shutters and 2 switches) and they don't do that.
Anyway, the dimmer changes the network address even alone often,

saveriol, 2024-11-07:
I tried to contact them, they said they passed the suggestion to the technical department, but I suggest you also open a case here: https://www.vimar.com/it/it/contact/form/request?type=technic-support

LeoeLeoeL, 2024-11-07:
I wrote.

LeoeLeoeL, 2025-03-09:
@saveriol 
I just noted Vimar released FW-updates for all devices:
Switch: 2.04 -> 2.10
Cover: 2.03 -> 2.08
Dimmer: 1.00 -> 1.15
I cannot find the changelog
Do you know if they fix the problem written in this thread?

saveriol, 2025-03-10:
I don't know, I still need to install the update. No news from Vimar from my side.
After update, they still have [this behaviour](https://github.com/Koenkk/zigbee2mqtt/issues/20045#issuecomment-2462621413)?

LeoeLeoeL, 2025-03-10:
Don't update the dimmer!!!
https://github.com/Koenkk/zigbee2mqtt/issues/26676

The behaviour is still present for the farthest shutter installed from the coordinator (even if there are many routers normally connected to it).

Anyway, previous firmwares reported a date of 2020. The "new" ones are 20 january 2023. Why so much time before releasing them?