Koenkk/zigbee2mqtt #20049: TS0601 erroneous values (spikes)

remezen, 2023-12-05:
### What happened?

TS0601 climate reports 1 million degrees temperature spikes
[image]

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20220219

### Adapter

SONOFF ZB Dongle-P

KameDomotics, 2023-12-06:
I also have some problems with the latest Z2m update.

remezen, 2023-12-07:
Maybe give users an option to set limits for values in configuration? I see there's a lot of similar cases with different devices.

closed 2024-07-05 by github-actions[bot]