Koenkk/zigbee2mqtt #20052: TuYa TH01Z (supported) does not start interview, nothing happens

dierochade, 2023-12-05:
### What happened?

I want to pair this device: 

https://www.zigbee2mqtt.io/devices/TH01Z.html It is listed as supported.
It is also working for other people ( https://smarthomescene.com/reviews/zigbee-temperature-humidity-sensor-with-a-screen-sz-t04-review/) and  a review on the sellers site shows a screenshot of device info on z2m (https://de.aliexpress.com/item/1005005097691633.html).

I tried to pair as described (long press on button, till animation of  Wifi icon starts) but just nothing happens at all. I cannot see anything useful in the logs either.

I tried

- many times
- very close to coordinator
- clicking every second or so to prevent sleep
- gobal join permission or just for coordintor

I got 2 of the devices with the same issue, so I think it is not just a defective unit.

I got no idea whats the problem.

### What did you expect to happen?

Start Interview, leading to pairing or unsopported

### Zigbee2MQTT version

 1.34.0 c

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP v12 (Nabu casa sky connect)

dierochade, 2023-12-05:
[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/13565087/log.1.txt)
[image]

dierochade, 2023-12-21:
Just worked now, can’t reproduce.

closed 2023-12-21 by dierochade

Stimmenhotel, 2024-01-04:
Hey, since I am having the same issue, what have you done before it worked again? Just waited? Started Interview again?

dierochade, 2024-01-04:
Probably I had reached the device limit of the coordinator (11 for skyconnect?). The device only pairs with the coordinator for me, so it just did nothing then.