Koenkk/zigbee2mqtt #20053: Cannot change operation_mode on Aqara Switches in dev branch

pusheax0, 2023-12-05:
### What happened?

Aqara opple switches do not switch into decoupled mode on dev branch

### What did you expect to happen?

switches to work

### How to reproduce it (minimal and precise)

add aqara opple switch

### Zigbee2MQTT version

1.34.0-dev (commit #299b37a7)

### Adapter firmware version

Latest

### Adapter

SONOFF ZBDongle-E and SONOFF ZBDongle-P 

### Debug log

[image]
[database.db.json](https://github.com/Koenkk/zigbee2mqtt/files/13565182/database.db.json)
[image]
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13565183/log.txt)
[image]
[image]

pusheax0, 2023-12-05:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/converters/toZigbee.ts#L2503

pusheax0, 2023-12-05:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6629

pusheax0, 2023-12-05:
@Koenkk please look again for ``` xiaomi_switch_operation_mode_opple: {```

Koenkk, 2023-12-05:
Could you provide the debug log when setting it fails?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

pusheax0, 2023-12-05:
```info  2023-12-05 14:52:56: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-12-05.14-52-56' filename: log.txt
info  2023-12-05 14:52:56: Starting Zigbee2MQTT version 1.34.0-dev (commit #cb74db47)
info  2023-12-05 14:52:56: Starting zigbee-herdsman (0.25.3)
info  2023-12-05 14:53:01: zigbee-herdsman started (reset)
info  2023-12-05 14:53:01: Coordinator firmware version: '{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"3","product":12,"revision":"7.3.1.0 build 176"},"type":"EZSP v12"}'
info  2023-12-05 14:53:01: Currently 0 devices are joined:
info  2023-12-05 14:53:01: Zigbee: disabling joining new devices.
info  2023-12-05 14:53:01: Connecting to MQTT server at mqtt://localhost
info  2023-12-05 14:53:01: Connected to MQTT server
info  2023-12-05 14:53:01: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2023-12-05 14:53:01: Started frontend on port 8080
info  2023-12-05 14:53:01: Zigbee2MQTT started!
info  2023-12-05 14:53:47: Successfully changed options
info  2023-12-05 14:53:47: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"1o984-1"}'
debug 2023-12-05 14:53:58: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"permit_join":true},"transaction":"1o984-2"}'
info  2023-12-05 14:53:58: Zigbee: allowing new devices to join.
debug 2023-12-05 14:53:58: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,3,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
info  2023-12-05 14:53:59: Successfully changed options
info  2023-12-05 14:53:59: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"1o984-2"}'
info  2023-12-05 14:54:42: Device '0x54ef4410009905c4' joined
info  2023-12-05 14:54:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410009905c4","ieee_address":"0x54ef4410009905c4"},"type":"device_joined"}'
info  2023-12-05 14:54:42: Starting interview of '0x54ef4410009905c4'
info  2023-12-05 14:54:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410009905c4","ieee_address":"0x54ef4410009905c4","status":"started"},"type":"device_interview"}'
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.switch.b2naus01"}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"appVersion":31}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{"dateCode":"12-28-2021"}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
debug 2023-12-05 14:54:42: Received Zigbee message from '0x54ef4410009905c4', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
debug 2023-12-05 14:54:42: Skipping message, definition is undefined and still interviewing
info  2023-12-05 14:54:42: Successfully interviewed '0x54ef4410009905c4', device has successfully been paired
info  2023-12-05 14:54:42: Device '0x54ef4410009905c4' is supported, identified as: Xiaomi Aqara smart wall switch (neutral, double rocker) (WS-USC04)
[... cut ...]
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:249:52)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:141:27)
    at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:130:27)
    at WebSocket.emit (node:events:517:28)
    at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1192:20)
    at Receiver.emit (node:events:517:28)```

Koenkk, 2023-12-06:
Found the issue, fixed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-06 by Koenkk