Koenkk/zigbee2mqtt #20054: No OTA  and no router aqara DLKZMK11LM

checho5613, 2023-12-05:
### What happened?

aqara DLKZMK11LM device has no router and no ota updates
[image]
[image]

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

zStack3x0 sonoff dongle p

thomdall, 2024-02-05:
Similar problem, I have 5 DLKZMK11LM, 1 of them showing router, the other 3 are showing as end device. Not sure what happened. 
[image]

checho5613, 2024-02-05:
the new zigbee2mqtt update solved the ota updates problem, but no device is a router and I have 5

Gmarapet, 2024-02-08:
I have the same issue. Have anyone solved the problem?

fox1047, 2024-07-07:
I have this problem too. Is any solution been found?

fox1047, 2024-07-07:
After some trial and error I was able to pair both my relays as routers. 
1. I force removed relay from z2m, 
2. Started pairing on relay for it to forget my zigbee network
3. Removed power from relay
4. Restarted Z2M
5. Permitted joining on coordinator only
6. Connected relay to power and started pairing
I don't know which of those steps might be unnecessary, but at least it worked.

spali, 2024-12-03:
> After some trial and error I was able to pair both my relays as routers.
> 1. I force removed relay from z2m,
> 2. Started pairing on relay for it to forget my zigbee network
> 3. Removed power from [...]

More or less the same worked for me too.
Skipped (3) and waited before (5) that the relay gave up to pair.

closed 2025-02-09 by github-actions[bot]