Koenkk/zigbee2mqtt #20056: [New device support]: RSH-HS06 _TZ3000_zl1kmjqx TUYA device - temperature and humidity sensor

panteLx, 2023-12-05:
### Link

https://de.aliexpress.com/item/1005005454107609.html?spm=a2g0o.order_list.order_list_main.11.2f6b18029xLReh&gatewayAdapt=glo2deu

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0xa4c13887c99c0115","nwkAddr":54813,"manufId":4417,"manufName":"_TZ3000_zl1kmjqx","powerSource":"Battery","modelId":"","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,1029],"outClusterList":[25],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2443}},"genBasic":{"attributes":{"modelId":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":211,"stackVersion":2,"hwVersion":1,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701801482170,"defaultSendRequestWhen":"immediate"}

### Comments

I really dont undersand how to create a new external converter. 
When I try to add the new device it won't even show the zigbee model. When I trust the package the model is RSH-HS06.

jmpmn-jr, 2023-12-07:
Hi, I have a similar item which seemed to be not supported:

https://www.aliexpress.com/item/1005005454153400.html?spm=a2g0o.order_list.order_list_main.11.21ef1802thGn8p

It seems that the only difference is that the modelId is empty. As first step I created an external converter:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    TS0201_battery: {
        cluster: 'genPowerCfg',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            // https://github.com/Koenkk/zigbee2mqtt/issues/11470
            if (msg.data.batteryPercentageRemaining == 200 && msg.data.batteryVoltage < 30) return;
            return fz.battery.convert(model, msg, publish, options, meta);
        },
    },
};

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: '',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_zl1kmjqx',
        },
    ],
    model: 'IH-K009',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor',
    //fromZigbee: [tuya.fz.datapoints],
    fromZigbee: [fzLocal.TS0201_battery, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity(), e.battery_voltage()],
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
`
I suppose this can be added in typescript as well, but what about the empty modelId? If it is ok, then a PR should be straightforward.

panteLx, 2023-12-07:
I'll get a FzLocal is undefined error. Is the external converter working for you? 
Is the converter working without the Tuya datapoints ?

I tried to create a external converter as well but the exposes won't work at all.

jmpmn-jr, 2023-12-07:
I use it as external converter but renamed to tz3000.js which could not upload and it works fine for me. 
But the interview for my device gives slightly different results as you can see here:

`{"id":21,"type":"EndDevice","ieeeAddr":"0xa4c138af32f0403a","nwkAddr":40013,"manufId":4417,"manufName":"_TZ3000_zl1kmjqx","powerSource":"Battery","modelId":"","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,1029],"outClusterList":[25],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2319}},"msRelativeHumidity":{"attributes":{"measuredValue":3759}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"manuSpecificTuya_2":{"attributes":{"53256":5}},"genBasic":{"attributes":{"modelId":"","manufacturerName":"_TZ3000_zl1kmjqx","powerSource":3,"zclVersion":3,"appVersion":211,"stackVersion":2,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":211,"stackVersion":2,"hwVersion":1,"zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1701921558184,"defaultSendRequestWhen":"immediate"}`

panteLx, 2023-12-07:
Converter and documentation added. Device should be supported in the next release

Ostromogilski, 2024-01-17:
Worked for me as well! Thank you!

FlorinCiuban, 2024-03-25:
_TZ3000_zl1kmjqx TUYA device - temperature and humidity sensor_, Show only the temperature.
[image]

FlorinCiuban, 2024-03-25:
> Converter and documentation added. Device should be supported in the next release
> 
> #20094 added to dev branch

Show only the temperature 😥
[image]

djcrawleravp, 2024-04-09:
On my case It works for a couple of minutes then stops reporting (even on Z2M still "online") after repairing the device starts reporting again but the problem appear again after a couple of minutes, I have another similar sensor "IHK-009" that updates its "last seen" every 10 minutes but this behaves differently

infinitytec, 2024-06-08:
> Show only the temperature 😥 [...]

Same issue here. I have some similar ones that are _TZ3000_dowj6gyi that work properly.