Koenkk/zigbee2mqtt #20057: incomplete device info for Bosch BTH-RM230Z

aaamoeder, 2023-12-05:
### What happened?

I tried pairing the device but got constant interview errors

the device [info page](https://www.zigbee2mqtt.io/devices/BTH-RM230Z.html) is incomplete since it mentions nowhere there is an install code needed to pair the device succesfully.

### What did you expect to happen?

since the device is mentioned as fully supported I expected it to be paired succesfully without any install code (never even heard of it before)

### How to reproduce it (minimal and precise)

attempt to pair without install code

### Zigbee2MQTT version

1.34.0 

### Adapter firmware version

20221226

### Adapter

cc2652 slaesh

### Debug log

failed interview log data.. irrelevant info...

h1nnak, 2024-01-04:
Hey aaamoeder, with the last update it is solved i think: [19377](https://github.com/Koenkk/zigbee2mqtt/issues/19377).
I paired the thermostat without problems.

metallmanu, 2024-01-09:
poorly the device is not completely supported. The system_mode cool is missing and also to set the valves either to no (normaly open) or nc (normaly closed)

metallmanu, 2024-07-08:
set the valves either to no (normaly open) or nc (normaly closed) is still missing

closed 2025-05-11 by github-actions[bot]

DerDreschner, 2025-11-02:
This is resolved with [version 2.6.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.3) which was released yesterday. 🥳