Koenkk/zigbee2mqtt #20058: WXKG11LM Stopps working after initial successful setup

linuxstony, 2023-12-05:
### What happened?

I added the button but the double click feature was missing in homeassistant somehow. So i removed it and the interview is successfull. Pushing the litte button on top produces the requested payload:

debug 2023-12-05 20:24:53Received Zigbee message from '0x00158d0009e05ea2', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3055,"10":33519,"3":28,"4":5032,"5":8,"6":[0,0]},"modelId":"lumi.remote.b1acn01"}' from endpoint 1 with groupID null
debug 2023-12-05 20:24:53WXKG11LM: unknown key 6 with value 0,0
debug 2023-12-05 20:24:53WXKG11LM: Processed data into payload {"voltage":3055,"battery":100,"device_temperature":28,"power_outage_count":7}
debug 2023-12-05 20:24:53WXKG11LM: Processed data into payload {"voltage":3055,"battery":100,"device_temperature":28,"power_outage_count":7}
info 2023-12-05 20:24:53MQTT publish: topic 'zigbee2mqtt/0x00158d0009e05ea2', payload '{"action":null,"battery":100,"click":null,"device_temperature":28,"linkquality":255,"power_outage_count":7,"voltage":3055}'

But the main button does not work at all anymore. No report in the logs nor the action in Zigbee2MQTT

### What did you expect to happen?

It should register the button pressed :D

### Zigbee2MQTT version

    1.33.1 commit: 7e63039

### Adapter firmware version

 0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

No entries are added to the button pressed - I assume the debut log will not help?

lesleypersyn, 2023-12-14:
Same here. Did last days HA and zigbee2mqtt update but uses the button not datly.
Action is 'null'

vaderag, 2024-02-12:
Slightly different but potentially related issue here
At some point in the past couple of releases I have found that my WXKG11LM still works on double / triple press, but the single press is now intermittently registering as a null / blank value. 

It does sometimes capture the single press, but additionally it seems to capture the single press sometimes _incredibly_ late, meaning that over the past two nights, the light it controls has come on randomly overnight, after registering a single click a few hours after it was sent

closed 2024-09-11 by github-actions[bot]