Koenkk/zigbee2mqtt #20064: [New device support]: 10 Key CCT Zigbee Remote Control R2(WZS)

neruch, 2023-12-06:
### Link

https://iskydance.com/index.php?c=product_show&a=index&id=2068

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0xe0798dfffec2f117","nwkAddr":38517,"manufId":4098,"manufName":"_TZ3000_xa9g7rxs","powerSource":"Battery","modelId":"TS1002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,1,3,4,4096,57345],"outClusterList":[25,10,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS1002","manufacturerName":"_TZ3000_xa9g7rxs","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701795059379,"defaultSendRequestWhen":"immediate"}

### Comments

Remote not supported zigbee2mqtt (sorry, I do not know English well).

closed 2024-07-04 by github-actions[bot]

masarovofis, 2024-12-24:
Кто то решил проблему с Внешний конвертером? удалось добавить устройство?

neruch, 2024-12-24:
> Кто то решил проблему с Внешний конвертером? удалось добавить устройство?

Тут русский язык не приветствуется. Сейчас бухтеть начнут.

leonwebcreators, 2025-01-13:
i'm interested in this too...

Sergei7885, 2025-01-16:
Tell me, is there a chance to get support for this remote control in Z2M? I have two remote controls and I can't use them, what do I need to add to the list?