Koenkk/zigbee2mqtt #20066: [New device support]: Ltech TY-150-24-G2Z2

VyckaLTU, 2023-12-06:
### Link

http://www.ltechonline.com/html/en/products/Intelligence/zigbee/TY-150-24-G2Z2.html

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x9035eafffed5520f","nwkAddr":5775,"manufId":4098,"manufName":"_TZE200_na98lvjp","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0007","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_na98lvjp","stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1701846699348,"defaultSendRequestWhen":"immediate"}

### Comments

This LTECH TY-150-24-G2Z2 driver is same as LTECH TY-75-24-G2Z2 which is already supported in Z2M. Difference between them is only power. One is 75W other 150W. And sadly 75W drivers integration is not working with 150W driver. It shows as unsupported device. 

Here is 75W model - https://www.zigbee2mqtt.io/devices/TY-75-24-G2Z2.html

User @bummzulu in old post said he edited external converter to make it work. But I tried with no luck. In external converter I added his modified external converter.

Thank you.

[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '**_TZE200_na98lvjp**'}
    ],
    model: 'WZ5_cct',
    vendor: 'Skydance',
    description: 'Zigbee & RF 5 in 1 LED controller (CCT mode)',
    fromZigbee: [legacy.fz.tuya_light_wz5],
    toZigbee: [legacy.tz.tuya_dimmer_state, legacy.tz.tuya_light_wz5],
    exposes: [
        e.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp', ea.STATE_SET),
    ],
};

module.exports = definition;
```

bummzulu, 2023-12-06:
Make sure you strictly follow the instruction on how to add an unsupported device (like change of configuration.yaml and adherence to correct file naming). In my case: WZ5_cct.js

Here my modified external convertert 
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_na98lvjp'}
    ],
    model: 'TY-75-24-G2Z2',
    vendor: 'Ltech',
    description: '150W 24V Zigbee CV tunable white LED driver',
    fromZigbee: [legacy.fz.tuya_light_wz5],
    toZigbee: [legacy.tz.tuya_dimmer_state, legacy.tz.tuya_light_wz5],
//    exposes: [
//        e.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),e.light().withColorTemp().setAccess('color_temp', ea.STATE_SET),
//    ],
//};
//
//module.exports = definition;

    exposes: [
        e.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp', ea.STATE_SET),
    ],
};

module.exports = definition;
```

Koenkk, 2023-12-06:
Could you create a pull request to add support for this device?

bummzulu, 2023-12-08:
would like to but have honestly no clue on how that works. Any instructions that you could share (didn't find when searching)

VyckaLTU, 2024-11-23:
> @VyckaLTU Can you tell if this device makes any noise when powers leds? Do you reccomend it? In lemona right now it cost 48eur so im thinking to buy it... Also i see it supports color temperature too but in Z2M description only brightness is mentioned.

Hi @Tommixoft. About noise. I have like 18 of them. But none of them are running in full power. So cant tell if they make sound while using i full power. There is 2 types of these drivers. One has color temperature control. Other dont have. Both model has brightness controll. I made a mistake and bought with color temperature ones. While my leds dont support it. I can confirm z2m has settings to control color temperature. But while my leds not supporting it im not using it. But one more thing about noise. This model with color temperature makes buzzing if i set it on not full brightness and max color temperature. If it is in max no noise at all for me. But if  i set less brightness. Lets say 50-60% it makes buzzing noise. I think it is related with with my led. I was thinking to buy one pcs of driver without color temperature control to test if it makes any noise with my leds on lower brigtness. But not did yet :)

bummzulu, 2024-11-23:
I dont notice any noise - having the version without color control. I can recommend it.

bummzulu, 2024-11-24:
I examined now the sound a bit closer. The dimmer is hidden a bit in my installation which of course provides some sound isolation. Between 20-80% I hear some soft sound. Noticeable only in a very quiet environment and when the dimmer is not sound isolated at all though some cover or whatever. Hope that helps.