Koenkk/zigbee2mqtt #20068: [Feature request]: Additional options for TS0601_Thermostat_1

mrslvd, 2023-12-06:
### Is your feature request related to a problem? Please describe

I bought TuYa TRV recognized as  "TS0601_thermostat_1" and manufacturer "_TZE200_rtrmfadk".
I have 2 issues:
1. Valve position is not shown;
2. I can't change mode from "Energy-saving mode" (this term is used in user manual) where TRV is working as ON/OFF to "Comfort mode", where PID is implemented. This can be done using TuYa gateway and App as stated in user manual

### Describe the solution you'd like

Possibility to change mode and show valve position.

### Describe alternatives you've considered

-

### Additional context

-

rexfue, 2023-12-14:
I have the same thermostat and urgently need the 'position' feature

mrslvd, 2023-12-17:
Hi rexfue,
Actually, I believe the position is not reported by TRV's firmware. If I am right then for z2m will not be possible to capture position for TRV recognized with "_TZE200_rtrmfadk" manufacturer. 
Changing the mode, I believe is possible to be implemented.
It will be great if I can find where I can buy TRVs with:
- PID implemented;
- 5% Position step (or less but for me it is pointless less then 5%);
- position reporting.
- QUIET operation;
- reliable operation.
And supported by z2m.
I got 3 different models here... but one of them is not quiet and is fragile (3 TRVs was changes for 3 winters), other has 25% step (in fact it is working as ON/OFF), and third does not report position and PID mode can't be turned on from z2m.

mrslvd, 2023-12-17:
Hi rexfue,
I did some experiments with following this:
- https://github.com/Koenkk/zigbee-herdsman-converters/issues/6482
- https://github.com/Koenkk/zigbee2mqtt/issues/18892
I made some mix between posted converters and now I have both position and PID
I can post my file but I am not sure whether I made things in right way....

tml89, 2024-01-23:
@mrslvd: can you please describe what you did?

mrslvd, 2024-01-26:
@tml89 , When in home I'll paste the ext.js file I assembled following the above links. I think z2m already add position  monitoring in standard converter, but enabling PID is still not implemented

mrslvd, 2024-01-27:
@tml89 , Attached is ext.js . This is external converter I am using for my TRV. Please, keep on mind that the file was assembled using the above links while I am not expert and some unnecessary/extra rows/data may exist. At my place everything works fine with this external converter.
My TRVs are  recognized as "TS0601_thermostat_1" and manufacturer "_TZE200_rtrmfadk". In the external converter some additional manufacturers are included which I did not own and test.
[ext.zip](https://github.com/Koenkk/zigbee2mqtt/files/14072805/ext.zip)

closed 2024-08-27 by github-actions[bot]