Koenkk/zigbee2mqtt #20069: Philips 9290012573A: setting effect does not work for some effects

Soukyuu, 2023-12-06:
### What happened?

Setting effect to `candle`, `fireplace` or `colorloop` does nothing.
The `okay`, `blink` and `breathe` effects work. Not sure what `channel_change` is supposed to do, it turns the light orange.

I've first encountered it with the newest deconz conbee II firmware but have since downgraded in an attempt to increase stability as recommended in #9554

Look for the `wohnzimmer_tv_links_licht` in the log, that's the one I attempted to set the non-working effects on.

### What did you expect to happen?

The `candle` and `fireplace` effect to make the light flicker, and for the `colorloop` to transition through the hue wheel as it does for Philips 929002294203

### How to reproduce it (minimal and precise)

Set the effect to `candle`, `fireplace` or `colorloop` from the frontend -> device -> details view

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

0x26580700

### Adapter

deconz Conbee II

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13578613/log.txt)

OUARZA, 2024-01-03:
Hello, were you able to find a solution?

Soukyuu, 2024-01-03:
Nope. Maybe this generation of the hue bulbs simply does not support those effecta and they were added by mistake. Or there is a bug, but I'm not seeing a user error here.

Soukyuu, 2024-01-03:
Actually... This is weird. Both the e14 candle and the e27 bulb identify as 9290012573A. 

With the 1.35.0 version, all effects work on the e14 white and color ambiance candle.

It still does not work for the e27 white and color ambiance bulb.

OUARZA, 2024-01-03:
I have a version that already worked.

nilsreiter, 2024-01-06:
I learned that this really also depends on the firmware of the bulbs. It's not exactly clear which firmware is needed for each device though. 1.108.7 seems to be the current one for LCT015, and there it works, but not with 1.101.2.

OUARZA, 2024-01-06:
Hello,
I am in 67.108.5 but for the moment I cannot update due to an update problem on the side of Philips and Zibgee2mqtt: https://github.com/Koenkk/zigbee-OTA/issues/420

Soukyuu, 2024-01-06:
My LCT007 are all on 67.88.1 and no OTAs available... is this related to the above bug?

OUARZA, 2024-01-06:
Good question, I don't remember if it's due to the firmware change that I have the problem.
I also know that Philips changed its API and moved to V2

closed 2025-01-09 by github-actions[bot]