Koenkk/zigbee2mqtt #20072: Problems and Improvements with Avatto Thermostat ZWT100 and ZWT198

MrEcosse, 2023-12-06:
### What happened?

There are a number of missing features with these thermostats and an issue with the rest day selection in programming mode.  
For the ZWT100-16A there is currently no ability to select the sensor (air or floor or both), the sound when you press the keys, the maximum temperature for the floor, and the ability to invert the relay output.

On both the ZWT100 and ZWT198 the rest day selection between 6-1 and 5-2 is incorrect.  Selecting 6-1 on the interface actually selects 5-2 on the thermostat and vice-versa.  This appears to be an error in tuya.valueConverter.workingDay.  I suspect this error occurs in all the thermostats which use the tuya.valueConverter.workingDay converter.

I have an example converter which corrects these issues

### What did you expect to happen?

For the features to be supported and correct.

### How to reproduce it (minimal and precise)

To reproduce the programming issue you need to put your thermostat into configuration mode from the thermostat itself.  Then issue the change of rest day commands from Z2M and you will see the thermostat selects the opposite of 5-2 and 6-1.

The other features were not supported and the existing converters usually show they don't know what datapoints 105, 106, 108 and 111 do.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230707

### Adapter

HamGeek

### Debug log

Not relevant as solution provided.

MrEcosse, 2023-12-06:
Here is a revised converter for the ZWT100-16A.  This includes the missing features and the correction of the workingDay setting.  The latter is accomplished by using a tuyaConvertrBasic.lookup function rather than the valueConverter.workingDay.  Also the Max temperature setting has a granularity of 1 degree not 0.5 so I changed that too.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const valueConverter = {
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),	
    runningState: tuya.valueConverterBasic.lookup({'heat': true, 'idle': false}),
    divideBy10: tuya.valueConverterBasic.divideBy(10),
};

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_lzriup1j']),
        model: 'ZWT100-16A',
        vendor: 'TuYa',
        description: 'Avatto wall thermostat for electric heating',
        onEvent: tuya.onEvent({timeStart: '1970'}),
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'key_sound', 'sensor_choice', 'sensor_limit', 'inverse']}],
        whiteLabel: [{vendor: 'AVATTO', model: 'ZWT-100-16A'}],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Full factory reset, use with caution!'),
            e.binary('key_sound', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Switches sound when interacting with thermostat'),
            e.child_lock(),
            e.climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET)
                .withPreset(['auto', 'manual', 'temporary_manual'])
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Antifreeze function'),
            e.max_temperature_limit()
                .withUnit('Â°C')
                .withValueMin(15)
                .withValueMax(90)
                .withValueStep(1)
                .withPreset('default', 60, 'Default value')
                .withDescription('Maximum upper temperature'),
             e.numeric('sensor_limit', ea.STATE_SET)
                .withUnit('Â°C')
                .withValueMin(35)
                .withValueMax(60)
                .withValueStep(1)
                .withPreset('default', 60, 'Default value')
                .withDescription('Floor sensor temperature limit'),
            e.numeric('deadzone_temperature', ea.STATE_SET)
                .withUnit('Â°C')
                .withValueMax(10)
                .withValueMin(0.5)
                .withValueStep(0.5)
                .withPreset('default', 1, 'Default value')
                .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
            e.binary('inverse', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Primary and secondary reverse selection.  ON is Auxillary reversal'),
            e.enum('sensor_choice', ea.STATE_SET, ['inside', 'outside', 'all'])
[... 29 lines cut ...]
                [109, 'schedule_holiday', tuya.valueConverter.ZWT198_schedule],
                [110, 'backlight_mode', tuya.valueConverter.backlightModeOffLowMediumHigh],
                [111, 'inverse', tuya.valueConverter.onOff], 

		
                // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
                // For each period:
                // 1st byte: hour
                // 2nd byte: minute
                // 3rd, 4th bytes: temperature multiplied by 10
                // On the device last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
                // scheduling can't be configured at all on the device.
                // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
                // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
                // Then the schedule is:
                // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
                // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
            ],
        },
    };

module.exports = definition;
```

Hope this helps

MrEcosse, 2023-12-06:
Also just to confuse us on the ZWT100 the Preset values for Auto and Manual are different to the ZWT198.  The settings above are correct for the ZWT100

tispokes, 2023-12-20:
@MrEcosse  another issue I've to address:
Setting the current_heating_setpoint with 0.5 steps is possible on the device but not via z2m. When you set a x.5 value, the value step back to 0 degrees.
Do you issuing the same behaviour?

Ronayke, 2024-02-04:
I can't find the correct input for 'Schedule weekday' and 'Schedule holiday'. Per definition, it should be for 'Schedule holiday': 2 times `hh:mm/cc.c°C`). Unfortunately, when I put '07:00/21.0°C 23:00/17.0°C', this error is shown: "Publish 'set' 'schedule_holiday' to 'Thermostat' failed: 'TypeError: Cannot read properties of undefined (reading 'trim')'"

Any idea how to solve this?

Ronayke, 2024-02-04:
I can't find the correct input for 'Schedule weekday' and 'Schedule holiday'. Per definition, it should be for 'Schedule holiday': 2 times hh:mm/cc.c°C). Unfortunately, when I put '07:00/21.0°C 23:00/17.0°C', this error is shown: "Publish 'set' 'schedule_holiday' to 'Thermostat' failed: 'TypeError: Cannot read properties of undefined (reading 'trim')'"

Any idea how to solve this?

I have model TS0601 (_TZE200_viy9ihs7)

tispokes, 2024-02-04:
> I can't find the correct input for 'Schedule weekday' and 'Schedule holiday'. Per definition, it should be for 'Schedule holiday': 2 times hh:mm/cc.c°C). Unfortunately, when I put '07:00/21.0°C 23:00/17.0°C', this error is shown: "Publish 'set' 'schedule_holiday' to 'Thermostat' failed: 'TypeError: Cannot read properties of undefined (reading 'trim')'"
> 
> Any idea how to solve this?
> 
> I have a ZWT198 (Zigbee Model TS0601; Zigbee Manufacturer _TZE200_viy9ihs7)

Look into the docs, w/o unit
https://www.zigbee2mqtt.io/devices/X5H-GB-B.html#schedule-text

Ronayke, 2024-02-04:
As written in the docs of TuYa X5H-GB-B, I tried the following, but still with the same error:

Schedule weekday: 06:00/20.5 08:00/15 11:30/15 13:30/15 17:00/22 22:00/15
Schedule holiday: 06:00/20.5 08:00/15

Also tried 8 periods in each, but to no avail.

MrEcosse, 2024-02-05:
I initially connected my stat using the Tuya app so I could get the end points.  I found that the schedule fields then populated with the default schedule.  Then I could edit the times if I needed.  If the fields are showing blank can you try editing the schedule on the stat itself (just make one change) and see if that then populates the field.  If not then pair with the Tuya app briefly and adjust the schedule just once as then it should show in HA

Ronayke, 2024-02-06:
@MrEcosse , you solved this problem for me. Since I don't have a Tuya bridge, I only had one option left, and that is manually changing the schedule on the device. And indeed, then the values in 'Schedule weekday' and 'Schedule holiday' are visible in Zigbee2MQTT. Thank you for this workaround!

Now there is only one small issue: manual and auto mode are switched with each other in this model (ZWT198)... Does anyone has an easy solution for this?

MrEcosse, 2024-02-06:
Here is my code for the battery one fingerprint _TZE200_viy9ihs7.  This should have the Auto and Manual the correct way around.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_viy9ihs7']),
        model: 'ZWT198',
        vendor: 'TuYa',
        description: 'Avatto Battery Wall Thermostat',
        onEvent: tuya.onEvent({timeStart: '1970'}),
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [{...tuya.tz.datapoints, key: [...tuya.tz.datapoints.key, 'key_sound', 'inverse']}],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Full factory reset, use with caution!'),
	    e.binary('key_sound', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Switches sound when interacting with thermostat'),
            e.child_lock(),
            e.climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET)
                .withPreset(['auto', 'manual', 'temporary_manual'])
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Antifreeze function'),
            e.max_temperature_limit()
                .withUnit('C')
                .withValueMin(15)
                .withValueMax(90)
                .withValueStep(0.5)
                .withPreset('default', 60, 'Default value')
                .withDescription('Maximum upper temperature'),
            e.numeric('deadzone_temperature', ea.STATE_SET)
                .withUnit('C')
                .withValueMax(10)
                .withValueMin(0.5)
                .withValueStep(0.5)
                .withPreset('default', 1, 'Default value')
                .withDescription('The delta between local_temperature (5<t<35)and current_heating_setpoint to trigger Heat'),
            e.enum('backlight_mode', ea.STATE_SET, ['off', 'low', 'medium', 'high'])
                .withDescription('Intensity of the backlight'),
	    e.binary('inverse', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Primary and secondary reverse selection.  ON is Auxillary reversal'),
            e.enum('working_day', ea.STATE_SET, ['disabled', '6-1', '5-2', '7'])
                .withDescription('Workday setting'),
            e.text('schedule_weekday', ea.STATE_SET).withDescription('Workdays (6 times `hh:mm/cc.cC`)'),
            e.text('schedule_holiday', ea.STATE_SET).withDescription('Holidays (2 times `hh:mm/cc.cC)`'),
            // ============== exposes for found, but not functional datapoints:
            /*
            e.min_temperature_limit() // dp 16
                .withValueMin(5)
                .withValueMax(15)
                .withValueStep(0.5)
                .withPreset('default', 10, 'Default value')
                .withDescription('dp16 is listed in Tuya, but no communication from device'),
            */
        ],
        meta: {
            tuyaDatapoints: [
[... 15 lines cut ...]
                [109, 'schedule_weekday', tuya.valueConverter.ZWT198_schedule],
                [109, 'schedule_holiday', tuya.valueConverter.ZWT198_schedule],
                [110, 'backlight_mode', tuya.valueConverter.backlightModeOffLowMediumHigh],
		[111, 'inverse', tuya.valueConverter.onOff], 
                // ============== found but not functional datapoints:
                // [16, 'min_temperature_limit', tuya.valueConverter.divideBy10],  // datapoint listed in Tuya, but no communication from device

                // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
                // For each period:
                // 1st byte: hour
                // 2nd byte: minute
                // 3rd, 4th bytes: temperature multiplied by 10
                // On the device last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
                // scheduling can't be configured at all on the device.
                // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
                // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
                // Then the schedule is:
                // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
                // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
            ],
        },
    };

module.exports = definition;
```

Ronayke, 2024-02-07:
I tried this code, and everything works as it should (as far as I could see). Thanks!

However, the key_sound doesn't do anything on my device. I guess the ZWT198 doesn't have this functionality.

uxdesignerhector, 2024-02-11:
I have this problem for manual  and auto mode, they are switched in front end for the  Zigbee Manufacturer:  `_TZE200_viy9ihs7` model [ ZWT198/ZWT100-BH](https://www.zigbee2mqtt.io/devices/ZWT198_ZWT100-BH.html#tuya-zwt198%252Fzwt100-bh). It happened after updating the Home Assistant Add-on to `1.35.3-1`, before it was working properly. 

I guess this commit could be related:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6912

I'm going to checkout if rejoining the device solves the problem

uxdesignerhector, 2024-02-11:
> I have this problem for manual and auto mode, they are switched in front end for the Zigbee Manufacturer: `_TZE200_viy9ihs7` model [ ZWT198/ZWT100-BH](https://www.zigbee2mqtt.io/devices/ZWT198_ZWT100-BH.html#tuya-zwt198%252Fzwt100-bh). It happened after updating the Home Assistant Add-on to `1.35.3-1`, before it was working properly.
> 
> I guess this commit could be related: [Koenkk/zigbee-herdsman-converters#6912](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6912)
> 
> I'm going to checkout if rejoining the device solves the problem

Reconfiguring the device didn't work. Removing the device from the ZigBee network and rejoining didn't work ... =(

ckegels, 2024-02-24:
i have the _TZE204_lzriup1j version and cant even set the temperature set point on? wont change it on the device i tried your code @MrEcosse but no dice. any clue what it could be?

my debug logs show that it should change but it doesnt

debug 2024-02-24 12:01:44Received MQTT message on 'zigbee2mqtt/Hoeveke living room /set' with data '{"current_heating_setpoint":15}'
debug 2024-02-24 12:01:44Publishing 'set' 'current_heating_setpoint' to 'Hoeveke living room '
info 2024-02-24 12:01:44MQTT publish: topic 'zigbee2mqtt/Hoeveke living room ', payload '{"backlight_mode":"high","child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":null,"factory_reset":"OFF","frost_protection":null,"inverse":"OFF","key_sound":null,"linkquality":72,"local_temperature":12.7,"local_temperature_calibration":0,"max_temperature_limit":null,"preset":"temporary_manual","running_state":"heat","schedule_holiday":null,"schedule_weekday":null,"sensor":"internal","sensor_choice":null,"sensor_limit":null,"system_mode":"heat","working_day":null}'

ckegels, 2024-02-24:
Having some issues with these.
i got 3 of them and one seems to be working fine. all 3 worked but after a couple minutes 2 started having issues.
i get updates from them like usual and can change the set temperature but the devices stay in heating.
one of them wont send the heating action and wont change it.

why does one work and not the others?
also they only wanted to pair with the coordinater directly but now are connected to 5 other devices and i get logs from each device every couple seconds.

MrEcosse, 2024-02-26:
Have you tried the code from these https://github.com/Koenkk/zigbee2mqtt/issues/21353

arkadi222, 2024-02-26:
@MrEcosse hi, could you tell me is this correct in zwt100 config:
`.withUnit('Â°C')`
what this sign means ?

tispokes, 2024-02-26:
> @MrEcosse hi, could you tell me is this correct in zwt100 config: `.withUnit('Â°C')` what this sign means ?

As mentioned by me, it's just a wrong space. Just replace the `Â` with a space

MrEcosse, 2024-02-26:
Sorry, the odd character is a text encoding issue.  Replace as @tispokes suggests.

MrEcosse, 2024-02-26:
@arkadi222 @tispokes Mine is similar too, so maybe you have a faulty one.  I agree they are decent thermostats and are replacing my Heatmiser system.

arkadi222, 2024-02-26:
I think it looks much better than mine. in fact, when viewed from above, it is slightly better.
I hope it was a defective one and the next 5 I ordered will "shine" better
[image]
[image]

arkadi222, 2024-02-26:
I've just return it to a seller :)
But I've ordered the same model from the same seller - I'll do some tests when I'll get it.
I've prepared cables on 1,4m height.

arkadi222, 2024-02-26:
ok, one more offtopic ;)
the latest zigbee-herdsman-converters is v.18.38.0.
But I've use it zigbee2mqtt as HA Addon. and there is z2m in v.1.35.3 and converters in v.18.28.3
is it possible to "override" converters with newest changes? or better to wait to next addon version from HA?

closed 2024-09-25 by github-actions[bot]

sundior, 2024-10-04:
Zakupiłem termostaty ZWT100-16A i mam z nimi kilka problemów. Brak szybkiej reakcji na zmianę temperatury, po załączeniu grzania na termostacie leci temperatura w dół i później ma problemy w osiągnięciu zakresu nastawu a w pokoju już upał, trzeci problem to w nwastawach jest możliwość co 0,5 stopnia załączanie ale niestety tylko co stopień się wyłącza a na dodatek 0,5 stopnia poniżej nastawu i powyżej. Czy jest jakaś aktualizacja oprogramowania i jak ją zrobić

arkadi222, 2024-10-09:
Sytuacja przez Ciebie opisywana może wynikać z tego, że źródło ciepła jest przewymiarowane w stosunku do pomieszczenia lub na zewnątrz jest jeszcze dość ciepło, przez co temperatura dość szybko rośnie po włączeniu ogrzewania i nawet po wyłączeniu termostatu jeszcze rośnie.
W sypialni nawet jak jest ciepło na zewnątrz problemu praktycznie nie ma:
[image]
histereza 0,5 stopnia dokładnie to oznacza - że wyłącza grzanie po ochłodzeniu do temp. zadanej -0,5, a wyłącza grzanie po przekroczeniu zadanej+0,5. W moim przypadku jest to wystarczające. 
ale już w salonie gdzie powierzchnia grzania jest dużo większa wykres już wygląda tak:
[image]
tutaj faktycznie mimo, że termostat wyłącza się przy 23.5 to temperatura jeszcze rośnie do 24.7
jak zrobi się chłodniej na zewnątrz prawdopodobnie takiego problemu nie będize

sundior, 2024-10-09:
Dzień dobryNie w tym problem, źle pan opisał.Termostat pokazuje poprawnie do momentu gdy załączy się grzanie i wtedy temperatura na termostacie cały czas spada .Termostat nie jest podpięty do niczego a tylko podane jest napięcie do zasilania i gdy ustawimy 19 stopni i załączy się grzanie przy 18,5 stopnia a  temperatura w pokoju jest stała mierząc innym termostatem,. Pokazuje cały czas spadek temperatury i potrafi przez parę godzin zejść do temperatury nawet 12 stopni gdy ogrzeję palcem termistor termostat się wyłączy i zacznie temperatura narastać do temperatury który pokazuje inny termostat. Raczej powinna być temperatura stała te 18,5 stopnia lub oscylować w dziesiętnych a nie w kilku stopniach i to stale malejąca. Co ciekawe po przełączeniu  na wyjście drugie EXTERNAL SENSOR  (czujnik zewnętrzny) jest wszystko OK termostat zachowuje się poprawnie no ale to jest kabel.Moim zdaniem jest jakiś błąd w oprogramowaniu lub uszkodzone termostaty no ale wszystkie 9 jakie zakupiłem?W dniu 2024-10-09 12:58:27 użytkownik arkadi222  napisał:Sytuacja przez Ciebie opisywana może wynikać z tego, że źródło ciepła jest przewymiarowane w stosunku do pomieszczenia lub na zewnątrz jest jeszcze dość ciepło, przez co temperatura dość szybko rośnie po włączeniu ogrzewania i nawet po wyłączeniu termostatu jeszcze rośnie. W sypialni nawet jak jest ciepło na zewnątrz problemu praktycznie nie ma: image.png (view on web) histereza 0,5 stopnia dokładnie to oznacza - że wyłącza grzanie po ochłodzeniu do temp. zadanej -0,5, a wyłącza grzanie po przekroczeniu zadanej+0,5. W moim przypadku jest to wystarczające. ale już w salonie gdzie powierzchnia grzania jest dużo większa wykres już wygląda tak: image.png (view on web) tutaj faktycznie mimo, że termostat wyłącza się przy 23.5 to temperatura jeszcze rośnie do 24.7 jak zrobi się chłodniej na zewnątrz prawdopodobnie takiego problemu nie będize

sundior, 2024-10-13:
Czy jest jakaś rada na złe wskazania lub jak wgrać oprogramowanie alternatywne jakieś

sundior, 2024-11-02:
Widzę brak chęci pomocy

lorchee, 2024-11-11:
@sundior czy udało się ogarnąć ten błąd ze spadkiem temperatury po uruchomieniu grzania? Mam ten sam problem z Avatto wt410 3A. Temperatura wsazuje dany pomiar, a po uruchomieniu styku grzania, dodatkowo ona spada. Mimo ze inny czujnik wskazuje wzrost temperatury w pomieszczeniu.

sundior, 2024-11-11:
Niestety ale nie, brak chęci pomocy. Wszyscy myślą że to jest niemożliwe i piłkę odbijają że to niby źródło ciepła przewymiarowane. Ale to niema nic wspólnego ze złym działanie samego termostatu. Aliexpres sprzedaje takie uszkodzone termostaty i udają że wszystko w porządku. Ja osobiście przepiąłem na zewnętrzny czujnik tam już działa poprawnie, patrząc na połączenia to są takie same jak wewnętrzny i zewnętrzny czujnik i oba idą na procesor tak więc to tylko źle działające oprogramowanie. Na jednym wejściu działa a na drugim już nie. Pytałem czy ktoś by pomógł w graniu i poprawieniu oprogramowania ale wszyscy nabrali wody do ust.

lorchee, 2024-11-11:
No u siebie wiele razy sprawdzałem i zwyczajnie niemożliwe jest żeby po załączeniu styku na listwe odczyt spadł o jakieś 0.5 stopnia w kilka minut. Kiedy osiagnie ustawiona temperature i grzanie wyłącza się to podbija o tą samą wartość. Ewidentnie jest coś w oprogramowaniu termostatu. U siebie nie podepnę zewnętrznego czujnika do niego. Nie wiem nawet jak dostać się do kodu oprogramowania tego mojego termostatu bo może i bym znalazł przyczynę. Wygląda na jakąś ustawiona korekcję temperatury ale nie wiem jak to mogę nawet sprawdzic

arkadi222, 2024-11-14:
@sundior tu nie chodzi o brak chęci pomocy - inaczej byśmy Ci nie odpisywali - tylko o to, że ja nie wiem, jak jeszcze mógłbym Ci pomóc. Też mam 5 tych termostatów i jeden z nich zachowuje się dość dziwnie - czasem po kilku godzinach od  restartu, czasem po kilku tygodniach zaczyna wariować w komunikacji z2m - zamiast standardowych ramek co ok. 15s zaczyna siać 1-2 ramek na sekundę, w związku z tą komunikacją świruje i tracę możliwość sterowania z poziomu HA - konieczny jest fizyczny reset termostatu. Podejrzewam, że po prostu ten typ tak ma - i zawsze trzeba zamówić z Chin jedną sztukę więcej do testów. Zresztą z takimi objawami ja bym sprzęt reklamował i odesłał - ja też już jeden termostat odesłałem (bo praktycznie w ogóle nie świecił).
A przy okazji- czy ktoś z Was mógłby wkleić zdjęcie z instrukcji - w szczególności zależy mi opisie parametrów 01-11 z poziomu termostatu (koniecznie dla tej wersji 16A, bo wydaje mi się, że dla wersji 3A sprzedawanej w Polsce parametry są ciut inne).

sundior, 2024-11-17:
Chodzi mi o informacje jak jest możliwość ściągnięcia oprogramowania z termostatu i ponownie go wgranie.