Koenkk/zigbee2mqtt #20082: [New device support]: Tuya Presence mmWave Radar Sensor 24G , TS0225 _TZFED8_o7oe4n9a

MicroFire33, 2023-12-07:
### Link

https://fr.aliexpress.com/item/1005006016522811.html

### Database entry

{"id":85,"type":"Router","ieeeAddr":"0xa4c138480c8f3d85","nwkAddr":35901,"manufId":4417,"manufName":"_TZFED8_o7oe4n9a","powerSource":"Mains (single phase)","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,4,1280,8,4096],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\f\u0000\u0000\u0000\u0000\u0007","65507":"sdk:0.0.0;kfg:0.0.0;ker:0.0.0;","65508":0,"65534":0,"modelId":"TS0225","manufacturerName":"_TZFED8_o7oe4n9a","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":1,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00256c4da8","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1701946040712,"defaultSendRequestWhen":"immediate"}

### Comments

Cannot retrive any information from the device.

MicroFire33, 2023-12-11:
**Add External converter**

> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const definition = {
    fingerprint: [{ modelID: 'TS0225',manufacturerName: '_TZFED8_o7oe4n9a'}],
    model: 'TS0225',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    onEvent: tuya.onEventSetTime, 
    configure: async (device, coordinatorEndpoint, logger) => {
                const endpoint = device.getEndpoint(1);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genTime', 'genBasic', 'ssIasZone']);
            },
    fromZigbee: [tuya.fz.datapoints, fz.ias_occupancy_alarm_1],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
            e.occupancy(),
            e.illuminance_lux(),
            e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        ],
    meta: {
        tuyaDatapoints: [
            [181, 'illuminance_lux', tuya.valueConverter.raw],
            [182, 'target_distance', tuya.valueConverter.divideBy10],
        ],
    },
};
module.exports = definition;

MicroFire33, 2023-12-11:
Add picture:

[image]

MicroFire33, 2023-12-11:
[image]
[image]
[image]
[image]

GitTheFOuttaHere, 2023-12-11:
I have a sensor that looks identical. It has the model _TZFED8_o7oe4n9a. I made a converter that looked almost like yours, excepting I listed a datapoint which I thought was for presence/occupancy because it appeared to correlate in the debug logs. Unfortunately I did not get the same results as you. My occupancy is NULL, the link sig quality is probably right....the others are way off.

GitTheFOuttaHere, 2023-12-11:
[image]

GitTheFOuttaHere, 2023-12-11:
Can I ask if you have tried turning yours off (simulating a power cut/blackout) & seeing if the values change to something like mine when it comes back online?. I'm sure I had working at some point on the external converter showing the correct values. After a power cycle the values went to what you see in the photo above.

MicroFire33, 2023-12-12:
To begin with, sorry for my English, I am French.
I test when I get home from work to unplug it and plug it back in to see.

MicroFire33, 2023-12-12:
Indeed, I have the same problem.
Data numbers change on reboot
[image]
[image]

MicroFire33, 2023-12-12:
On the other hand, even after unplugging "Occupancy" still works but the other 2 values ​​are crazy.

MicroFire33, 2023-12-14:
After several reconnection tests, the "Occupancy" works every time, however now the distance is completely crazy but after restarting Z2M the Lux becomes normal again.

GitTheFOuttaHere, 2023-12-14:
I received my Tuya gateway. I found some data points (in Chinese) and have translated a few & added them to the external converter.

It's as far as I've got but the "Distance" & "Lux" change values at random still.

My sensor also does not work properly in the Tuya app either. I think these are faulty sensors.

GitTheFOuttaHere, 2023-12-14:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{ modelID: 'TS0225',manufacturerName: '_TZFED8_o7oe4n9a'}],
    model: 'TS0225',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    onEvent: tuya.onEventSetTime, 
    configure: async (device, coordinatorEndpoint, logger) => {
                const endpoint = device.getEndpoint(1);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genTime', 'genBasic', 'ssIasZone']);
            },

    fromZigbee: [tuya.fz.datapoints, fz.ias_occupancy_alarm_1],
    toZigbee: [tuya.tz.datapoints],

    exposes: [
            e.occupancy(),
            e.illuminance_lux(),
            e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
			e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(60).withValueStep(1).withUnit('s')
            .withDescription('Occupancy keep time'),
            e.numeric('sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(20).withValueStep(1).withUnit('S')
            .withDescription('Motion Sensitivity - 1 = Highest 20 = Lowest'),
    ],
    meta: {
        tuyaDatapoints: [
		    [1, 'occupancy', tuya.valueConverter.trueFalse1],
            [102, 'fading_time', tuya.valueConverter.raw],
			[110, 'sensitivity', tuya.valueConverter.raw],
			[181, 'illuminance_lux', tuya.valueConverter.raw],
            [182, 'target_distance', tuya.valueConverter.divideBy100],
        ],
    },
};

module.exports = definition;

MicroFire33, 2023-12-14:
I test when I get home from work, however I have an idea for the errors in my distance sensor.
Because when it was working it was on the floor and it aimed at the ceiling.
And now it is positioned horizontally to do the tests and in the manual it says that the distance must be 3 meters maximum with the ground.

GitTheFOuttaHere, 2023-12-14:
Sorry about the indentations on "102" & "182". Have to work out how to post without ruining the formatting.

Good find on the sensor being vertical rather than horizontal.

MicroFire33, 2023-12-14:
[image]
Remote update, I'm testing the sensor position this evening

GitTheFOuttaHere, 2023-12-14:
These were the datapoints I discovered, not sure what all of them mean.

Trigger sensitivity      - DP114                 触发灵敏度
Status interval change   - DP101-                 状态变化间隔
Light Illuminance        - DP181                 光照度
No One Delays            - DP102                 无人延时
Distance to target       - DP182                 距离
Distance Interval Data   - DP183                 距离区间数据
Distance Interval Switch - DP105-                距离区间开关
Sensitivity              - DP110
Request to send Someone  - DP126
Pattern Changes          - DP176-

GitTheFOuttaHere, 2023-12-14:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
fingerprint: [{ modelID: 'TS0225',manufacturerName: '_TZFED8_o7oe4n9a'}],
model: 'TS0225',
vendor: 'TuYa',
description: 'Smart Human presence sensor',
onEvent: tuya.onEventSetTime,
configure: async (device, coordinatorEndpoint, logger) => {
const endpoint = device.getEndpoint(1);
await reporting.bind(endpoint, coordinatorEndpoint, ['genTime', 'genBasic', 'ssIasZone']);
},

fromZigbee: [tuya.fz.datapoints, fz.ias_occupancy_alarm_1],
toZigbee: [tuya.tz.datapoints],

exposes: [
        e.occupancy(),
        e.illuminance_lux(),
        e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(60).withValueStep(1).withUnit('s')
        .withDescription('Occupancy keep time'),
        e.numeric('sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('S')
        .withDescription('Occupancy Sensitivity - 1 = Highest 10 = Lowest'),
		e.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(20).withValueStep(1).withUnit('S')
        .withDescription('Radar Sensitivity - 1 = Highest 20 = Lowest'),
],
meta: {
    tuyaDatapoints: [
    [1, 'occupancy', tuya.valueConverter.trueFalse1],
    [102, 'fading_time', tuya.valueConverter.raw],
    [110, 'sensitivity', tuya.valueConverter.raw],
    [181, 'illuminance_lux', tuya.valueConverter.raw],
    [182, 'target_distance', tuya.valueConverter.divideBy100],
	[114, 'motion_sensitivity', tuya.valueConverter.raw],
    ],
},

};

module.exports = definition;

GitTheFOuttaHere, 2023-12-14:
The above converter works okay for me.

Maybe your occupancy is different to mine & you can remove that line referencing the datapoint?.

Mine is certainly on DP1

All the best

MicroFire33, 2023-12-15:
[image]
The first tests are positive, thank you

MicroFire33, 2023-12-29:
> > The first tests are positive, thank you
> 
> You are most welcome. Hope everything goes well for you.
> 
> Merry Christmas

Good morning,
Thank you, happy New Year to you too.
And I installed the detector in my house and the results are conclusive.
Thanks again for your help.

GitTheFOuttaHere, 2023-12-30:
That's great news. Hopefully others will find the converter useful as well as I'm sure more people will receive these.

Have a great new year, friend. Stay safe in 2024 :)

closed 2024-07-29 by github-actions[bot]