Koenkk/zigbee2mqtt #20087: Bosh BSIR-EZ:  Cannot set alarm state

lieblinger, 2023-12-07:
### What happened?

It's not possible to set the alarm state and sensor values are not available (see screenshot):

[image]

````
Publish 'set' 'alarm_state' to '0x003c84fffee61514' failed: 'Error: Cluster 'ssIasWd' has no command '240''
````

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle-E

johnsethwatkins, 2023-12-07:
I have the same issue with a new Bosch siren paired today.

Identical situation to original poster. 1.34.0 z2m, Sonoff 3.0 stick, firmware 20210708.

Debug log:

ParalaXEngineering, 2023-12-07:
Hello, I've to report the same issue. As a complement:
- Setting the warning sound to "burglar" set the alarm on, putting it back to "stop" set the alarm off
- Getting the power source doesn't work either (Publish 'get' 'power_source' to 'Rue Sirène' failed: 'Error: Unhandled key toZigbee.rbshoszbeu.convertGet power_source')

Also on 1.3.4, with home assistant and home assistant stick as coordinator

ParalaXEngineering, 2023-12-24:
I tried again to pair it multiple times, on a repeater or a coordinator, with or without power supply but no luck.

Some stuff works, for instance I can set the siren sound to "burglar", which make some noise, but it is not really a possible workaround. Any news on this?

ParalaXEngineering, 2024-01-02:
I confirm the issue is still there on the latest version 1.35...

johnsethwatkins, 2024-01-08:
Yes, I agree, still not working

Archimede2, 2024-01-26:
Hello everyone. 

I am experiencing  exactly the same issue. 
I have two different system based on Raspberry 4 there are almost the same appart the zigbee Dongle. In my first system I'm using a zzh! CC2652R Multiprotocol RF Stick from Tindie and everything is working well including the Bosch alarme.
In my second system I'm using a brand new SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator but in this case I'm not able to use properly the Bosch Alarm. the both alarme have the same firmware.
The strange thing is the state respond and the both system are not giving the same answer. 
In the first system (the good one) I got this:
{
    "ac_status": true,
    "alarm": false,
    "battery": 100,
    "battery_low": false,
    "linkquality": 24,
    "restore_reports": true,
    "supervision_reports": false,
    "tamper": false,
    "test": false
}

and in the second one (the bad one) I got this:
{
    "battery": 71,
    "linkquality": 90,
    "power_source": "solar_panel",
    "siren_and_light": "only_light",
    "siren_volume": "high",
    "voltage": 3700
}
So some information are not able to go the Sonoff dongle. I don't know why. 
I'm gone wait few weeks, maybe someone will find a solution. Otherwise the only solution will be to buy an other dongle. 
Thanks for the next answer and the next solution !!!!!
I can't wait to read the next message. !!!

WEZANGO, 2024-01-26:
Same issue here. I've got Sonoff CC2652P as well. Will try my Skyconeect and see if that works.

ParalaX002, 2024-01-27:
For me the Sky Connect doesn’t work :( I’ve got the same symptoms as described before.

WEZANGO, 2024-01-30:
Was this working on previous z2m versions?

johnsethwatkins, 2024-01-30:
No. this was a new device added in the release 1.34 and did not work then.

Archimede2, 2024-01-30:
> Was this working on previous z2m versions?

I do not know. My second system with the dongle sonoff is a new one.

IsThisACow, 2024-01-31:
+1 
Just updated to 1.35.1 to get it working.
Sonoff Dongle with 6.10.3.0 build 297
Availability is Disabbled. Don't have any idea how to change that.
Errormessage is the same as @lieblinger has it.

ParalaXEngineering, 2024-02-07:
Hi,

This is working now, thanks a lot to all :) I can set the state of the alarm so this is very nice.

A detail remaining is that the status of the energy source is not working correctly now, with this error:

Publish 'get' 'power_source' to 'Rue Sirène' failed: 'Error: Unhandled key toZigbee.rbshoszbeu.convertGet
power_source'

Strangely, also, it seems that the battery level is not updated automatically, but that setting the power source (which doesn't make sense because I'm only solar), is getting the battery level. Someone else know how to get reliably the battery level?

Chrunchy, 2024-04-13:
Besides the battery status, should everything now be working?
I can set sirene on and off, by setting alarm_state to on. But shouldnt there be a state to put the alarm into some active state, so the siren goes off if the tamperswitch is touched?

{
    "alarm_state": "OFF",
    "battery": 55,
    "battery_low": true,
    "light_delay": 0,
    "light_duration": 7,
    "linkquality": 102,
    "siren_and_light": "siren_and_light",
    "siren_delay": 0,
    "siren_duration": 7,
    "siren_volume": "medium",
    "voltage": 3600,
    "ac_status": null,
    "power_source": "solar_panel",
    "tamper": null,
    "test": null
}

Chrunchy, 2024-04-14:
I tried creating an external converter trough Z2M - that results in the tamperswitch working and reporting its currect state - but then other stuff stops working. I guess this device isn't fully compatible with Z2M yet before some code update has been performed by someone with the right knowledge :)

Alepat62, 2024-08-08:
> I tried creating an external converter trough Z2M - that results in the tamperswitch working and reporting its currect state - but then other stuff stops working. I guess this device isn't fully [...]

I've got the same issue with "tamper": null, did you found a solution ?

closed 2025-01-09 by github-actions[bot]