Koenkk/zigbee2mqtt #20098: Mercator Ikuü SSRM-ZB (TS0601) -  Rotary dimmer mechanism

bjammin, 2023-12-08:
### What happened?

This related to the device added here:
https://github.com/Koenkk/zigbee2mqtt/issues/18847

The issue seems the seem as the one here, for the same device from a different brand:
https://github.com/Koenkk/zigbee2mqtt/issues/13800

1. Changing brightness level (either with physical dial, slider on "exposes" tab, or in HA) generates lots of these warnings:

`Zigbee2MQTT:warn  2023-11-22 13:32:38: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x84ba20fffe97ca04 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,55]}}': type='commandDataResponse', datatype='value', value='55', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]`

2. The min brightness and max brightness on the "exposes" tab do not seem to save or the state get reloaded. However they seem to have some effect, but it's hard to tell exactly what as they are always reset when you come back to the page.

### What did you expect to happen?

No warning messages in log when changing brightness.

Min and max brightness state preserved and reloaded.

### How to reproduce it (minimal and precise)

1. Change brightness through any means (either with physical dial, slider on "exposes" tab, or in HA)
2. Attempt to set min or max brightness on exposes tab for device.

### Zigbee2MQTT version

1.33.2-1

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

info  2023-12-08 11:26:07: MQTT publish: topic 'zigbee2mqtt/dimmer_corridor_ceiling', payload '{"brightness":156,"level":618,"linkquality":25,"state":"OFF"}'
warn  2023-12-08 11:26:07: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x84ba20fffe97ca78 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,106]}}': type='commandDataResponse', datatype='value', value='618', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]
info  2023-12-08 11:26:07: MQTT publish: topic 'zigbee2mqtt/dimmer_corridor_ceiling', payload '{"brightness":156,"level":618,"linkquality":29,"state":"OFF"}'
info  2023-12-08 11:26:14: MQTT publish: topic 'zigbee2mqtt/dimmer_corridor_ceiling', payload '{"brightness":65,"level":264,"linkquality":18,"state":"OFF"}'
warn  2023-12-08 11:26:14: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x84ba20fffe97ca78 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,8]}}': type='commandDataResponse', datatype='value', value='264', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]
info  2023-12-08 11:26:14: MQTT publish: topic 'zigbee2mqtt/dimmer_corridor_ceiling', payload '{"brightness":65,"level":264,"linkquality":32,"state":"OFF"}'

scottyphillips, 2024-03-15:
I think this might be related to #21791 which is for the ikuu button version SSWM-DIMZ as I have exactly the same problems.

scottyphillips, 2024-03-16:
I have had a look at the code in tuya.js and I think 'TS0601_dimmer' is missing the tuyaDatapoints. Compare TS0601_dimmer with  TS0601_dimmer_1:

TS0601_dimmer: No TuyaDataPoints
```
      {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_whpb9yts'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ebwgzdqq'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ctq0k47x'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_9i9dt8is'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_dfxkcots'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_w4cryh2i'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_ojzhk75b'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_swaamsoy'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_9cxuhakf'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_a0syesf5'},
        ],
        model: 'TS0601_dimmer',
        vendor: 'TuYa',
        description: 'Zigbee smart dimmer',
        fromZigbee: [legacy.fromZigbee.tuya_dimmer, fz.ignore_basic_report],
        toZigbee: [legacy.toZigbee.tuya_dimmer_state, legacy.toZigbee.tuya_dimmer_level],
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
        exposes: [e.light_brightness().withMinBrightness().withMaxBrightness().setAccess(
            'state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).setAccess(
            'min_brightness', ea.STATE_SET).setAccess('max_brightness', ea.STATE_SET)],
        whiteLabel: [
            {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
            {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
            {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
            {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
            {vendor: 'Mercator Ikuü', model: 'SSWD01'},
            {vendor: 'Moes', model: 'ZS-USD'},
            {vendor: 'Moes', model: 'EDM-1ZBB-EU'},
            tuya.whitelabel('Mercator Ikuü', 'SSWM-DIMZ', 'Switch Mechanism', ['_TZE200_9cxuhakf']),
            tuya.whitelabel('Mercator Ikuü', 'SSWRM-ZB', 'Rotary dimmer mechanism', ['_TZE200_a0syesf5']),
        ],
    },
```

TS0601_dimmer_1: TuyaDataPoints included
```
    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ip2akl4w', '_TZE200_1agwnems', '_TZE200_la2c2uo9', '_TZE200_579lguh2',
            '_TZE200_vucankjx', '_TZE200_4mh6tyyo', '_TZE204_hlx9tnzb', '_TZE204_n9ctkb6j', '_TZE204_9qhuzgo0']),
        model: 'TS0601_dimmer_1',
        vendor: 'TuYa',
        description: '1 gang smart dimmer',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [tuya.exposes.lightBrightnessWithMinMax(), tuya.exposes.countdown(), tuya.exposes.lightType(),
            e.power_on_behavior().withAccess(ea.STATE_SET),
            tuya.exposes.backlightModeOffNormalInverted().withAccess(ea.STATE_SET)],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
                [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
                [4, 'light_type', tuya.valueConverter.lightType],
                [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
                [6, 'countdown', tuya.valueConverter.countdown],
                [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
                [21, 'backlight_mode', tuya.valueConverter.backlightModeOffNormalInverted],
            ],
        },
        whiteLabel: [
            {vendor: 'Lerlink', model: 'X706U'},
            {vendor: 'Moes', model: 'ZS-EUD_1gang'},
            tuya.whitelabel('Moes', 'ZS-SR-EUD-1', 'Star ring smart dimmer switch 1 gang', ['_TZE204_hlx9tnzb']),
            tuya.whitelabel('Moes', 'MS-105Z', 'Smart Dimmer module', ['_TZE200_la2c2uo9']),
        ],
    },
```

scottyphillips, 2024-03-16:
Hey @bjammin, I think I may have fixed this. Do you want to try the following external converter? It confirmed my assessment and its solved all of the problematic issues for the SSWM-DIMZ.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy')

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_whpb9yts'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ebwgzdqq'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ctq0k47x'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_9i9dt8is'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_dfxkcots'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_w4cryh2i'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ojzhk75b'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_swaamsoy'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_9cxuhakf'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_a0syesf5'},
    ],
    model: 'TS0601_dimmer_test',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.lightBrightnessWithMinMax()],
    meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
                [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
                [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
            ],
        },
        whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
        {vendor: 'Moes', model: 'ZS-USD'},
        {vendor: 'Moes', model: 'EDM-1ZBB-EU'},
        tuya.whitelabel('Mercator Ikuü', 'SSWM-DIMZ', 'Switch Mechanism', ['_TZE200_9cxuhakf']),
        tuya.whitelabel('Mercator Ikuü', 'SSWRM-ZB', 'Rotary dimmer mechanism', ['_TZE200_a0syesf5']),
    ],
};

module.exports = definition;
```

bjammin, 2024-03-18:
Hi @scottyphillips 

Thanks so much for this.  I can confirm that I was having the same issue as you mentioned in the other issue (after setting brightness to 100 in HA, manual control no longer works) in addition to the problems I reported.

I can also confirm that this converter seems to fix all of the issues!  I can now set min and max brightness as expected, and everything works smoothly.

Thanks for fixing this, and I hope it gets merged to main.

closed 2024-10-16 by github-actions[bot]