Koenkk/zigbee2mqtt #20108: EGLO Light Model EGLO_ZM_RGB_TW does not turn of with Transition enabled

dgruber99, 2023-12-08:
### What happened?

I activated the transition feature in Zigbee2MQTT with the aim of having the lights fade in and out. I set the value to 1 for the fading effect, which worked well. However, there's an issue - the lights no longer turn off; they stop at 1% brightness and remain on regardless of my attempts to turn them off. Removing the value completely restores their functionality.

thats in my Status page. "state" is "OFF" but the light is still on at around 1% 
{
    "brightness": 76,
    "color": {
        "h": 36,
        "hue": 36,
        "s": 70,
        "saturation": 70,
        "x": 0.315,
        "y": 0.245
    },
    "color_mode": "color_temp",
    "color_temp": 370,
    "color_temp_startup": 370,
    "linkquality": 192,
    "state": "OFF",
    "power_on_behavior": null
}

i hope anybody can help me with that issue 
Thanks alot!! 

Vendor:
EGLO
Modell
900024/12253

### What did you expect to happen?

the Lights to fade in and out, until the lights turn out 

### How to reproduce it (minimal and precise)

Enter 1 or 0  in the Transition field in Zigbee2MQTT 

### Zigbee2MQTT version

 1.34.0

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

SONOFF Dongle Plus Modell-E

### Debug log

[debug_Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13620269/debug_Log.txt)

closed 2024-07-06 by github-actions[bot]

khongi, 2024-11-15:
I'm also experiencing this issue, however even with transition set to 0, the light stays turned on at minimum brightness.

khongi, 2024-11-15:
The solution was to completely remove transition setting -> remove the number from the Settings (specific) / Transition.

MichaelUray, 2025-01-01:
So is this a bug int the firmware of the Eglo light, or a problem with Zigbee2MQTT?