Koenkk/zigbee2mqtt #20110: [Wrong device]: model/description

smart-ctrl, 2023-12-08:
### Link

https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html

### Model

TS011F_plug_1

### Description

Smart plug (with power monitoring)

### Vendor

TuYa

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/TS011F_plug_1.jpg

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xREMOVED","nwkAddr":58862,"manufId":4107,"manufName":"_TZ3000_0zfrhq4i","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"manufacturerName":"_TZ3000_0zfrhq4i","zclVersion":3,"appVersion":66,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1702069307817,"defaultSendRequestWhen":"immediate"}

### Notes

Expected: Link of this device (product page) https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html

Detected as a TS011F_plug_3 even it have instant automatic reporting of power, voltage and current values. Changing or disable the poll interval (-1) doesn't affect the reporting interval at all, it's always instant. So I'm sure it doesn't need to poll anything.

Koenkk, 2023-12-09:
Only the energy is polled for this device (voltage/power/current are reported).

smart-ctrl, 2023-12-09:
Well, I'm not sure about that. I did set the polling to -1 (disabled) and it's still reporting new value after I changed it.

Info 2023-12-09 19:00:02MQTT publish: topic 'zigbee2mqtt/0xa4c1388e020857be', payload '{"child_lock":"UNLOCK","current":0.333,"energy":0.05,"indicator_mode":"off/on","linkquality":255,"power":67,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":233}'
Info 2023-12-09 19:00:35MQTT publish: topic 'zigbee2mqtt/0xa4c1388e020857be', payload '{"child_lock":"UNLOCK","current":0.333,"energy":0.06,"indicator_mode":"off/on","linkquality":255,"power":67,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":233}'

Koenkk, 2023-12-10:
After setting this, did you restart z2m?

smart-ctrl, 2023-12-10:
Yes, I did.

Koenkk, 2023-12-10:
cool, will be detected as `TS011F_plug_1` now 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-10 by Koenkk