Koenkk/zigbee2mqtt #20111: [Wrong device]: TuYa valve actuator

tlc76, 2023-12-08:
### Link

https://www.aliexpress.com/item/1005005499231547.html?spm=a2g0o.order_list.order_list_main.5.310b18028GbGuD

### Model

ZN231392 or similar

### Description

Water/Gas smart valve actuator

### Vendor

tuya (?)

### Picture (link)

https://ibb.co/nM19q9W

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xa4c13875e2208989","nwkAddr":61169,"manufId":4417,"manufName":"_TZ3000_wpueorev","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000fU\u0006-\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_wpueorev","powerSource":1,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe9cd9d4","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1702070632295,"defaultSendRequestWhen":"immediate"}

### Notes

The device is wrongly recognised as a TS0001 switch:
https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001

However, the exposed channels are correct for this type of device (smart water/gas valve actuator):
- State (on/off)
- Power-on behavior (off/previous/on)
- Linkquality

_Notes:_

- The actual product is slightly different than the "regular" tuya ZN231392.
- The LED is very close to the power button, both being centered vertically.
- The "regular" ZN231392 has the power button on the left side, and the LED on the right side.

I opened the device, the gear mechanism is very similar to the "regular" ZN231392. All gears are metallic.

mrmaximas, 2023-12-09:
Sure, "original" is the
{"id":42,"type":"Router","ieeeAddr":"0xa4c13807f04b19ac","nwkAddr":64302,"manufId":4417,"manufName":"_TZ3000_iedbgyxt","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\f\u0000\u0000\u0000\u0012\r\u0000\u0000\u0000\u0012)\u0000\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_iedbgyxt","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"moesStartUpOnOff":2,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"tuyaBacklightSwitch":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1259610371},"lastSeen":1702113063714,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-12-09:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-09 by Koenkk

mrmaximas, 2023-12-09:
@Koenkk it's possible to change exposes for such valve types from switch state on/off to valve state?