Koenkk/zigbee2mqtt #20118: Error in DeconzAdapter when connecting Conbee3 with Xiaomi Aqara E1 Radiator

jannewulf, 2023-12-09:
### What happened?

zigbee2mqtt throws an exception when trying to configure my Aqara Smart Radiator Thermostat E1. 
Yesterday it was able to read values from the radiator but not write to it. Now, after I tried some things it already fails to configure the device.

See the logs
```
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:31: Logging to console and directory: '/app/data/log/2023-12-09.01-22-31' filename: log.txt
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:31: Starting Zigbee2MQTT version 1.34.0 (commit #aae7312)
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:31: Starting zigbee-herdsman (0.25.0)
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:33: zigbee-herdsman started (resumed)
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:33: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":77,"product":0,"revision":"0x264d0900","transportrev":0},"type":"ConBee3"}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:33: Currently 0 devices are joined:
zigbee2mqtt  | Zigbee2MQTT:warn  2023-12-09 01:22:33: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt  | Zigbee2MQTT:warn  2023-12-09 01:22:33: Allowing new devices to join.
zigbee2mqtt  | Zigbee2MQTT:warn  2023-12-09 01:22:33: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:33: Zigbee: allowing new devices to join.
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:35: Connecting to MQTT server at mqtt://mosquitto
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:36: Connected to MQTT server
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:36: Started frontend on port 10003
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"aae7312","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":77,"product":0,"revision":"0x264d0900","transportrev":0},"type":"ConBee3"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.34.0"}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 01:22:36: Zigbee2MQTT started!
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:35:28: Device 'living-room/heating' joined
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:35:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"living-room/heating","ieee_address":"0x54ef44100073f50c"},"type":"device_joined"}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:35:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"living-room/heating"},"type":"device_connected"}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:35:28: Starting interview of 'living-room/heating'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:35:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"living-room/heating","ieee_address":"0x54ef44100073f50c","status":"started"},"type":"device_interview"}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:35:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"living-room/heating"},"type":"pairing"}'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:36:29: Successfully interviewed 'living-room/heating', device has successfully been paired
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:36:29: Device 'living-room/heating' is supported, identified as: Xiaomi Aqara Smart Radiator Thermostat E1 (SRTS-A01)
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-09 09:36:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara Smart Radiator Thermostat E1","exposes":[{"access":1,"description":"Indicates if the device is in setup mode (E11)","label":"Setup","name":"setup","property":"setup","type":"binary","value_off":false,"value_on":true},{"features":[{"access":7,"description":"Temperature setpoint","label":"Occupied heating setpoint","name":"occupied_heating_setpoint","property":"occupied_heating_setpoint","type":"numeric","unit":"°C","value_max":30,"value_min":5,"value_step":0.5},{"access":1,"description":"Current temperature measured by the internal or external sensor","label":"Local temperature","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":7,"description":"Mode of this device","label":"System mode","name":"system_mode","property":"system_mode","type":"enum","values":["off","heat"]},{"access":7,"description":"Mode of this device (similar to system_mode)","label":"Preset","name":"preset","property":"preset","type":"enum","values":["manual","away","auto"]}],"type":"climate"},{"access":7,"description":"Select temperature sensor to use","label":"Sensor","name":"sensor","property":"sensor","type":"enum","values":["internal","external"]},{"access":1,"description":"Indicates if this valve is calibrated, use the calibrate option to calibrate","label":"Calibrated","name":"calibrated","property":"calibrated","type":"binary","value_off":false,"value_on":true},{"access":7,"description":"Calibrates the valve","label":"Calibrate","name":"calibrate","property":"calibrate","type":"enum","values":["calibrate"]},{"features":[{"access":7,"description":"Enables/disables physical input on the device","label":"State","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"label":"Child lock","type":"lock"},{"features":[{"access":7,"description":"Enables/disables window detection on the device","label":"State","name":"state","property":"window_detection","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"label":"Window detection","type":"switch"},{"access":1,"label":"Window open","name":"window_open","property":"window_open","type":"binary","value_off":false,"value_on":true},{"description":"Determines if temperature control abnormalities should be detected","features":[{"access":7,"description":"Valve detection","label":"State","name":"state","property":"valve_detection","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"label":"Valve detection","type":"switch"},{"access":1,"description":"Notifies of a temperature control abnormality if valve detection is enabled (e.g., thermostat not installed correctly, valve failure or incorrect calibration, incorrect link to external temperature sensor)","label":"Valve alarm","name":"valve_alarm","property":"valve_alarm","type":"binary","value_off":false,"value_on":true},{"access":7,"description":"Away preset temperature","label":"Away preset temperature","name":"away_preset_temperature","property":"away_preset_temperature","type":"numeric","unit":"°C","value_max":35,"value_min":-10},{"access":1,"description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Remaining battery in %, can take up to 24 hours before reported.","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Number of power outages (since last pairing)","label":"Power outage count","name":"power_outage_count","property":"power_outage_count","type":"numeric"},{"access":1,"description":"Temperature of the device","label":"Device temperature","name":"device_temperature","property":"device_temperature","type":"numeric","unit":"°C"},{"features":[{"access":7,"description":"When being ON, the thermostat will change its state based on your settings","label":"State","name":"state","property":"schedule","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Smart schedule configuration (default: mon,tue,wed,thu,fri|8:00,24.0|18:00,17.0|23:00,22.0|8:00,22.0)","label":"Schedule settings","name":"schedule_settings","property":"schedule_settings","type":"text"},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SRTS-A01","options":[{"access":2,"description":"Controls the temperature unit of the the [... cut ...]
```

### What did you expect to happen?

That I can read values from the radiator and also write values to the radiator. 

### How to reproduce it (minimal and precise)

Have Conbee3 plugged in and have the Aqara Radiator ready. Run the compose.yml. Connect the Aqara Radiator.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

0x264d0900

### Adapter

Conbee 3

### Debug log

[zigbee2mqtt-debug-logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/13623798/zigbee2mqtt-debug-logs.txt)

herbetom, 2023-12-27:
I ran into the same problem (interstingly with quite a similar setup (Pi3b+, ConBeeIII, Aqara E1 Radiator)).

Looks a lot like #11647. Even if that's for the ConBee II.

I tried a few things. What helped the most is probably setting the adapter_delay to 200ms in the config:

```
advanced:
  adapter_delay: 200
```

But i also
- updated the firmware from 0x264d0900 to 0x264e0900: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
- Tried a powered USB Hub (that didn't seem to help)
- added a USB Extenstion cable. Not sure about that.

and there might have been other steps i already forgot.

It's still not 100% reliable, but it currently works mostly. That's better then the not working at all from before.

cnaslain, 2024-01-02:
This delay config apparently fix my case too. It is described into https://github.com/Koenkk/zigbee2mqtt/issues/4884 and now the parameter is also documented into https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration

herbetom, 2024-01-02:
Just for completeness: Because I've had, in addition to the receive (send from zigbee adapter but never received by the Thermostat) errors, the problem that the Xiaomi Aqara E1 Radiator did many (in my opinion useless and quite annoying) micro-adjustemns every few minutes once the target temperature had been reached.

I've since swapped it with an SONOFF TRVZB and it works much better for me.
Not only are those annoying micro adjustments gone but the SONOFF also seems to receives commands faster und with a higher reliability.

I haven't tested yet if I can remove the adapter_delay but while that helped with the Aqara it still didn't work every time and it took some time while the SONOFF receives the commands almost always and much quicker.

closed 2024-07-31 by github-actions[bot]